Edzelf/ESP32-Radio #2: CPU halted

zajaklis, 2017-05-23:
sometimes after pressing "save" and then after a couples of seconds "restart" im getting:

D: Command client available
D: handlehttp started
D: Get command is: reset
D: Start reply for reset
D: Command: reset with parameter 0
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
assertion "false && "item should have been present in cache"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_item_hash_list.cpp", line 85, function: void nvs::HashList::erase(size_t)
abort() was called at PC 0x400da8db on core 0
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x40008155:0x3ffccdf0 0x40007d16:0x3ffcce10 0x4010a99f:0x3ffcce40 0x4010aff5:0x3ffcce60 0x4010b2cc:0x3ffccea0 0x4010a4a7:0x3ffccef0 0x4010a03e:0x3ffccf40 0x4010a08e:0x3ffccf60 0x400d7aeb:0x3ffccf80 0x4011f7ee:0x3ffccfa0 0x400e474d:0x3ffccfd0

CPU halted.

Edzelf, 2017-05-23:
Yes, I think I can fix this.

Edzelf, 2017-05-23:
Are you sure you have the last version of NVS? Ther was a bug in in it until a week ago.
See https://github.com/espressif/arduino-esp32/issues/365
There is also a testprogram for this bug at the URL above.

zajaklis, 2017-06-05:
i was not able to reproduce this issue again, closing

closed 2017-06-05 by zajaklis