Edzelf/ESP32-Radio #3: Add support for Adafruit Feather ESP32 board

bbx10, 2017-05-24:
@edzelf  Many thanks for the new ESP32 version. I had no problems getting it work on the Feather ESP32 board. The only other change is to undefine USETFT since I do not have the display. It works great using the Web interface. The Web interface is very fast even when playing 256K streams.

The patch follows. I can submit a PR if desired.

```
--- a/Esp32_radio.ino
+++ b/Esp32_radio.ino
@@ -106,9 +106,17 @@
 #define WHITE   BLUE | RED | GREEN
 // Digital I/O used
 // Pins for VS1053 module
+#if defined(ARDUINO_FEATHER_ESP32)
+// Adafruit ESP32 Feather with Adafruit VS1053 Wing
+#define VS1053_CS     32
+#define VS1053_DCS    33
+#define VS1053_DREQ   15
+#define SDCARDCS      14
+#else
 #define VS1053_CS     5
 #define VS1053_DCS    16
 #define VS1053_DREQ   4
+#endif
 // Pins CS and DC for TFT module (if used, see definition of "USETFT")
 #define TFT_CS 15
 #define TFT_DC 2
@@ -1786,6 +1794,11 @@ void setup()
     
   Serial.begin ( 115200 ) ;                              // For debug
   Serial.println() ;
+#if defined(SDCARDCS) && (SDCARDCS >= 0)
+  // Deselect SDCARD
+  pinMode(SDCARDCS, OUTPUT);
+  digitalWrite(SDCARDCS, HIGH);
+#endif
   // Print some memory and sketch info
   dbgprint ( "Starting ESP32-radio Version %s...  Free memory %d",
              VERSION,
```

Edzelf, 2017-05-24:
Thanks, I will add this to the sourcecode.

bbx10, 2017-05-24:
Thanks, I tried it the latest commit and it works fine on Feather ESP32!

closed 2017-05-26 by Edzelf