Edzelf/ESP32-Radio #4: Feature Request: Add A2DP when available

68erDirk, 2017-05-29:
One of the great improvements of ESP32 is the integrated bluetooth feature. For ESP-idf there is already an example for a a2dp sink (https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/a2dp_sink). I hope this will soon be available in the arduino esp32 as well. At that point I would appreciate to have it integrated in order to be able to connect a smartphone vie bluetooth and play the a2dp stream on my esp-radio.

locki-cz, 2017-07-27:
A2DP is very nice idea! I hope Edzelf will implement it :)

vadpol, 2018-01-18:
Well, I think that using of ESP32 web radio as A2DP source would be also very much interesting https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/a2dp_source/
In this case it will be possible to connect bluetooth enabled speakers to it, for example to listen to the web radio via  quality audio system.