Edzelf/ESP32-Radio #8: Does not enter AP mode after erasing Flash

bbx10, 2017-06-02:
Erase flash (IDF make erase_flash) then upload ESP32-radio firmware. ESP32-radio does not enter local AP mode. The following fixes the issue.

```C
@@ -1547,10 +1547,14 @@ bool connectwifi()
     if (  WiFi.waitForConnectResult() != WL_CONNECTED ) // Try to connect
     {
       localAP = true ;                                  // Error, setup own AP
     }
   }
+  else
+  {
+      localAP = true ;                              // no ssid/pw so setup local AP
+  }
   if ( localAP )                                        // Must setup local AP?
   {
     dbgprint ( "WiFi Failed!  Trying to setup AP with name %s and password %s.", NAME, NAME ) ;
     WiFi.softAP ( NAME, NAME ) ;                        // This ESP will be an AP
     delay ( 5000 ) ;
@@ -3572,6 +3576,6 @@ String httpheader ( String contentstype )
```

Edzelf, 2017-06-03:
I will include this code in the next realease.

lonelyjack, 2017-06-09:
Where can i paste this part to the code?

bbx10, 2017-06-09:
The file is Esp_radio.ino. The function name is connectwifi(). The line number is 1547 or thereabouts.

lonelyjack, 2017-06-23:
bbx10
Could you, please, make .ino and share it

bbx10, 2017-06-25:
Rename this to Esp32_radio.ino. github does not allow file names ending in ino.

[Esp32_radio.txt](https://github.com/Edzelf/ESP32-Radio/files/1100615/Esp32_radio.txt)

Edzelf, 2017-06-26:
Fixed.

lonelyjack, 2017-06-26:
Thank you, guys)))

closed 2017-06-27 by Edzelf