Edzelf/ESP32-Radio #9: Compilation?

lonelyjack, 2017-06-03:
Please, give some tips how to compile this project. Your tips for 8266 was great. Started to install libraries, but there's a lot mistakes like this:

C:\Users\Eugene\Desktop\ESP32_Radio\Esp32_radio.ino:90:26: fatal error: Adafruit_GFX.h: No such file or directory

compilation terminated.

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Eugene\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board ESP32 Dev Module.

I used this way
https://github.com/espressif/arduino-esp32/blob/master/doc/windows.md

Edzelf, 2017-06-03:
The warning about multiple WiFi library is normal.  GFX can be downloaded from Adafruit github.

lonelyjack, 2017-06-03:
C:\Users\Eugene\Desktop\ESP32_Radio\Esp32_radio_init.ino: In function 'void setup()':

Esp32_radio_init:20: error: redefinition of 'void setup()'

 void setup()

      ^

C:\Users\Eugene\Desktop\ESP32_Radio\Esp32_radio.ino:2004:6: note: 'void setup()' previously defined here

 void setup()

      ^

C:\Users\Eugene\Desktop\ESP32_Radio\Esp32_radio_init.ino: In function 'void loop()':

Esp32_radio_init:69: error: redefinition of 'void loop()'

 void loop()

      ^

C:\Users\Eugene\Desktop\ESP32_Radio\Esp32_radio.ino:2736:6: note: 'void loop()' previously defined here

 void loop()

      ^

C:\Users\Eugene\Desktop\ESP32_Radio\prefbug.ino: At global scope:

prefbug:3: error: redefinition of 'Preferences preferences'

 Preferences preferences ;

             ^

C:\Users\Eugene\Desktop\ESP32_Radio\Esp32_radio_init.ino:14:13: note: 'Preferences preferences' previously declared here

 Preferences preferences ;         // Instance of Preferences

             ^

C:\Users\Eugene\Desktop\ESP32_Radio\prefbug.ino: In function 'void setup()':

prefbug:8: error: redefinition of 'void setup()'

 void setup()

      ^

C:\Users\Eugene\Desktop\ESP32_Radio\Esp32_radio.ino:2004:6: note: 'void setup()' previously defined here

 void setup()

      ^

C:\Users\Eugene\Desktop\ESP32_Radio\prefbug.ino: In function 'void loop()':

prefbug:65: error: redefinition of 'void loop()'

 void loop()

      ^

C:\Users\Eugene\Desktop\ESP32_Radio\Esp32_radio.ino:2736:6: note: 'void loop()' previously defined here

 void loop()

      ^

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Eugene\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Multiple libraries were found for "SD.h"
 Used: C:\Users\Eugene\Documents\Arduino\hardware\espressif\esp32\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
redefinition of 'void setup()'

Edzelf, 2017-06-03:
Esp32_radio_init.ino is a separate sketch.  Put it in another directory.  And read the documentation first.

lonelyjack, 2017-06-03:
read the documentation a lot of times. thank you!!! flashed!!!

closed 2017-06-03 by lonelyjack