Edzelf/ESP32-Radio #10: Button connection

lonelyjack, 2017-06-04:
[image]
is it right?

Edzelf, 2017-06-04:
There is no pull-up required for GPIO12.  At least not on a standard Dev Board.  A switch to GND is sufficient.  A pull-up does not harm either.

lonelyjack, 2017-06-04:
Out of the resistor, the diagram is right?

Edzelf, 2017-06-05:
Yes, try it and look at the debug logging at the serial output.

lonelyjack, 2017-06-08:
What about other GPIOs? 13,14,17,22,25,26,27,32,33? Is there no pull-up required?

Edzelf, 2017-06-08:
Try it, or read the documentation.

lonelyjack, 2017-06-08:
There's an empty fields in your overview for these pins

lonelyjack, 2017-06-13:
Well, it's rebooting without pull-ups

closed 2017-06-13 by lonelyjack

Edzelf, 2017-06-13:
Now, take a wire and ground pins 13, 14,17,.... one by one.  Watch the logging.

Edzelf, 2017-06-13:
If you mean "it keeps rebooting", then you have a weak power supply.  Add a big capacitor like 3300 microFarad between 5 Volt and ground.