Edzelf/ESP32-Radio #11: Help with IHR?

koskee, 2017-06-06:
I have been trying to get the IHR stations working and have continued to fail both on this project as well as the esp8266 version.  I feel like I must be doing something wrong? 

The specific station I require access to has a call sign: `CHBD`
IHeartRadio:  http://www.iheartradio.ca/big-dog-92-7/
My preset is something like :` preset_00 = ihr/CHBDFM      # 00 - BigDog 92.7`

I'm wondering: 
do I have the formatting correct? 
Do I need to log in or something like that? 
Is that a valid call sign? 
Any other pointers? 

It seems lately that all of the examples will load and play, with the exception of anything that has to do with IHR. I can't get anything from ihr to play (although it does seem to pull the metadata properly,  at least) 

I'm going to attempt to figure iout how to cross reference this post and the ihr post from the other platform, since it looks like this one doesn't have nearly as much traffic as the other. 

(PS. please don't judge me based on this shitty channel, it's not for me, it's for the guys at work to help ease them into the idea of listening to something else for a change - this is a local station for me).

Edzelf, 2017-06-06:
I am not able to do some testing in the coming weeks, but I'm pretty sure that the examples in the default configuration were working.

koskee, 2017-06-07:
Hmm. And yet for some reason I don't think I've EVER  been able to play any of the ihr content, on neither of the two plattforms, and yet I can listen with my computer on their website just fine, so it doesn't really seem like a networking/connection issue. 

I've browsed through the code as well and nothing jumped out at me, however I must admit the XML functions are a bit beyond my understanding. All of the surrounding code appears to be pretty much airtight from what I can tell though.

koskee, 2017-06-19:
As i mentioned on the other issue referenced above, the issue turned out to be hardware. I bought a VS1053B board, but received a board with a VS1003B chip installed on it instead. The substituted chip does not have support for the AAC audio format, thus no stream that utilizes this format will work properly.

closed 2017-06-19 by koskee