Edzelf/ESP32-Radio #14: How to power without powerbank?

lonelyjack, 2017-06-11:
That's a great project, but when I power it with powerbank - it's ok, but when i try power it with samsung charger or ac/dc modules like these:
https://ru.aliexpress.com/item/AC-DC-5V-700mA-3-5W-Power-Supply-Buck-Converter-Step-Down-Module-for-Arduino/32395148959.html?spm=2114.13010608.0.0.yKp14D
https://ru.aliexpress.com/item/Free-Shipping-1pcs-lot-HLK-PM01-AC-DC-220V-to-5V-mini-power-supply-module-intelligent/32408565688.html?spm=2114.13010608.0.0.JET9WZ
there's a big trouble. There's no sound or it's squashing

koskee, 2017-06-11:
What does your setup look like? I've had good results placing a mid to large sized electrolytic capacitor (~1000uF+, I use a 2200uF) across the power rails (5v and ground pins, observe polarity). 
 The esp8266 was notoriously power hungry and would cause large spikes on the supply voltage/current, which would cause all kinds of weird behaviour in practice. I'm not sure if these chips are the same way, but a stable source of power can only be a good thing in terms of stability & clean output. 

Also, Ive found that if the voltage level on the XRST pin of the 1053 board is not kept fairly stable, you'll start to get mpeg compression glitches showing up in the output. So, if the above doesn't help, if the output is still dodgy, try putting a capacitor between the EN/XRST pins and ground. I currently have a 4.7uF electrolytic that I'm using, but probably almost any value would be beneficial as there shouldn't be much fluctuation on these pins. My only advice here would be to keep it fairly small, maybe between 1uF and 47uF maximum. You may also see benefit in the sub uF range as well, you'll have to see what works for you

koskee, 2017-06-12:
Also, the charger specs shouldn't matter all that much, as I can run mine reliably from a computer usb port (usb is 500ma max) and this is shared between a few different peripherals (although I have noticed that mine will change to the next preset if I plug a flash drive into my computer while it's running on the USBs power so it's obviously not a lot of headroom)

lonelyjack, 2017-06-12:
koskee, I have not esp8266, I have esp32. I've placed 4700uf 10v electrolytic capacitor across the power rails of esp32 devboard, and the same one across the power rails of vs1053.  These capacicators don't cross with each other, because I'm powering these boards from different ports of powerbank.

koskee, 2017-06-13:
> koskee, I have not esp8266, I have esp32. 

No, I understand, I was just saying that the esp32 could have a big appetite for power, depending whether espressif has done anything to change the fact. 

I would suggest connecting the ground of the esp32 to the ground of the vs1053, however it's likely that the two separate powerbank  ports have a common ground internally. That is,  unless youre using more than one powerbank..

Based on your description of "squishing" the output, it sounds a lot like the same issue I was having. The only thing that actually seemed to work was placing a (smaller.. 1~100uF) capacitor between  the XRST pin on the vs1053 board -> ground. This will cause the voltage on this pin to stabilize as well as filter out some of the noise/fluctuations,  which essentially solved all of my issues. Im fairly confident it should work for you as well.

lonelyjack, 2017-06-14:
Thank you, I'll try. Capacicator between xsrt and ground should be electrolitic?

koskee, 2017-06-14:
Yeah,  that's what I'm using,  4.7uF electrolytic. 

I'm guessing that just about any sort of cap would do the trick in this case, as all we're trying to do is to store a bit of power for the moments when the supply voltage takes a dive. The cap helps to hold that line @5V over any voltage sags (not for very long mind you, but that's all it should need for better output.) 

Make sure to observe the correct polarity when using electrolytics though or they can pop and all the goop inside will leak out everywhere.. 

Let me know how it goes.

zajaklis, 2017-06-15:
dont use those rubbish power supplies, they cannot deliver more than 250ma

lonelyjack, 2017-06-17:
Koskee, thank you again. I've connected cap between xrst and gnd - all works fine. At first I used 4,7 uF, there was a little squash sometimes. Then I used 47uF - all is good.

arlaor, 2017-06-17:
Hi  @lonelyjack, @koskee, @Edzelf, @zajaklis 
Can you please confirm if this project on the esp32 model is stable compared to the esp8266 module or have the same performance?

Edzelf, 2017-06-17:
Better stability.  Better performance.  More features.

koskee, 2017-06-18:
@arlaor 
See here:  #https://github.com/Edzelf/Esp-radio/issues/92 (last comment at bottom) 
The difference will absolutely amaze you. You'll likely wonder why you even bothered with the other version.

arlaor, 2017-06-18:
thanks @Edzelf  @koskee 
I'm waiting for the arrival of a pair of wemos lolin.
This is a Excellent Project.

closed 2017-06-19 by lonelyjack