Edzelf/ESP32-Radio #15: On via vol+ after reset

ilioss, 2017-06-18:
Hi Ed,

Finnaly the TFT received and connected.
Meanwhile the 8266 replaced by the ESP32 changed  the software for the latest version.
Indeed the 32 looks like more stable. The 8266 I actualy didn,t get it working after all in combunation with the geeetech shield.( lots of so called "glitching" ) there is a signal and data but not understandeble.
In the same time I placed the capacitors across the 5 Vcc -gnd and betweeh XRST-gnd hopefully will this prevent problems with the power supply spikes.
During my playing/testing I noticed that after a reset from the config page I had to give vol+ a push to get sound again. Is this a known bug?
Also after I received my TFT I didn,t get it working. Connect as discribed and tested again. I changed the TFT_ILI963C.cpp file on line 1096/7
No result. Than the blue 1.8 SPI 128x160 board type I realy do not understand this Most problaby this is the key to get my display working. Do you have a clue?
Further more I realy enjoy this project and like to learn a lot not only from the esp32 also from networking etc. A realy nice step further than just arduino.
Greeting
Theo

Edzelf, 2017-06-18:
Your config should cantain something like "volume=76" and NOT "mute."
Try to get your display working without the vs1053 connected.

ilioss, 2017-06-18:
Hi,Ed thanks for your dwift anser.I will test the display without the 1053.Let you know.Regards,Theo

ilioss, 2017-06-18:
Hi, All must be,

I tested the TFT display with the benchmark file as found in the ILI9163c zip and it works on a NANO.
A small stroke was not used!
Than I connect the display again to my radio and disconnect fully the VS1053( Alientek) no picture at all a clear white screen. Double checked the connections.
Compaired to the both files the benchmark.ino and the ESP32Radio.ino there is some slide difference in defining the CS and DC
 #define TFT_CS 15
#define TFT_DC 2  ESPRadio.ino ( yours so to say )

#define __CS 10
#define __DC 6   benchmark.ino

I.am not meaning the i/o number.

Can this make the difference?
What exactely you mean with the "bleu 1.8 SPI 128x160 board"  in your documentation?
I can only found a red or black board 

I,am cont. to have my display working.
Your advise is highly appreciated.

Kind regards,
Theo

Edzelf, 2017-06-18:
I used this one: [https://www.aliexpress.com/item/Smart-Electronics-1-8-Inch-128-160-Serial-SPI-TFT-LCD-Module-Display-PCB-Adapter-Power/32597366825.html?spm=2114.13010608.0.0.cqL62P](TFT).

Make sure your power supply is strong enough.  The ESP32 is very sensitive on this.
The GPIO pins are correct in the ESP32-Radio sketch.

ilioss, 2017-06-19:
Hi, all

Succes. I must admit it is fully my fault. As "penny wise pond foolish". I used a experiment pcb from a former project and some copper traces where still open D15 among them. This solved and tested my set-up and display gives nice data in full collor.
Now I have to find out if the power supply is 100% relaiable in combination with sufficient voltage and current, I,am using a 6 volt 400mA power supply 3.3 volts by a LD33VC 1amp, also related to the used capaciters.
I also have to check the volume setting as default mute is not the meaning.
Ed, many thanks for your support.
As far as I can check I have the same display as mentioned.
Kind regards,
Theo

So far So good I,am realy trilled by this project.

closed 2017-06-26 by Edzelf