Edzelf/ESP32-Radio #17: KY040 und IR Control?

djuscha, 2017-06-21:
Can one connect KY040 and IR remote control?

Edzelf, 2017-06-21:
I don't know what a KY040 is.  There is currently no support fo IR remote control.

djuscha, 2017-06-21:
KY040 is 360 Rotary Encoder Module Brick Sensor Development 
[https://ru.aliexpress.com/item/360-Degree-Rotary-Encoder-Module-Rotary-Code-Module-Encoder-Brick-Sensor-Development-Board-KY-040-Count/32629616836.html?spm=2114.10010208.1000016.1.36Vagi&isOrigTitle=true](url)

koskee, 2017-06-24:
Although more features is always going to be better, personally I don't think it would necessarily be worth the time invested for the IR remote. The radio already has the web interface, which is quite easily accessible with any phone or tablet - I personally have my phone within arms reach pretty much 24/7 so it's never been a problem. Not to mention, controlling over WiFi doesn't require line of sight, so I can turn mine off after I've already left and gotten into my vehicle outside,  for example. 

Also, IR requires fairly strict timings for proper communication if I recall correctly (unless the task was delegated to an external module of course, but this adds physical bulk to the system). 
The required frequency (40-50khz, don't remember exact number) might not play nice since the mcu already has a decent sized task between filling/transferring data to/from the ringbuf fer and hosting a webserver. 

It's not a bad idea, and if you could turn off the either the webserver or the IR, it should be able to keep up in that case.. It's just that I don't think too many people would be keen on adding another remote to the collection on their coffee table. The technology seems a bit dated Imo. 

That being said, if you have the time and/or skills to implement such a feature, I say go for it. 

OTOH, I had considered the rotary encoder myself, I think that would be beneficial (cut down on necessary buttons for physical control)

djuscha, 2017-06-24:
Understand :), but encoder would be super :)

Edzelf, 2017-06-28:
IR interface is implemented now.

djuscha, 2017-06-29:
:))

djuscha, 2017-06-29:
And how is it connected? PDF not yet available :(

Edzelf, 2017-06-29:
Forgot to upload it.  Fixed.

djuscha, 2017-06-30:
When I write in the config
Ir_40BF = upvolume = 2
Ir_C03F = downvolume = 2
Ir_1F6B = mute

Is in debug log

D: writeprefs ir_8A31 = uppreset = 1
D: writeprefs ir_719A = station = us1.internet-radio.com:8105
D: writeprefs ir_1F6B = mute
D: End of writing preferences
  After the save and restart is the stored is no longer there?
That was yes and if I press buttons does not happen in the log?
what am I doing wrong?

Edzelf, 2017-06-30:
Try lowercase.

djuscha, 2017-06-30:
Do you mean that?
ir_40bf = upvolume = 2
ir_c03f = downvolume = 2
ir_1f6b = mute

log
D: writeprefs ir_40bf = upvolume = 2
D: writeprefs ir_c03f = downvolume = 2
D: writeprefs ir_1f6b = mute
D: HTTP input shorter than expected
D: End of writing preferences

Is the same .. after save and restart everything away

Edzelf, 2017-06-30:
Yes, lower case.  I will check it.  Check ik you can change other prferences, for exmple "volume".

GabeHC, 2017-07-02:
No where to found any IR related code, Did you not yet committed?

Edzelf, 2017-07-02:
Uploaded last version.  IR should work now.  Improved MP3 player.

djuscha, 2017-07-02:
Now everything is wonderful. Everything works! Thank you very much
Now I'm waiting for integration of the encoder :)

Edzelf, 2017-07-02:
What would you like to with (one?) encoder?

djuscha, 2017-07-02:
Switch Radiosender, stop / play and volume control.

Edzelf, 2017-07-03:
I ordered some encoders.

djuscha, 2017-07-03:
👍 👍 👍

djuscha, 2017-07-03:
I have that made with Termite simply connected with USB with computer and buttons at the Vernbedinung actuated

 

https://www.compuphase.com/software_termite.htm

Edzelf, 2017-07-04:
Serial monitor in Arduino IDE will do as well.

Edzelf, 2017-07-28:
Rotary encoder is implemented now.  See documentation for functional description.

lonelyjack, 2017-08-01:
How can i activate ecncoder?

Edzelf, 2017-08-02:
See the documentation under "Preferences".
I used:
```
ir_pin = 35                                       # GPIO Pin number for IR receiver VS1838B
vs_cs = 5                                         # GPIO Pin number for VS1053 "CS"
vs_dcs = 16                                       # GPIO Pin number for VS1053 "DCS"
vs_dreq = 4                                       # GPIO Pin number for VS1053 "DREQ"
tft_cs = 15                                       # GPIO Pin number for TFT "CS"
tft_dc = 2                                        # GPIO Pin number for TFT "DC"
enc_clk = 25                                      # GPIO Pin number for rotary encoder "CLK"
enc_dt = 26                                       # GPIO Pin number for rotary encoder "DT"
enc_sw = 27                                       # GPIO Pin number for rotary encoder "SW"
sd_cs = 21                                        # GPIO Pin number for SD card "CS"

```

djuscha, 2017-08-10:
Rotary Encoder Works great! Thanks!!!