Edzelf/ESP32-Radio #21: flash read err, 1000

ilioss, 2017-07-03:
Hi all,

The latest version downloaded and flashes into my esp32 devkit (DOIT). So far so good.
Also installed the pushbuttons for the volume and preset.
Some where some how I run into problems which gives the follow malfunctions, it is not possible to run thru my station list up/down. not even preset is possible also the sd card is not readable no files on sd ( there are ). 
Looking on  the internet I found some issues related to the flash error "flash read err, 1000"  looks like this is most probably the problem. So I would like to know if this is the case?
Howe to reset the flash read error to (0) zero it must somehow related to flash encryption or something else.
I double checked the hardware etc.
Like to hear.
Kind regards
Theo

ps I have the pyton esptools installed in arduino hardware folder and I am using the arduino ide for flashing my esp32
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034

D: Starting ESP32-radio Version Wed, 31 May 2017 12:05:00 GMT...  Free memory 194828
D: GPIO0 is HIGH

Edzelf, 2017-07-03:
Try if flashing works without any (VS1053, TFT) connected to it.  Make sure you USB port is capable to provide enough power to the devkit, otherwise use external power supply.

ilioss, 2017-07-03:
Hello,

Just tested the esp32 devkit module out of his socket ( only connect to the pc) and flased again. Now the err is not seen.
As soon as I flash it in his socket ( radio setup ) it go,s wrong.
Is it possible it has somewhere a connection with the gpio12?
I reprogram the gpio 13 to vol+
                         gpio 14 to vol-
                         gpio 26 to stop
                         gpio 27 to resume
this works well.
Still the number list from the sd card in not showing any songs howeever there are songs on the FAT card.
ts Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034

D: Starting ESP32-radio Version Wed, 31 May 2017 12:05:00 GMT...  Free memory 194828
D: GPIO0 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO17 is HIGH
D: GPIO22 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW
D: GPIO35 is LOW
D: GPIO0 will execute uppreset = 1

My powersupply is sufficient and fitted  with a capacitor expect no problems from this site.
Keep you posted with news if any.
I like to solve this err, 1000 issue.
Regards,
Theo

ilioss, 2017-07-05:
Hi,

No answer on my post related to the test on a solo esp32.
I destil that you asume there is a h/w fault some where.
The situation is as follow at present;
1) SD card still not accessable gives no songs howeever there are songs on the sd.
2) The station shift up/down is working perfect.( I now dare to have the radio operational and this gives use a lot of joy nice clear music good quality)
3) Glitsing no longer noticed.
4) The flash read err, 1000 still exist ( if it gives no problem it is fine by me)

Ed which gpio do I have to program for station up and which for station down like the prev and next on the thml page?
The volume up and down I did program and are working real nice. 
So I like to change the stop and resume for station up/down.
Meanwhile I ordered a complete new set of parts esp32,display and vs1053 so I will build a one out of the box and hope it will afrain faults.
Like to hear ,
Greetings
Theo 
btw a complete separate SPI SD will do the job as well???

Edzelf, 2017-07-05:
I'm sorry that I can't help you with the SD.

You are free to use any (unused) GPIO pin for a function, just put them in the preferences in the config page of the webinterface.  See the examples in the documentation, like:
gpio_00 = uppreset = 1
gpio_12 = downpreset = 1
gpio_13 = upvolume = 2
gpio_14 = downvolume = 2

"upvolume = 2" means 2 volume steps up.  1 step is too small to make a difference.

Have you seen that IR-remote is also possible for these functions?

Sure you can use a separate SD module instead of the one in the TFT.

ilioss, 2017-07-05:
Hello Ed,
Thanks for your swift reply.
I will program gpio for next,
Prev. Accordingly.
Regards,
Theo

Edzelf, 2017-07-06:
The SD card software seems to be pretty unreliable.  See:
https://github.com/espressif/arduino-esp32/issues/272
I have problems with the SD interface on the TFT, a standalone SD module and SD card connected directly to the ESP32.

koskee, 2017-07-06:
Hm. Well I suppose that's a fair point. I haven't exactly tried out the SD interface, as I haven't really needed it. Wasn't aware it had issues. 
When you say that it's unreliable, what sort of symptoms do you have with it?

Edzelf, 2017-07-06:
The SD card is not recognized at start-up.  The error showed up after some weeks without errors.

ilioss, 2017-07-06:
Hi all,

Interesting this issue is open again. 
Following the link https://github.com/espressif/arduino-esp32/issues/272 it looks a rather complicated error.
My SD card refuses from very beginning.
( on my esp32 I have this flash read err, 1000 failure still.)
I specially made a SD with 6 songs mp3 on FAT.  And connected as per documentation.
As my radio is fully operational it is hard to make some more tests.
However I ordered a complete new set of esp32, display and vs1053b.
If I have received this I will also make some tests and report the results.
btw on my request for a windows7 procedure to reset this error I have upon now no reaction.
Linux is described windows not. I use windows.
To reset this failure has something to do with flash encryption
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 1 R/W (0x1) must be =0
on github:  flash read err, 1000 after deep sleep reset #117
Ongoing.
Kind regards,
Theo

koskee, 2017-07-06:
Well, by the looks of it, we've got ladyada and me-no-dev on the case. They're both fantastic developers in my humble opinion, so I'm sure there is little for anyone here to do until they're finished their bug hunt, unless of course, anyone wants to dive into the library code and give them a hand. 
If the dependent library is the issue, then I think we'll just have to wait for the fix to come from upstream and then once it is committed, compatibility can be verified. I'd be surprised if this wasn't resolved pretty quickly with those guys looking into it.

bbx10, 2017-07-07:
An SD card fix has been committed a few hours ago. I still cannot play from SD card but the SD card test program now works on my vs1053 board. I will be investigating more today.

Edzelf, 2017-07-07:
Will try then new SD linrary as soon as possible.

ilioss, 2017-08-16:
Hi All,
My SD is working.
Case closed.
Ed thanks for your support.
Kind regards.
Theo

closed 2017-08-16 by ilioss

vadpol, 2018-01-18:
Looking at all above, what should be the latest working SD library for ESP32 used now? Could you provide the link? The one provided at https://github.com/espressif/arduino-esp32/tree/master/libraries/SD seems to be working but only shortly - sometimes after powering up the displayed number of mp3 files on SD card is not correct (usually less than it is) or if it is even correct the mp3 file playback starts on command but stops in a few seconds. Then ESP32 switches back to WiFi reception or hangs up. No debug info is shown on error, so expecting a SD library's bug. Also I tried to use this one https://github.com/adafruit/arduino-esp32/commit/15bace7a1a77374915632a3d1a7b6d8003244aa6 but it doesn't compile with the current code. Many thanks!

ilioss, 2018-01-18:
Hi vadpol,

Originaly I started this issue while I had the flash read err, 1000.
This still ocure but it looks like it do not interfereer with my program. Working normal.
So for me it is no issue any longer. 
SD card problems I faced where solved by the use of a SD card sutable for the job.
Fast enough etc. I started with mickey mouse SD cards I learned my lesson.
So do not work with sheap Charly SD cards.
If you have still problems maybe it is a good ideer to open a new issue.
This one is closed in fact.
The SD lib I use with the radio is SD.h  `
SD - a slightly more friendly wrapper for sdfatlib

 This library aims to expose a subset of SD card functionality
 in the form of a higher level "wrapper" object.

 License: GNU General Public License V3
          (Because sdfatlib is licensed with this.)

 (C) Copyright 2010 SparkFun Electronics`
Hopefully it will help you.
Kind regards,
ilioss

vadpol, 2018-01-19:
Hi Theo,
Many thanks for your comment! I use a Kingston branded class 10 2Gb microSD just out of the box, therefore I don't think that this is a hardware problem. What I will do next - try to connect a standalone SD module to ESP32 instead of the one integrated in my TFT display board. If it works than it is a library problem for sure. 

BR,
Vadim

Edzelf, 2018-01-19:
You may try to lower the value in the definition:
#define SDSPEED 1000000

vadpol, 2018-01-21:
Got it working just by connecting of the standalone microSD module, no software change. It seems that the embedded in TFT display SD slot is buggy in my case.