Edzelf/ESP32-Radio #22: Feature Request: Offline Support

koskee, 2017-07-05:
Hey Ed, I just followed a link someone else posted and found this product. It seems that the current state of your hardware and even software are likely pretty much already set up to support this feature, and I think it would be really awesome to have available if a stream becomes temporarily offline/unusable due to lag. There has been a few times when I couldn't get a decent enough stream to provide smooth playback (due to my shitty internet, when there's too many people on it).  I think that having the option to record data while the unit is turned off to fill in some of those service gaps would be a nice touch. 

I'm fairly certain it would be fairly straightforward to implement, (likely just a few lines of code really...) I think you'd just have to find a way to store the incoming data into files that could be picked up and read later on. Would likely be as simple as generating a mpeg compatible file headers/padding/etc. I personally haven't looked into the specifics of the mpeg layer 3 specification before, so I'm not sure  what would be necessary exactly, but it seems that all of the necessary hardware is already in place. 

https://www.compuphase.com/dm440/dm440_en.htm

Edzelf, 2017-07-05:
This is quite a different device.  It retrieves files from the server according to a playlist and play them later.  Its is in fact much simpler than the ESP32-Radio.  I doubt if there are public accessible servers that provide this service.  If there are, I will be happy to develop a device like the DM440 on ESP32-Radio hardware.

ESP32-Radio is a real-time player.  That makes it impossible to repair gaps in the data stream caused by a slow Internet connection..

koskee, 2017-07-05:
I understand that "fixing" small gaps in the stream with other content wouldnt really be possible/would likely sound terrible lol.. 

What I was trying to say is.. 
Say for instance that when I stopped the player for the night, that for the first while it continues to stream, but instead of playing the audio, it is simply diverting the data and writing it into  a cache on the SD card (available free space). 

Essentially the ring buffer becomes huge, (up to the 8,16,or 32Gb capacity of the card or whatever is specified in configuration. Since you can already read files from the SD, and already play the live stream, then say if my cache is set to record a few hours, then it effectively caches that duration of the stream directly to the SD, before turning off. That way, if it is having trouble maintaining a decent connection the next day, it can detect a buffer underrun , and automatically switch to reading the cached data from the SD card while it waits for the network latency / throughput to improve. 

Might need some creativity to figure out at what point it decides to switch back, as I'm guessing accessing two streams at once would be either resource intensive, or just not possible. 

Perhaps that is a better explanation? 🙂

Edzelf, 2017-07-06:
I do understand it.  But connection errors will always cause missing data which cannot be repaired.
If during the night the connection is better, you might choose to buffer the stream during the night and play the buffered contents during the day.  But the good thing of radio is the real-time aspect.  Who is interested in yesterdays news?

koskee, 2017-07-08:
well, to be fair, there are a lot more uses for radio than just news.. 
when it is used for music, it becomes a lot less time sensitive, as most music is recorded in the first place.
I'm just interested in having some data from the stream cached to provide a fallback that won't require selecting a new stream, whether that selection happens automatically or not.

Going back to the use case i have previously described in a workplace setting, since everyone seems to have different music preferences, it is hard to select something that everyone will be happy with. So I am looking to propose a rotating station selection scheme. That will work great as long as the tech holds up, but I've noticed that at times of increased network latency, the stream can't be grabbed fast enough to be listenable. Having the sound cutting out every 1-10s becomes annoying VERY fast, and ultimately requires switching to a different stream. 
Now in terms of the rotating station selection, I can see some people getting pretty upset if its rightfully their turn, but they get skipped over because the network is lagging. So I just thought it would be a neat feature if the radio could preemptively acquire data when the stop button is pressed to be made available the next day as a backup data stream