Edzelf/ESP32-Radio #24: Display refresh ?  Text remains and text over text seen

ilioss, 2017-07-07:
Hello,

After using the radio for a couple of day,s without problems I noticed a " as it looks like a refresh " problem.
On the display a line of text remains and it looks like text is displayed over a old text. See picture.
Stop and resume cleared the display and cont. with a correct text.
This phenonemom happened more frequent now.
[image]

As the display is also controled by the SPI and this looks a problem which came up after some time my idee is. Is there a relation with ths SD problem?
@Ed, I just wants to let you know this phenonemom.
If I can do something please let me know hopefully I can help.
Regards,
Theo

Edzelf, 2017-07-07:
Try to run it without SD, i.e. remove the MISO line from the SD interface.

ilioss, 2017-07-07:
Hallo Ed,

Thanks for your anser.
I will disconnect the miso signal from the display.
Let you know the result.
Regards,
Theo

Edzelf, 2017-07-08:
Install the new SD library as well.

Edzelf, 2017-07-08:
I was able to reproduce this bug.  Some stations like "NPO Radio 4" transmit very long stream titles.  I add some more space to the stream title on TFT.  The space for the station name is decreased by the same amount.

ilioss, 2017-07-08:
Just busy to up=grade the sw.
Also the SD lib. but where can I find the new patched SD.lib??
Sorry for the inconvinience.
Theo

closed 2017-07-26 by ilioss