Edzelf/ESP32-Radio #27: second wi-fi?

lonelyjack, 2017-07-20:
It's not working

Edzelf, 2017-07-21:
What happens if you remove the first SSID?  Could be a problem with WiFiMulti.

derpicknicker1, 2017-10-16:
For me multiWiFi does not work. As soon as i put a second SSID into the settings, i can not connect to any network. With a single SSID in settings it works on every reboot.

Edzelf, 2017-10-16:
Looks like WiFiMulti has a problem.  Please try to make a minimal script that uses WifiMulti that shows the problem and report it to the author of the library.

derpicknicker1, 2017-10-17:
This is a problem with `WiFi.scanNetworks()` not with _WiFiMulti_. But _WiFiMulti_ uses `scanNetworks()`.
A discussion about this issue can be found at https://github.com/espressif/arduino-esp32/issues/242

As is found, this also affects the list of networks on the config webpage (it's empty for me).

A known workaround is to do a `WiFi.begin(ssid,pwd)` before starting the network scan. _ssid_ and _pwd_ can be just any string and must not represent a valid connection.