Edzelf/ESP32-Radio #29: Hick-up revision 19th July

ilioss, 2017-07-25:
Hi Ed,

Already long time not contacted. The radio play,s very well.( we only suffer from a really slow internet) as soon as we surf on the internet the radio looses some bits. Not the problem of the radio KPN lack,s "buitengebied"interest.
This evening I checked the site and sow a new revision dated 19th July.
So I installed and tested it. I must say I have the impression it became less stable. And starts up quit a bit slower. 
Now my experience,
The operational error direct noticed was the channel operation.
On my system I have gpio13 vol+ gpio14 vol- , gpio26 channel+, gpio27 channel-, and gpio25 preset6
It is not possible to operate the channel via the pushbuttons once you have channel for instance 5 it is not possible to lower the channel. It looks like the only way is up and remains there. 
After some tests I noticed that only after you pressed the "control" on the html page banner it will lower the channel.
An other thing. OTA seems not working on my system.
I tested it with the arduino basic OTA script and this is working.
So with the radio script it is not?
Not a big deal but I would like to feel the luxure to program the radio over the air while the radio and the esp32 is build in in a nice case.( no external possibility for the usb connector.)
The time indicater is a nice feature. Howe ever it looks like it is not always shown on the display.
The lower indicater banner showed preset 8 but I have only 0 till 7 programmed?? This is 8 but still.

The serial monitor gives the following: after I pressed the gpio27 for request lower channel
( it has worked perfectly )

push gpio27   I did.
Serial monitor showed,
GPIO27  is now low , executed down preset  =-1
command downpreset with parameter -1
preset is now 3
Stop request
Stopping client

( is it necessary to note  -1  for downpreset?  Volume is not like that it is just 2 for both up or down?
Like to hear.
ps. The radio is in our living-room operational. If the wife likes it it is ok. But if it has some unlogical hick-ups she will complain and I have to go to the shop I,am afraid.
We will see.
Kind regards,
Theo

ilioss, 2017-08-12:
Hi Ed,

Sorry no respond on my measage above. Still things are going on.
Hopefully I do not bther you with my issue's?
Due to bad weather I started to assamble my 2nd radio. All parts recieved.
Initially test fails. No display.
So what to do?
Again I checked all connections and so forth. Looks ok to me.
Loaded a previous version and now it worked all.
Not surprisingly while the 28-July version didn.t work on my system as well.
I installed also the rotary switch om 3.3 v ( I keep the radio at 3.3Volt true out in order to avoid mistakes somewhere some how. Look to me logic.)
The switch didn,t function on my new radio with the sw revision of 28-07.
So, I have my first radio running on the previous sw version not the latest.
My new radio loaded with the 28-07 version.
Status:
see serial print out.
First what seems to me looks strange is direct on top of the list I see that parameters are with pin set to xxx -1???  What can course that?
My programmed buttons for vol+ and vol- and preset  station are working.
What still not will work is the SD card and the display:
The vs1053 woorks pretty well.

Ed my new radio I would like to use as MP3 player as well to be able to insert the SD card I must rotate the display. No problem parameter from 3 to 1 looks logical to me and so I did and with succes the display is 180 dgr. rotated. But the offset from the originated point seems to be wrong the text start not exact at the beginning of the display but some where 30 or so pixels from the left?? 
Any ideer where to adjust this one?
I already had a look in the FTFXXX.setting.h file and the lib but no succes.

Hope to have some responce.
Kind regards,
Theo

ps. I will have a look in the other SD card issue

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034

D: Starting ESP32-radio Version Fri, 28 Jul 2017 12:05:00 GMT...  Free memory 194940
D: ir_pin pin set to -1
D: enc_clk pin set to -1
D: enc_dt pin set to -1
D: enc_sw pin set to -1
D: tft_cs pin set to -1
D: tft_dc pin set to -1
D: sd_cs pin set to -1
D: vs_cs pin set to -1
D: vs_dcs pin set to -1
D: vs_dreq pin set to -1
D: GPIO0 is HIGH
D: GPIO1 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO3 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO12 will execute enc_sw
D: GPIO13 will execute upvolume = 2
D: GPIO14 will execute downvolume = 2
D: GPIO25 will execute preset = 6
D: GPIO26 will execute uppreset = 1
D: GPIO27 will execute downpreset = 1
D: GPIO32 will execute enc_dt
D: GPIO33 will execute enc_clk
D: SD Card Mount Failed!
D: Added Leeuwarderweg29wifi to list of networks
D: Added Leeuwarderweg29 to list of networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 2
D:  1 - Leeuwarderweg29           Signal: -46 dBm, Encryption WPA_WPA2_PSK, Acceptable
D:  2 - Leeuwarderweg29wifi       Signal: -74 dBm, Encryption WPA_PSK, Acceptable
D: End of list
D: Command: mqttbroker with parameter none
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: mqttpasswd with parameter ******
D: Command: volume with parameter 72
D: Command: toneha with parameter 4
D: Command: tonehf with parameter 7
D: Command: tonela with parameter 11
D: Command: tonelf with parameter 13
D: Command: preset with parameter 6
D: Command: clk_server with parameter pool.ntp.org
D: Command: clk_offset with parameter 1
D: Command: clk_dst with parameter 1
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Connected to Leeuwarderweg29
D: IP = 192.168.2.15
D: Start server for commands
D: Rotary encoder is enabled
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (6/0) from icecast.omroep.nl/radio6-bb-mp3
D: Connect to new host icecast.omroep.nl/radio6-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio6-bb-mp3
D: Connected to server
D: Switch to HEADER
D: Server: Icecast 2.4.2
D: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Cache-Control: no-cache
D: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Pragma: no-cache
D: icy-br:192
D: ice-audio-info: samplerate=48000;channels=2;bitrate=192
D: icy-br:192
D: icy-genre:Soul & Jazz
D: icy-metadata:1
D: icy-name:NPO Soul&Jazz
D: icy-pub:0
D: icy-url:http://www.nposoulenjazz.nl/
D: icy-metaint:16000
D: Switch to DATA, bitrate is 192, metaint is 16000
D: First chunk:
D: D3 86 1E 22 61 4B 6A 7A
D: 65 E5 3A 55 9D 77 38 4E
D: E1 2D 59 FB 2F A8 0D A5
D: 8E 18 E8 8A 0D A1 BF C9
D: Metadata block 64 bytes
D: Streamtitle found, 48 bytes
D: StreamTitle='NPO Soul & Jazz - Non-stop - VPRO';

Edzelf, 2017-08-12:
In the new version you have to provide the pinnumber that you use for TFT, VS105, etc.  in the preverences.  See the documentation and the new default settings in the web interface.  I will check the 180 degrees display later.

Edzelf, 2017-08-12:
In the new version of the software, you have to provide the pinnumbers for TFT, VS1053, ... in the preferences like this:
```
vs_cs = 5                                            # GPIO Pin number for VS1053 "CS"
vs_dcs = 16                                          # GPIO Pin number for VS1053 "DCS"
vs_dreq = 4                                          # GPIO Pin number for VS1053 "DREQ"
```
Read the documentation.  I will look into the 180 degrees rotation of the TFT.

Edzelf, 2017-08-12:
180 degrees works perfect on my hardware.  I quess you are using an unpatched version of the TFT library.  Read the documentation under "TFT library patches".

ilioss, 2017-08-12:
Hi Ed,

Thanks for your swift reply.
Yes, I will check the preference list.
In the mean time I was able to have some SD card function. It appeared that I get the music files from the sd card. How ever no song names. In the html page from inderx I get the songnames by pressing the status. 
So it looks like there is some progres.
I keep you informed.
Kind regards,
Theo

Thanks again very preciated.

Edzelf, 2017-08-12:
The only info for mp3-files on SD is the filename.

ilioss, 2017-08-12:
Hi Ed,

Thanks for your answer.
I patched up the preferences and the lines at the start of the list ( serial printout ) are now corresponding the gpio. Thats fixed.
I get the radio working to start it I had to push several times on the switch from the rotary encoder.
The SD is also giving his secrets to me the songs name is given and also the number of songs.
I made really nice progress. 
Question the SD was resently under revision by ladyada where is the new sd lib. to find. Also the patch for the TFT?
Like to hear.
Kind regards

Theo

Edzelf, 2017-08-13:
Just copy the last version of the ESP32 Arduino Core libraries to your PC.  The patch for the TFT is decribed in the documentation of ESP32-Radio.

ilioss, 2017-08-16:
Hi All,
Most issues solved.
Due to support from Ed.
So case closed.
Kind regards.
Theo

closed 2017-08-16 by ilioss