Edzelf/ESP32-Radio #31: 5 Volts in my Laptop

arlaor, 2017-08-05:
Hi @Edzelf 

More than a problem in your project, it is a general doubt for the ESP32 modules
I have a question about the voltage supported in my esp32.
When I connect the module to my pc laptop, it receives 5V input sent by the usb port, but recommends using maximum 3.9V.
Should I use some type of connection that reduces the voltage of my laptop to connect my esp32 module to my pc laptop?
I use my mobile phone connector which has a 5V output, to power my esp32. Can i damage my esp32 module?
The heat generated by the ESP32 module to be connected to my laptop is high and therefore I am worried to generate any damage to this

Another question, Is there any option to increase the number of stations that ESP32-Radio can support?

thanks

Edzelf, 2017-08-05:
Most ESP32 modules have a build-in voltage regulator for the 3.3 volt.  Do not try to connect an ESP32 to 5 volt directly: it will die within seconds....
Currently the number of presets (not the number of stations) is limited to 90.  It is relatively easy to extend the number of presets.  How much presets do you want?

arlaor, 2017-08-05:
Hi @Edzelf 
I tried to enter 150 stations, but the only one shows me 49 stations

To program the esp32 module in my computer, connected to the USB port that normally has a 5V output, how should I do it?

Edzelf, 2017-08-05:
Tell me what module you have.  If it has a micro USB input, it wiil probably have a voltage regulator as well.

arlaor, 2017-08-05:
A wemos lolin32 and MH-ET live D1 mini esp32

Edzelf, 2017-08-06:
Both modules have an on-board voltage regulator and can be connected to 5 volt through the micro USB interface.
I will check the maximum number of presets is and maybe increase it.  The amount of preference lines is limited by the size of the NVS partition in flash, normally 24 kB.  Maybe there is a way to extend this partition.
It will help if you mail me the output of the debug logging.

arlaor, 2017-08-06:
I'm sorry, I do not know how to debug

this is my stations list
```javascript
preset_01=prem2.radiotunes.com:80/hit00s?203bc6099569e11872e584dc #1 hit00s RT
preset_02=prem2.rockradio.com:80/00srock_aac?203bc6099569e11872e584dc #2 00srock RR
preset_03=streaming.hotmix-radio.net/hotmixradio-80-128.mp3 #3 80 HotMixRadio (80's)
preset_04=prem2.rockradio.com:80/alternative80s_aac?203bc6099569e11872e584dc #4 alternative80s RR
preset_05=prem2.radiotunes.com:80/the80s?203bc6099569e11872e584dc #5 the80s RT
preset_06=prem2.rockradio.com:80/80srock_aac?203bc6099569e11872e584dc #6 80srock RR
preset_07=prem2.radiotunes.com:80/80srock?203bc6099569e11872e584dc #7 80srock RT
preset_08=prem2.radiotunes.com:80/hit90s?203bc6099569e11872e584dc #8 hit90s RT
preset_09=prem2.rockradio.com:80/90srock_aac?203bc6099569e11872e584dc #9 90srock RR
preset_10=prem2.radiotunes.com:80/altrock?203bc6099569e11872e584dc #10 altrock RT
preset_11=prem2.rockradio.com:80/alternativerock_aac?203bc6099569e11872e584dc #11 alternativerock RR
preset_12=live3.rcnmundo.com/antena2.aac #12 Antena 2
preset_13=14073.live.streamtheworld.com:3690/MUSICABLU.mp3 #13 Blu Musica (Pop)
preset_14=18123.live.streamtheworld.com/BLURADIO.mp3 #14 Blu Radio
preset_15=20423.live.streamtheworld.com/CADENADIAL.mp3 #15 Cadena Dial (Pop Latino)
preset_16=18553.live.streamtheworld.com/CARACOL_RADIOAAC.aac #16 Caracol Radio Bogota
preset_17=18243.live.streamtheworld.com:3690/CARACOL_RADIO_SC #17 Caracol Radio
preset_18=18703.live.streamtheworld.com/CR_CALIAAC.aac #18 Caracol Radio Cali
preset_19=prem2.di.fm:80/classiceurodance?203bc6099569e11872e584dc #19 classiceurodance DI
preset_20=prem2.di.fm:80/classiceurodisco?203bc6099569e11872e584dc #20 classiceurodisco DI
preset_21=prem2.rockradio.com:80/classichardrock_aac?203bc6099569e11872e584dc #21 classichardrock RR
preset_22=wms-15.streamsrus.com:12310/; #22 Classic Rock Hits AddictedToRadio.com (80's)
preset_23=prem2.di.fm:80/clubdubstep?203bc6099569e11872e584dc #23 clubdubstep DI
preset_24=prem2.radiotunes.com:80/christian?203bc6099569e11872e584dc #24 christian RT
preset_25=194.58.97.117:8000/coreradio #25 CORE RADIO! (Metal)
preset_26=prem2.radiotunes.com:80/country?203bc6099569e11872e584dc #26 country RT
preset_27=real2.streaming-co.com:8524/;stream.mp3 #27 Cristal Stereo
preset_28=prem2.radiotunes.com:80/dancehits?203bc6099569e11872e584dc #28 dancehits RT
preset_29=streams.deltaradio.de/100/mp3-128/ #29 delta radio - Event 100
preset_30=streams.deltaradio.de/101/mp3-128/ #30 delta radio - Event 101
preset_31=streams.deltaradio.de/delta-alternative/mp3-128/ #31 delta radio ALTERNATIVE
preset_32=streams.deltaradio.de/delta-poprock/mp3-128/ #32 delta radio Der beste RockPop reloaded
preset_33=streams.deltaradio.de/delta-grunge/mp3-128/ #33 delta radio GRUNGE
preset_34=streams.deltaradio.de/delta-foehnfrisur/mp3-128/ #34 delta radio Hard Rock & Heavy Metal
preset_35=streams.deltaradio.de/delta-indie/mp3-128/ #35 delta radio INDIE
preset_36=streams.deltaradio.de/delta-live/mp3-128/ #36 delta radio LIVE
preset_37=streams.deltaradio.de/delta-sommer/mp3-128/ #37 delta radio Sommer
preset_38=streams.deltaradio.de/delta-unplugged/mp3-128/ #38 delta radio UNPLUGGED
preset_39=prem2.di.fm:80/djmixes?203bc6099569e11872e584dc #39 djmixes DI
preset_40=prem2.radiotunes.com:80/edmfest?203bc6099569e11872e584dc #40 edmfest RT
preset_41=prem2.di.fm:80/epictrance?203bc6099569e11872e584dc #41 epictrance DI
preset_42=prem2.di.fm:80/eurodance?203bc6099569e11872e584dc #42 eurodance DI
preset_43=prem2.radiotunes.com:80/rteurodance?203bc6099569e11872e584dc #43 rteurodance RT
preset_44=mp3.ffh.de/ffhchannels/hqbrandneu.aac/ffhchannels/hqbrandneu.aac #44 FFH BRANDNEU
preset_45=mp3.ffh.de/ffhchannels/hq80er.aac/ffhchannels/hq80er.aac #45 FFH DIE 80ER
preset_46=mp3.ffh.de/ffhchannels/hqeurodance.aac/ffhchannels/hqeurodance.aac #46 FFH EURODANCE
preset_47=mp3.ffh.de/ffhchannels/hqtop40.aac/ffhchannels/hqtop40.aac #47 FFH iTUNES TOP 40
preset_48=mp3.ffh.de/ffhchannels/hq90er.aac/ffhchannels/hq90er.aac #48 FFH JUST 90s
preset_49=mp3.ffh.de/ffhchannels/hqvoting.aac/ffhchannels/hqvoting.aac #49 FFH LEIDER GEIL
preset_50=mp3.ffh.de/ffhchannels/hqlounge.aac/ffhchannels/hqlounge.aac #50 FFH LOUNGE
preset_51=mp3.ffh.de/ffhchannels/hqparty.aac/ffhchannels/hqparty.aac #51 FFH PARTY
preset_52=mp3.ffh.de/ffhchannels/hqrock.aac/ffhchannels/hqrock.aac #52 FFH ROCK
preset_53=mp3.ffh.de/ffhchannels/hqschlager.aac/ffhchannels/hqschlager.aac #53 FFH SCHLAGERKULT
preset_54=mp3.ffh.de/ffhchannels/hqsoundtrack.aac/ffhchannels/hqsoundtrack.aac #54 FFH SOUNDTRACK
preset_55=mp3.ffh.de/ffhchannels/hqxmas.aac/ffhchannels/hqxmas.aac #55 FFH WEIHNACHTSRADIO
preset_56=prem2.rockradio.com:80/grunge_aac?203bc6099569e11872e584dc #56 grunge RR
preset_57=prem2.di.fm:80/handsup?203bc6099569e11872e584dc #57 handsup DI
preset_58=prem2.di.fm:80/harddance?203bc6099569e11872e584dc #58 harddance DI
preset_59=prem2.radiotunes.com:80/hardrock?203bc6099569e11872e584dc #59 hardrock RT
preset_60=prem2.rockradio.com:80/hardrock_aac?203bc6099569e11872e584dc #60 hardrock RR
preset_61=prem2.di.fm:80/hardtechno?203bc6099569e11872e584dc #61 hardtechno DI
preset_62=mp3.ffh.de/radioffh/hqlivestream.aac/radioffh/hqlivestream.aac #62 HIT RADIO FFH
preset_63=streaming.hotmixradio.fm/hotmixradio-hits-128.mp3 #63 Hits HotmixRadio (Pop)
preset_64=stream01.iloveradio.de/iloveradio2.mp3 #64 I Love 2 Dance (iloveradio.de)
preset_65=stream01.iloveradio.de/iloveradio4.mp3 #65 I Love Bass (iloveradio.de)
[... cut ...]
preset_144=19253.live.streamtheworld.com/TROPICANAAAC.aac #144 Tropicana (Tropical)
preset_145=16693.live.streamtheworld.com/TR_CALIAAC.aac #145 Tropicana Cali (Tropical)
preset_146=19273.live.streamtheworld.com/TR_NEIVAAAC.aac #146 Tropicana Neiva (Tropical)
preset_147=prem2.di.fm:80/umfradio?203bc6099569e11872e584dc #147 umfradio DI
preset_148=live3.rcnmundo.com/vallenatoventiao.mp3 #148 Vallenato Ventia'o RCN
preset_149=prem2.di.fm:80/vocaltrance?203bc6099569e11872e584dc #149 vocaltrance DI
preset_150=19173.live.streamtheworld.com/WRADIOAAC.aac #150 W Radio
```

arlaor, 2017-08-06:
Make a modification to the texts, reducing some words and increase to 69 stations

Edzelf, 2017-08-07:
The space for preferences (like the presets) is limited to 20480 bytes.  You may change the partitioning of your ESP32 by editing the file in .../hardware /espressif/esp32/tools/partitions/default.csv.  It now contains:

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 1M,
app1,     app,  ota_1,   0x110000,1M,
spiffs,   data, spiffs,  0x210000,0x1F0000,

```
You may use the space of the "spiffs" partition to increase the size of the "nvs" partition.  Before you change anything in the partition table: read the ESP32 documentation on this subject carefully.
Note that changing the default.cvs file will affect all your ESP32 projects.  As far as I know, there is now way to have separate partition tables per project.

For debug: In the IDE, just set the baudrate of your com port to 115200 and start the Serial Monitor.