Edzelf/ESP32-Radio #32: SD Card Mount  Failed!

djuscha, 2017-08-10:
Get the error message  SD Card Mount  Failed!
And the mp3 will not play. What am I doing wrong?

Edzelf, 2017-08-10:
Hard to say.  Most likely:
1 - wiring fault.
2 - Pins for SD card not in preferences.  Pins should be logged in debug logging like "D: Command: sd_cs with parameter xx".
3 - no FAT filesystem on SD card.  Every other filesystem will fail.
4 - no SD card in slot.

djuscha, 2017-08-10:
1. I have checked, is correct
2. How can I do that?
defaultprefs stands   sd_cs = 21   # GPIO Pin number for SD card "CS"
3. Have checked. Is FAT
4. SD card is in it ;)
Have tried two different card readers

Edzelf, 2017-08-10:
To assign pin number 21 to sd_cs, add this line to the preferences in the config page of the webinterface:
sd_cs = 21.
Restart ESP32 and check the logging for "D: Command: sd_cs with parameter 21"

djuscha, 2017-08-10:
That is my log.
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034

D: Starting ESP32-radio Version Fri, 28 Jul 2017 12:05:00 GMT...  Free memory 194988
D: ir_pin pin set to 35
D: enc_clk pin set to 25
D: enc_dt pin set to 26
D: enc_sw pin set to 27
D: tft_cs pin set to 15
D: tft_dc pin set to 2
D: sd_cs pin set to 21
D: vs_cs pin set to 5
D: vs_dcs pin set to 16
D: vs_dreq pin set to 4
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO21 is LOW, probably no PULL-UP
D: GPIO22 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is HIGH
D: GPIO0 will execute uppreset = 1
D: GPIO12 will execute upvolume = 2
D: GPIO13 will execute downvolume = 2
D: GPIO14 will execute stop
D: GPIO17 will execute resume
D: GPIO21 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
D: Start TFT
D: SD Card Mount Failed!
D: Added FRITZ!Box Fon WLAN 7390 to list of networks
D: Added SSID2 to list of networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 3
D:  1 - FRITZ!Box Fon WLAN 7390   Signal: -44 dBm, Encryption WPA_WPA2_PSK, Acceptable
D:  2 - WLAN-78EF46               Signal: -75 dBm, Encryption WPA2_PSK, 
D:  3 - FRITZ!Box Fon WLAN 7390   Signal: -75 dBm, Encryption WPA_WPA2_PSK, Acceptable
D: End of list
D: Command: mqttbroker with parameter none
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: mqttpasswd with parameter ******
D: Command: volume with parameter 84
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: preset with parameter 0
D: Command: clk_server with parameter pool.ntp.org
D: Command: clk_offset with parameter 1
D: Command: clk_dst with parameter 1
D: Command: ir_pin with parameter 35
D: Command: vs_cs with parameter 5
D: Command: vs_dcs with parameter 16
D: Command: vs_dreq with parameter 4
D: Command: tft_cs with parameter 15
D: Command: tft_dc with parameter 2
D: Command: enc_clk with parameter 25
D: Command: enc_dt with parameter 26
D: Command: enc_sw with parameter 27
D: Command: sd_cs with parameter 21
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Connected to FRITZ!Box Fon WLAN 7390
D: IP = 192.168.178.70
D: Start server for commands
D: Rotary encoder is enabled
D: Sync TOD, new value is 21:46:19
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (0/0) from retroserver.streamr.ru:8043/retro128?.mp3
D: Connect to new host retroserver.streamr.ru:8043/retro128?.mp3
D: Connect to retroserver.streamr.ru on port 8043, extension /retro128?.mp3
D: Connected to server
D: Switch to HEADER
D: Accept-Ranges: none
D: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: icy-br:128
D: ice-audio-info: channels=2;samplerate=44100;bitrate=128
D: icy-br:128
D: icy-description:Stream transcoder based on liquidsoap
D: icy-genre:various
D: icy-name:retro
D: icy-pub:1
D: icy-url:http://95.213.143.165:8010/retro320.mp3
D: Server: Icecast 2.3.3-kh11
D: Cache-Control: no-cache, no-store
D: Pragma: no-cache
D: Access-Control-Allow-Origin: *
D: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Connection: close
D: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: icy-metaint:8192
D: Switch to DATA, bitrate is 128, metaint is 8192
D: First chunk:
D: FF FB 92 64 E4 80 03 4A
D: 42 5D FB 0C 2A 60 3F 05
D: 0C 4D 60 A9 6F 8F 8D 79
D: 6D EC A4 AF C1 2C 1E AD
D: Metadata block 64 bytes
D: Streamtitle found, 60 bytes
D: StreamTitle='Viktor Saltykov/Elektroklub - Koni V Yablokah';

Edzelf, 2017-08-10:
As you can see, GPIO21 is pulled LOW.  So there is probably a short to ground.  Check the wiring again.

djuscha, 2017-08-11:
Got card reader again connected again, still the same error. I find no mistake in my contempt. Have tried with the other card reader, also the same. Everything else works. Only card reader does not. Perhaps this is due to the card reader. Have these both tried https://ru.aliexpress.com/item/3-3V-5V-Micro-SD-TF-Dual-Card-Reader-Module-SPI-SDIO-Dual-Mode-Storage-Board/32817214968. html? spm = & a2g0v.10010108.1000016.1.34f197c2JEr2vo isOrigTitle = true
and
https://ru.aliexpress.com/item/1pcs-Micro-SD-card-mini-TF-card-reader-module-SPI-interfaces-with-level-converter-chip-for/32583289463.html?spm= a2g0v.10010108.1000016.1.18830581SL0Syq & isOrigTitle = true
:(

ilioss, 2017-09-17:
@djuscha 
Hi, as I suffer also a long time with SD card faults finaly it turned out that the used SD ( micro ) was causing the problem.
I also get the meassage SD card not mounted and the err. flash read err, 1000.
Just after I installed a micro SD card with HC notation and 10 class speed my problems where solved.
https://www.google.nl/url?sa=t&rct=j&q=&esrc=s&source=web&cd=6&cad=rja&uact=8&ved=0ahUKEwjpooLUgazWAhUrBcAKHV1jBt4QFgiWAjAF&url=http%3A%2F%2Fwww.makeuseof.com%2Ftag%2F5-mistakes-avoid-buying-next-microsd-card%2F&usg=AFQjCNGyu3jqsS3p6cDxcPhaMq2XZedQjg
look into this link for SD card do,s.
Succes,
Theo
ps. Power supply quality can also be an issue!!!!!

fetze82, 2018-08-15:
Hey, i have an SD Card Problem as well.
I cant find files at my file. If i check my setup with an SD card demo - i can read the mp3 file names. But with the ESp32 Radio, no mp3 files will be found.

i cant find an error at the debug information. Maybe somebody has an idea.
Greatings
Chris

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 06 Aug 2018 12:12:32 GMT.  Free memory 171672
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 46 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is HIGH
D: GPIO39 is LOW, probably no PULL-UP
[... 87 lines cut ...]
D: Connect to new host bob.hoerradar.de/aac-radiobob?sABC=5o745317%230%23spr822816552r64sn3n61ss16p16q465%23zrqvncynlre&amsparams=playerid:mediaplayer;s
D: Connect to bob.hoerradar.de on port 80, extension /aac-radiobob?sABC=5o745317%230%23spr822816552r64sn3n61ss16p16q465%23zrqvncynlre&amsparams=playerid
D: Connected to server
D: Switch to HEADER
D: Headerline: icy-br: 64
D: Headerline: icy-pub: 1
D: Headerline: icy-description: RADIO BOB! BOBs Livestream
D: Headerline: icy-audio-info: channels=2;samplerate=44100;bitrate=64
D: Headerline: icy-url: http://www.radiobob.de/
D: Headerline: Instance-id: c03c6cec793576f150c977a82c5a4f51
D: Headerline: Cache-Control: no-cache
D: Headerline: Server: sec-bob.hoerradar.de 8.0.13
D: Headerline: icy-genre: Rock
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint: 16000
D: Headerline: Pragma: no-cache
D: Headerline: icy-name: RADIO BOB! BOBs Livestream
D: Headerline: Connection: close
D: Headerline: Content-Type: audio/aacp
D: audio/aacp seen.
D: Switch to DATA, bitrate is 64, metaint is 16000
D: Duration mp3loop 55
D: Metadata block 48 bytes
D: Streamtitle found, 37 bytes
D: StreamTitle='xxx';

Edzelf, 2018-08-15:
In debug log:
`D: pin_sd_cs set to -1`
So the pin number for SD chip select is not defined.  Add a line for it in the config page of the web interface.

hshafiq99, 2019-10-18:
I am also having some issue. @Edzelf 
Following are the debug logs:
.....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.8.103, MASK: 255.255.255.0, GW: 192.168.8.1
.
WiFi connected.
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
Card Mount Failed

MHz000, 2021-02-09:
I'm on Version Thu, 21 Jan 2021 
Happy to get things running but got error msg
"SD Card Mout Failed"
all SD-pins connected _separatly_ 
my init-file looks like
  preferences.putString ( "pin_tft_cs",  "15                               # GPIO Pin number for TFT CS" ) ;
  preferences.putString ( "pin_tft_dc",  "2                                # GPIO Pin number for TFT DC" ) ;
  preferences.putString ( "pin_sd_cs",   "21                             # GPIO Pin number for SD card CS" );
However I don't like to use SD-card not at all. How to get ride of the check?
Thank you

Edzelf, 2021-02-09:
Just comment out the definition of SDCARD in the sketch.  By the way, the contents of the ini file do not reflect the real sutuation.  It is just meant to get you starting.  As soon as the webinterface works, you should configure the pins in the config page.

MHz000, 2021-02-09:
**success!** No mor SD card mount failed msg

Sorry for asking next question: now I got a lot of 
VS1053 SPI error(s) and
This is not a VS1053, but a VS1003 instead!
I bought this from AliExpress :-(
does this message "game over" to me?
Any suggestion where to by?

Thank-you for advice and patience

Edzelf, 2021-02-09:
VS1003 does not have all the capabilities, but it seems to work with some modification.  I can not test it, because I only have the VS1053.  Search for this in other issues on this repositry.

MHz000, 2021-02-11:
Just received a VS1053. Checked the wiring twice but still get error messages concerning the VS1053

D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 SPI error. SB:0BB8 R1:0BE0 R2:0BB8
D: VS1053 SPI error. SB:2B5C R1:2B20 R2:2B20
D: VS1053 SPI error. SB:43F8 R1:43F8 R2:43E0
D: VS1053 SPI error. SB:4FB0 R1:4FC0 R2:4FC0
D: VS1053 SPI error. SB:5334 R1:5340 R2:5334
D: VS1053 SPI error. SB:6BD0 R1:6BD0 R2:6B80
D: VS1053 SPI error. SB:71AC R1:71AC R2:71E0
D: VS1053 SPI error. SB:79E0 R1:79E0 R2:79C0
D: VS1053 SPI error. SB:7B0C R1:7B0C R2:7B20
D: VS1053 SPI error. SB:8340 R1:8340 R2:8300
D: VS1053 SPI error. SB:846C R1:8600 R2:8600
D: VS1053 SPI error. SB:9150 R1:8000 R2:8000
D: VS1053 SPI error. SB:9AB0 R1:BC00 R2:9AB0
D: VS1053 SPI error. SB:9D08 R1:9D20 R2:9D08
D: VS1053 SPI error. SB:9E34 R1:B400 R2:B400
D: VS1053 SPI error. SB:A9EC R1:A9EC R2:A9E0
D: VS1053 SPI error. SB:AC44 R1:AC44 R2:E000
D: VS1053 SPI error. SB:B0F4 R1:CC00 R2:B0F4
D: VS1053 SPI error. SB:B220 R1:D400 R2:B220
D: VS1053 SPI error. SB:BDD8 R1:BDA0 R2:BDA0
D: Start CH376
Recompile, reload or restart the ESP32 dosn't help. 
Setup: ESP32 on breadboard, TFT (diplaying start message) and VS1053 via Dupond wire cable 10cm.
Any suggestion?
Thank-you