Edzelf/ESP32-Radio #33: ESP32 Radio + Oled 0.98" scrolling + U8g2lib

blotfi, 2017-08-11:
Dear All
I want to share my modifications, that displays, IP , web radio name and title on a small OLED screen 
[#Esp32_radio_oled.zip](https://github.com/Edzelf/ESP32-Radio/files/1218099/Esp32_radio_oled.zip)

Edzelf, 2018-04-25:
OLED is now supported.

fbsln, 2018-06-04:
Ich habe das 0,98 OLED Display ("1306") 

SCL an GPIO 15 und SDA an GPIO 4 

DREQ habe ich ähnlich wie bei dem HetTec Beispiel von GPIO 4 auf 2 verlegt. 

Radio läuft nur Display bleibt dunkel. 

Wer erkennt meinen Fehler. Danke für Unterstützung.

Viele Grüße und danke für die tolle Arbeit.

blotfi, 2018-06-04:
try my post of 11/08/2017
it works

Edzelf, 2018-06-05:
You have to be sure that the pins for SDA and SCL are in the preferences. You may do so in the config page of the webinterface. Restart the radio and look at the debug output with the serial monitor. Check if the pins match the ones you used for the connection with the display.

h1aji, 2020-04-03:
@blotfi is there any chance to update [SSD1306.h](https://github.com/Edzelf/ESP32-Radio/blob/master/Esp32_radio/SSD1306.h) ?

blotfi, 2020-04-03:
I only use U8g2lib ( I didn't develop the lib) so you can update it via Arduino lib manager
I didn't worked on this project since my last post
Best regards