Edzelf/ESP32-Radio #35: Bluetooth

blotfi, 2017-08-12:
Hi
Is it possible instead of 1053 to use Bluetooth BLE module of ESP32 to send the mp3 to a BT enabled device (speaker) which can play it?
thanks

Edzelf, 2017-08-12:
Not with this version of the software.

blotfi, 2017-08-12:
yes but is it possible ?
is ESP32 enough to trnasmit and get rid of the 1053

Edzelf, 2017-08-13:
Yes it possible, but it is a lot of work.

koskee, 2017-08-14:
Also I'm not aware of any Bluetooth receivers that also integrate mp3/aac/etc decoding, so unless there is some hardware out there that I'm unaware of (entirely possible), you would either need to implement the decoding in software (which is also possible, but will only slow everything down in the end)  or you'd still need the 1053.
The reason the 1053 is nice is because it offloads a good chunk of the heavyweight processing to a secondary, designed for that exact purpose, processor. Purpose built processors like that also have the advantage of being much more efficient carrying out the dsp workload, where the esp32, fast as it is, may lack some of the core instructions for efficient decoding (i'm not sure). 

Sure,  the cost of the 1053 modules is slightly higher than *some  other sensors, shields and the like, but one must also consider that using audio compression algorithms like that often requires a licence because the standards are normally patented, and I'd bet that is where the majority of the extra cost comes from.. (although I think I heard somewhere that the mp3 patent/copyright protection period is now over? )

NonaSuomy, 2017-08-25:
Just a note, Sprite_tm (One of the guys behind expressif) had this to say about the ESP32,

```
Sprite_tm says:
August 22, 2017 at 11:37 pm
Kinda meh that he still uses a hardware mp3 decoder. The ESP8266 already was pretty capable of decoding mp3s at 80MHz (the mp3 decoder was limited to mono because of memory usage); the ESP32 can do stereo MP3s at any rate while still idling its CPUs most of the time.
```

I'm sure it would take a lot of hardcore hacking to get it working, also it wouldn't support the other codecs like AAC that a lot of the streams work with now, unless someone did work on that as well to decode on the fly with the ESP32.

Edzelf, 2017-08-26:
The facility for OGG/AAC justifies the use of the VS1053 module more than enough, not to mention the output connector for the earphones/speakers....

bickyb, 2017-08-28:
Here's an article about the end of the licensing associated with the MP3 codec. 
https://www.theatlantic.com/news/archive/2017/05/the-end-of-the-mp3/526716/

bickyb, 2017-08-28:
Here's another project like Ed's ESP-32 project.  It does not use the Arduino IDE.
Unfortunately, Bluetooth is not available yet for the Arduino IDE.  I agree it would be awesome to be able to use the built in bluetooth capability of the ESP32 to output to a bluetooth speaker.  

https://github.com/MrBuddyCasino/ESP32_MP3_Decoder

NonaSuomy, 2017-08-28:
Nice @bickyb it seems it has some beta support for decoding AAC and it also has some method there to attach to a Bluetooth speaker.

There's some support for bluetooth & BLE in ESP32 Arduino most involves a lot of digging in the issues tracker at github and the forums, reusing includes from IDF https://github.com/espressif/arduino-esp32/files/1231768/A2DP_Test.zip

ESP32Arduino is basically just a wrapper on top of IDF so basically anything new you do with IDF you could essentially port to Arduino, most of the work just went into compatibility with Arduino Existing Libraries.

vupqtran, 2018-03-01:
@NonaSuomy Can you tell me how to use your A2DP_Test.zip?