Edzelf/ESP32-Radio #36: correct wifi acces without wifi connection

student1704, 2017-08-13:
Thank you for the big job, Ed.

How can I easily correct the wifi access data without loading new init file and without wifi connection.
 Background is: I want to prepare a web radio for a friend of mine.  How can he connect the  radio to his wifi environmet without reprogramming by Arduino SDK?
 BR

Edzelf, 2017-08-13:
If you know his SSID and password, you may add an extra "wifi-xx" line in the configuration.
Otherwise: When no WiFi connection can be made, the ESP32 will act as its own AP and your friend can connect to the SSID "ESP32Radio" and browse to 192.168.4.1 and add his network to the configuration.  See also the configuration part in the documentation.

student1704, 2017-08-15:
Ok, got it.
Thanks and BR

closed 2017-08-15 by student1704