Edzelf/ESP32-Radio #37: Controls ALL together?

ilioss, 2017-08-14:
Hi Ed,

Still impressed by the ESP32 radio.
I have progres to get the 28-07 version working. Indeed placing he correct preferences. SD is working now.
As I liked the ideer to have the IR, Encoder and free programmable buttons all available for control the radio.
So I had the gpio 13,14,26,27 and 25 working for volume station control working as a charm.
On the new radio I installed also the rotary switch. 
My question is is it posible to have these working all or is control only possible or buttons or ir, or rotary switch?
 The ir-remote control is on order in China this will take some time.

Like to hear.
Kind regards,
Theo

ps. Is it correct you removed the TinyXML lib.? If so remove also from page two of the pdf.

Edzelf, 2017-08-14:
You can have all options in parallel :serial, mqqt, webinterface, IR, pushbuttons and rotary switch. The XML library is replaced by a simple parser.  I will check the docs for that.

ilioss, 2017-08-15:
Hi Ed,

Here a small up-date.
I installed the last version 28-07 in my 2nd radio. ( pic.)
By programming the i/o via the HTML config It was possible to get the SD and display working?
Must be via .ini as well? Correct?
If you see the list it is mentioned that the i/o 13,14,25,26,27 will be executed. However they are not working?  Tested and they will go low.
Any ideer?
The encoder is working but it looks like i have a dodgy one ( puls failure,s ) but with missing pulses it work but like cheap Charlie. No problem for me.
If you see the pic. the white band on the display. I still bussy to find out howe to start the display @ 0,0 .
[image]
rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x7fffdabd,len:-108547
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0xffffffff,len:-8193
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
cmd len -108545
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fffdaad,len:-257336
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034

D: Starting ESP32-radio Version Fri, 28 Jul 2017 12:05:00 GMT...  Free memory 194988
D: ir_pin pin set to -1
D: enc_clk pin set to 33
D: enc_dt pin set to 32
D: enc_sw pin set to 12
D: tft_cs pin set to 15
D: tft_dc pin set to 2
D: sd_cs pin set to 21
D: vs_cs pin set to 5
D: vs_dcs pin set to 16
D: vs_dreq pin set to 4
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO13 will execute upvolume = 2
D: GPIO14 will execute downvolume = 2
D: GPIO25 will execute preset = 6
D: GPIO26 will execute uppreset = 1
D: GPIO27 will execute downpreset = 1
D: GPIO32 will execute enc_dt
D: GPIO33 will execute enc_clk
D: Start TFT
D: Locate mp3 files on SD, may take a while...
D: 11 tracks on SD
D: Added Leeuwarderweg29wifi to list of networks
D: Added Leeuwarderweg29 to list of networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 1
D:  1 - Leeuwarderweg29wifi       Signal: -68 dBm, Encryption WPA_PSK, Acceptable
D: End of list
D: Command: mqttbroker with parameter none
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: mqttpasswd with parameter ******
D: Command: volume with parameter 63
D: Command: toneha with parameter 4
D: Command: tonehf with parameter 7
D: Command: tonela with parameter 11
D: Command: tonelf with parameter 13
D: Command: preset with parameter 5
D: Command: clk_server with parameter pool.ntp.org
D: Command: clk_offset with parameter 1
D: Command: clk_dst with parameter 1
D: Command: vs_cs with parameter 5
D: Command: vs_dcs with parameter 16
D: Command: vs_dreq with parameter 4
D: Command: tft_cs with parameter 15
D: Command: tft_dc with parameter 2
D: Command: enc_clk with parameter 33
D: Command: enc_dt with parameter 32
D: Command: enc_sw with parameter 12
D: Command: sd_cs with parameter 21
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Connected to Leeuwarderweg29
D: IP = 192.168.2.15
D: Start server for commands
D: Rotary encoder is enabled
D: Sync TOD, new value is 12:37:51
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (5/0) from icecast.omroep.nl/radio5-bb-mp3
D: Connect to new host icecast.omroep.nl/radio5-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio5-bb-mp3
D: Connected to server
D: Switch to HEADER
D: Server: Icecast 2.4.2
D: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Cache-Control: no-cache
D: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Pragma: no-cache
D: icy-br:192
D: ice-audio-info: bitrate=192
D: icy-br:192
D: icy-genre:Mixed
D: icy-name:Radio 5
D: icy-pub:0
D: icy-url:http://www.radio5.nl
D: icy-metaint:16000
D: Switch to DATA, bitrate is 192, metaint is 16000
D: First chunk:
D: FD C2 A0 D0 70 1B 91 55
D: BD 51 48 30 00 40 20 00
D: 01 53 A8 1A 0A 00 8E DF
D: 7C DC 3F 8C 46 50 F6 67
D: Metadata block 48 bytes
D: Streamtitle found, 44 bytes
D: StreamTitle='Peter&Gordon - True Love Ways';

Edzelf, 2017-08-15:
.ini does not exist in this version.

You have to patch the TFT driver, see documentation.

I see:
D: GPIO32 will execute enc_dt
D: GPIO33 will execute enc_clk

If you push the buttons, it should be logged in the debug output.  Test if the signals are pulled to ground.
That is not correct, you cannot bind commands to pins that are used for encoder switch, so remove the preferences for "gpio_32 = ... " and "gpio_33 = ..."

GPIO02 is LOW,  should be HIGH.  probably a wiring problem.

ilioss, 2017-08-15:
Hi Ed,

Thanks for the swift reply.
I realy do appreciate.
I will re-solder the encoder to 25.26,27
Double check the gpio02. Hardware as well as with the debug feature.
As it is not quit clear to me about the free to program i/o,s.
What is the gpio 21 function in the preferences it is mentioned for sd_cs = 21  and gpio_21 = station = icecast.omroep.nl:80/radio1-bb-mp3?
For my vol+,vol-, station+,station-,station preset=6 I will use gpio 12,13,14,17 and  will this work?
Sorry for the inconviniously questions.
Kind regards,

ps. Did you experience dodgy encoder pulses?
Theo

Edzelf, 2017-08-15:
Show me you preferences (without WiFi settings for safety) and I will correct them.  You may also mail it to me directly.  My rotary encoder seems to work correct.  I used this one: https://www.aliexpress.com/item/1PCS-FREE-shipping-Rotary-Encoder-Module-Brick-Sensor-Development-for-Dropshipping-KY-040/32726570531.html.
The encoder should work fine on any 3 free I/Os.
You may change line 1600 like:
  if (  (  newtime - oldtime ) < 2 )                                // Debounce

ilioss, 2017-08-26:
As the question is ansered  all controls can be in parallel.
I close the issue.
Ed thanks a lot.
Kind regards.
Theo

closed 2017-08-26 by ilioss