Edzelf/ESP32-Radio #38: Using D-duino32, Wemos 32 with built-in LCD

fenyvesi, 2017-08-26:
I tried the ESP8266 version also, it is excellent.
I'd use the ESP32 verison now and I have some modules with built-in I2C OLED. 
I have this one: 
[https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042647.6.2.v07Jjm&orderId=85346295631073&productId=32807483723](url)

Unfortunately the number of exposed pins is minimal and stupid a bit. The PINs you used for SPI are not exposed.

The question: Can I use another set of SPI PINs (not the standard VSPI)  in Arduino?
I tried:
SPI.begin(12,13,14); without success. 

I checked the connections, seems to be OK.
Do you have any idea/advice?

Thank you.

George
[image]

Edzelf, 2017-08-26:
SPI should work with different pins.
The header file shows:
void begin(int8_t sck=SCK, int8_t miso=MISO, int8_t mosi=MOSI, int8_t ss=-1);

fenyvesi, 2017-08-26:
Yeah. I checked myself there. 
The result:
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 error retry SB:0000 R1:0773 R2:E730
D: VS1053 error retry SB:012C R1:0000 R2:0000
D: VS1053 error retry SB:0258 R1:0000 R2:FFFF
D: VS1053 error retry SB:0384 R1:0000 R2:0000
D: VS1053 error retry SB:04B0 R1:0000 R2:0003
.....

fenyvesi, 2017-08-28:
Just for the records, a working setup:

vs_sck  = 14                                         # GPIO Pin number for VS1053 SCK
vs_miso	= 12                                     # GPIO Pin number for VS1053 MISO
vs_mosi	= 13                                     # GPIO Pin number for VS1053 MOSI
vs_cs = 15                                            # GPIO Pin number for VS1053 "CS"
vs_dcs = 16                                          # GPIO Pin number for VS1053 "DCS"
vs_dreq = 2                                          # GPIO Pin number for VS1053 "DREQ"

SPI.begin(vs_sck,vs_miso,vs_mosi);

Edzelf, 2017-08-28:
SPI pins are now configurable in the preferences.

fenyvesi, 2017-08-29:
Thank you.

In oreder to modify the program because of the different OLED scrren, PINs etc. I had to look through it.
I must say that the richness of functionality, the clear structure, the comments, in one word the quality is admirable.

I can't even estimate the effort you took.

Thank you once more.

George

fenyvesi, 2017-08-29:
One more additon: I included mDNS responder in order not to check the IP on the screen.

closed 2017-08-30 by fenyvesi

fenyvesi, 2017-08-31:
Just a question. You use a different VS1053 module as mine. Can you connect
the amplifier directly or did you use a separate driving?

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mentes
a vírusoktól. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Edzelf, 2017-08-31:
I do not know what kind of module you have.  But disconnect the amplifier and meassure the voltage on the 3 pins of the audio connector with respect to GND.  If this is not 0 Volt (I expect 1.25 V), then you cannot connect an amplifier that shares the GND with the VS1053 circuit.  A separate power supply for the amplifier is necessarary.  See the amplifier circuit in the documentation.

fenyvesi, 2017-08-31:
I’ve already checked and this is the bare „floating” ground of VS1053. I need a passive driving circuit. 
The module you use is dirrerent and I see on Aliexpress that it is called line-out several times. 
So the question that do you connect the output of your module directly to the amplifier?
( Does your module consist of the driving circuit for line-in?)

George

Küldte a Windows 10 Posta

Feladó: Ed Smallenburg
Elküldve: 2017. augusztus 31., csütörtök 13:51
Címzett: Edzelf/ESP32-Radio
Másolatot kap: George Fenyvesi; State change
Tárgy: Re: [Edzelf/ESP32-Radio] Using D-duino32, Wemos 32 with built-in LCD(#38)

I do not know what kind of module you have. But disconnect the amplifier and meassure the voltage on the 3 pins of the audio connector with respect to GND. If this is not 0 Volt (I expect 1.25 V), then you cannot connect an amplifier that shares the GND with the VS1053 circuit. A separate power supply for the amplifier is necessarary. See the amplifier circuit in the documentation.
—

---
Az e-mailen az AVG vĂ­rusellenĹ‘rzĂ©st vĂ©gzett.
http://www.avg.com

Edzelf, 2017-08-31:
Yes, connect the 3 lines of the ouput connector to:
Common - ground of the amplifier and the ground of the 2nd power supply.
L - to the amplifiers left input
R - to the amplifier right input