Edzelf/ESP32-Radio #39: Someones tindie project based off esp-radio

NonaSuomy, 2017-08-27:
google now spotted this for me, do you know him @Edzelf ?
 https://www.tindie.com/products/masihvahida/esp8266-internet-radio/

koskee, 2017-08-27:
well, the board looks like its well made, at least. I certainly wouldn't be coughing up... woah. I just realized that the price at the top is JUST for the PCB. You want the fully assembled version? SURE - that'll be a cool $239US, which is almost $300CAD. Seriously? 
Those boards can't have been more than 5 bucks/piece to have made up, and i think we all know how much the components cost.. Highway robbery is what that is, WOW.

... At least they gave credit (no link though). There's a ton of shady people out there that wouldn't even do that.

NonaSuomy, 2017-08-27:
I didn't even look at that... that is pretty messed up.

Edzelf, 2017-08-28:
I was not aware of these publications.  I do not know "masivahida".  The price must be a typo, USD 5.90 looks reasonable to me...

NonaSuomy, 2017-08-29:
haha, I noticed that there was only one bought so he probably obligatory "bought" it himself to show interest... Nice bodge wire on the lower left of the back of the board too.

Edzelf you should do a proper affordable Tindie project for fun.