Edzelf/ESP32-Radio #41: Feature request: I2S DAC and Touch control

harmverbeek, 2017-08-27:
The ESP32 supports I2S, so why not use the I2S output to connect a Audio DAC like this one: https://www.aliexpress.com/item/Interface-I2S-PCM5102-DAC-Decoder-GY-PCM5102-I2S-Player-Module-For-Raspberry-Pi/32807703240.html

It is very cheap and will probably solve the noise issue as wel.

Another thing is, the ESP32 supports touch control. That could replace the mechanical controls.

locki-cz, 2017-08-28:
Noise in audio is terrible and i cant use batteries :(

Edzelf, 2017-08-28:
Just use 2 separate power supplies, or one power supply and buy DC-DC converter for 80 cents like https://www.aliexpress.com/item/1PCS-B0505S-1W-5V-to-5V-converter-DC-DC-power-module-converter-1000VDC-Isolation/32712121720.html.

Edzelf, 2017-08-28:
@harmverbeek : Programming the software for mp3, aac, ogg, .... conversion is a lot of work, so I don't think I will step in to this.

harmverbeek, 2017-08-28:
@locki-cz : I used an I2S DAC on a Raspberry Pi Zero, all on breadboard, no noise at all. Or do you mean the touch sensors generate noise? These are also nice: https://www.aliexpress.com/item/10-pcs-red-color-HTTM-Series-2-7V-6V-Capacitive-Touch-Switch-Module-Strong-anti-interference/32814839036.html

harmverbeek, 2017-08-28:
@Edzelf : I don't see how you could completely isolate the power supplies with a DC-DC converter, they would have to share the ground line.

Edzelf, 2017-08-28:
Who says so?

Edzelf, 2017-08-28:
Touch control is now implemented.

harmverbeek, 2017-08-28:
Cool!

Landor-Nirsan, 2017-09-01:
Hi,
I got rid of the noise by modifying the VS1053 board's output. Cut the track from the chip (pin 42) to the output connector's ground pin between the resistor and the output plug. This is important, because the GBUF output needs to have a resistor/capacitor combo to ground. Then tie the connector's ground pin to "real" ground. Also you need to cut the track coming from pins 39 and 46 (left/right outputs), again after the resistor/cap combo, and insert a 2.2 microfarad capacitor into the gap you made by cutting the track. This must be done to decouple the DC on the chip's outputs. Not every amp has AC coupled inputs.
See drawing.

[image]

It's a bit fiddly but worth the effort.

Landor-Nirsan, 2017-09-01:
Or just take out the entire output connector and make the mods off-board... Of course keeping the wires from where the connector used to be to your mod as short as possible.

rihani, 2017-11-09:
any help to convert this o play through PCM5102A please

sansjunk, 2019-09-08:
I came across this modification to produce better analog output for connecting to amps. [Line out, Line in and Headphone connections](http://www.vsdsp-forum.com/phpbb/viewtopic.php?f=9&t=69)

eXadra, 2019-09-10:
Hello guys, the thing with the VS1053 output is that like said before the signal of left and right audio is not referenced to GND but to a Vreference, this is used to boost the audio output level of the phones.
T**he  only  way** to solve this problem without loosing audio quality  is  by using 2 isolating 600:600R audio micro transformers (check ebay), one for left and another for right aoudio signals. the transformers are connected in one side between audio (L or R) and reference and in the other side between ground and amplifier input. The audio is cristal clear without any distortion or wifi noise.