Edzelf/ESP32-Radio #42: MQTT - Ping

fenyvesi, 2017-08-29:
Just a question: if my MQTT server because of network problems is unreachable then the program tries to connect all the time and playing doesn't work.

I tried to ping the server but I couldn't find a ping library.

Any idea?

Edzelf, 2017-08-29:
It was designed to stop after a number of tries.  I will check it.

fenyvesi, 2017-08-30:
I've waited cca. 5 minutes and the MQTT routine tried and tried and tried.

In case of a working solution the MQTT functionality should be disabled because you can't even reach the WEB settings, and if you restart the program, it is in vain, the situation remains the same. Only uploading a modified program through USB helped.

Edzelf, 2017-08-30:
The number of retries was 50.  I changed this number to 5.

fenyvesi, 2017-08-30:
Tried, works. Cca. 3 minutes.

closed 2017-08-30 by fenyvesi