Edzelf/ESP32-Radio #44: Crash in readhostfrompref ( enc_preset ) routine

fenyvesi, 2017-09-06:
If I turn the encoder in PRESET mode back and forth 15-20 times, than the progrem crashes.
( I must say that the encoder interrupt handler is changed, it is much quicker. I will publish it here. Can it happen that ithe isr routine tries to write something at "the same time"?)

It seems to me, that it happens in the mentioned routine. It happens more often in the 1st call, this example is in the 2nd one.

The last existing preset station is 7 . Shouldn't we check it for max before calling readhostfrompref ( enc_preset )?

The program excerpt in chk_enc() routine is :

       dbgprint ( "chk_enc 1 preset:%d, ",enc_preset);
       delay(100);                                                             // crashes in readhostfrompref ( enc_preset ) !!
       tmp = readhostfrompref ( enc_preset ) ;                 // Get host spec and possible comment
       dbgprint ( "chk_enc 11");
       if ( tmp == "" )                                                     // End of presets?
        {
          dbgprint ( "chk_enc 12");
          delay(100);                                                       // crashes in readhostfrompref ( enc_preset ) !!
          enc_preset = 0 ;                                               // Yes, wrap
          tmp = readhostfrompref ( enc_preset ) ;           // Get host spec and possible comment
          dbgprint ( "chk_enc 13");
      }

and the debug messages (it calls the routine with non-existing sation 9):

D: chk_enc 1 preset:9, 
D: chk_enc 11
D: chk_enc 12
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x40080df2  PS      : 0x00060034  A0      : 0x80081104  A1      : 0x3ffc0bc0  
A2      : 0x0000000b  A3      : 0x00000000  A4      : 0x3ffc3c2c  A5      : 0x3ffcf9e0  
A6      : 0x00000020  A7      : 0x3ffc8c50  A8      : 0x00000000  A9      : 0x3ffc0ba0  
A10     : 0x3f4022d9  A11     : 0x3ffc36aa  A12     : 0x80082daf  A13     : 0x3ffcf9c0  
A14     : 0x3ffc487c  A15     : 0xffffffff  SAR     : 0x00000004  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000164d  LEND    : 0x40001667  LCOUNT  : 0xfffffffd  

Backtrace: 0x40080df2:0x3ffc0bc0 0x40081104:0x3ffc0be0 0x40082205:0x3ffc0c00

CPU halted.

Edzelf, 2017-09-06:
readhostfrompref returns an empty string if the parameter "preset" is off-limit.  So there is no need for an extra check.  The empty string will cause a wrap-around to preset zero.
I wouldn't put any delays in chk_enc.

fenyvesi, 2017-09-06:
Yeah, I saw. The delay was an experiment to balance the quicker isr routine.
The crash theoritically can happen in the timer or encoder isr-s, also. I can't check. A memory map would be useful, I don't now how to get it from Arduino translation.

The strange situation I had that translated the program and I got an illegel instaruction crash.

Restareted Arduino retranslated and none of the crashes appear now. Strange. I will test a bit further.

fenyvesi, 2017-09-06:
I deleted the delay() from chk_enc(). Unfortunately it crashed again.

It seems that the return of the readhostfrompref(enc_preset) crashes. May be stack problem.
The code:
```
The calling in chk_enc():
    case PRESET :
	  enc_preset += rotationcount ;                           // Change preset
      if ( enc_preset < 0 )                                   // Check limits
      {
        enc_preset = 0 ;                                      // Not lower than 0
      }
//	  delay(100);												// crashes in readhostfrompref ( enc_preset ) !!
 dbgprint ( "chk_enc 1 preset:%d, ",enc_preset);
      tmp = readhostfrompref ( enc_preset ) ;                 // Get host spec and possible comment
  dbgprint ( "chk_enc 11");
       if ( tmp == "" )                                        // End of presets?
      {
    dbgprint ( "chk_enc 12");
       enc_preset = 0 ;                                      // Yes, wrap
        tmp = readhostfrompref ( enc_preset ) ;               // Get host spec and possible comment
   dbgprint ( "chk_enc 13");

The routine called:
String readhostfrompref ( int8_t preset )
{
  char           tkey[12] ;                            // Key as an array of chars

  dbgprint ( "rhfp 1");
  sprintf ( tkey, "preset_%02d", preset ) ;            // Form the search key
  dbgprint ( "rhfp 2");
  if ( nvssearch ( tkey ) )                            // Does it exists?
  {
    // Get the contents
  dbgprint ( "rhfp 3");
    return nvsgetstr ( tkey ) ;                        // Get the station (or empty sring)
  }
  else
  {
  dbgprint ( "rhfp 4");
    return String ( "" ) ;                             // Not found
  }
}

```
The result:

D: chk_enc 1 preset:5, 
D: rhfp 1
D: rhfp 2
D: rhfp 3
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x40080df2  PS      : 0x00060034  A0      : 0x80081104  A1      : 0x3ffc0bc0  
A2      : 0x0000000b  A3      : 0x00000000  A4      : 0x3ffc3c2c  A5      : 0x3ffcf9a0  
A6      : 0x00000000  A7      : 0x3ffcf9d0  A8      : 0x00000000  A9      : 0x3ffc0ba0  
A10     : 0x3f402235  A11     : 0x3ffc36aa  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000003  A15     : 0x00060e23  SAR     : 0x00000004  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000164d  LEND    : 0x40001667  LCOUNT  : 0xfffffffd  

Backtrace: 0x40080df2:0x3ffc0bc0 0x40081104:0x3ffc0be0 0x40082205:0x3ffc0c00

CPU halted.

I believe that it is a compilation error, may be in esp-idf itself. 

I am definitely not a guru but will meditate, what to do.

fenyvesi, 2017-09-06:
May be not the return but the nvsgetstr() routine itself.
```

//**************************************************************************************************
//                                      N V S G E T S T R                                          *
//**************************************************************************************************
// Read a string from nvs.                                                                         *
// A copy of the key is used because nvs_get_str will fail after some time otherwise....           *
//**************************************************************************************************
String nvsgetstr ( const char* key )
{
  uint32_t      counter ;
  static char   nvs_buf[NVSBUFSIZE] ;       // Buffer for contents
  size_t        len = NVSBUFSIZE ;          // Max length of the string, later real length

  nvsopen() ;                               // Be sure to open nvs
  nvs_buf[0] = '\0' ;                       // Return empty string on error
  nvserr = nvs_get_str ( nvshandle, key, nvs_buf, &len ) ;
  if ( nvserr )
  {
    dbgprint ( "nvs_get_str failed %X for key %s, keylen is %d, len is %d!",
               nvserr, key, strlen ( key), len ) ;
    dbgprint ( "Contents: %s", nvs_buf ) ;
  }
  return String ( nvs_buf ) ;
}
```
You comment is:

> A copy of the key is used because nvs_get_str will fail after some time otherwise...

The routine doesn't print the error message, so the most suspicious element is:

`nvs_get_str ( nvshandle, key, nvs_buf, &len ) `

fenyvesi, 2017-09-06:
nvsopen() ;  works, so the problem is the nvs_get_str() routine. I know that it is buggy and you can't do anything.

Edzelf, 2017-09-06:
The comment is not relevant after the update of the nvs library.  The copy of the key is already removed from the code.
Are you sure you have the last version of the nvs-library (after 12 may 2017)?
See [here](https://github.com/Edzelf/ESP32-Radio/issues/2)

fenyvesi, 2017-09-06:
I don't know hows to check. The ESP32 Arduino was originally installed in July. Today I updated for the last one. The error remained the same, maybe occurs not so often but I am not sure.
( I can reproduce the error if I turn the encoder 15-20 times back and forward. )

D: nvsg 1
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x40080df2  PS      : 0x00060034  A0      : 0x80081104  A1      : 0x3ffc0bc0  
A2      : 0x000000ba  A3      : 0x00000001  A4      : 0x3ffc3c2c  A5      : 0x3ffcf9a0  
A6      : 0x00000000  A7      : 0x3ffcf9d0  A8      : 0x00000000  A9      : 0x3ffc0ba0  
A10     : 0x3f40223b  A11     : 0x3ffc36aa  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000003  A15     : 0x00060823  SAR     : 0x00000004  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000164d  LEND    : 0x40001667  LCOUNT  : 0xfffffffd  

Backtrace: 0x40080df2:0x3ffc0bc0 0x40081104:0x3ffc0be0 0x40082205:0x3ffc0c00

CPU halted.

fenyvesi, 2017-09-06:
The date of
C:\Users\...\Documents\Arduino\hardware\espressif\esp32\tools\sdk\lib

23.07.2017

fenyvesi, 2017-09-06:
nvs.h has the same date.

C:\Users\...\Documents\Arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\lib
is today.

Edzelf, 2017-09-07:
I added my binary Esp32_radio.ino.bin to the "bin" directory.  Please try to flash this to your ESP32 and test it.  The command to flash this is like:

fenyvesi, 2017-09-07:
I uploaded. ( No screen.) I am sending my source in mail.
The result:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034

D: Starting ESP32-radio Version Wed, 30 Aug 2017 08:30:00 GMT...  Free memory 194516
D: ir_pin pin set to -1
D: enc_clk pin set to 26
D: enc_dt pin set to 36
D: enc_sw pin set to 25
D: tft_cs pin set to -1
D: tft_dc pin set to -1
D: sd_cs pin set to -1
D: vs_cs pin set to 15
D: vs_dcs pin set to 16
D: vs_dreq pin set to 2
D: spi_sck pin set to -1
D: spi_miso pin set to -1
D: spi_mosi pin set to -1
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
Guru Meditation Error of type LoadStoreError occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x40001277  PS      : 0x00060930  A0      : 0x800ebd2c  A1      : 0x3ffcfe50
A2      : 0x40080dbc  A3      : 0x3f4020c2  A4      : 0x00000004  A5      : 0x3ffcc20c
A6      : 0x44530064  A7      : 0x44530064  A8      : 0x80000000  A9      : 0x3ffcfe70
A10     : 0x3ffcc208  A11     : 0x00000002  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x3ffcfe20  SAR     : 0x00000010  EXCCAUSE: 0x00000003
EXCVADDR: 0x40080dbc  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

Backtrace: 0x40001277:0x3ffcfe50 0x400ebd2c:0x3ffcfe60 0x400ebd45:0x3ffcfe80 0x400d984b:0x3ffcfef0 0x400d8d04:0x3ffcff20 0x400d5898:0x3ffcff50 0x401266be:0x3ffcffa0

CPU halted.

fenyvesi, 2017-09-07:
Küldte a Windows 10 Posta

Feladó: Ed Smallenburg
Elküldve: 2017. szeptember 7., csütörtök 10:16
Címzett: Edzelf/ESP32-Radio
Másolatot kap: George Fenyvesi; Author
Tárgy: Re: [Edzelf/ESP32-Radio] Crash in readhostfrompref ( enc_preset )routine (#44)

I added my binary Esp32_radio.ino.bin to the "bin" directory. Please try to flash this to your ESP32 and test it. The command to flash this is like:
—

---
Az e-mailen az AVG vĂ­rusellenĹ‘rzĂ©st vĂ©gzett.
http://www.avg.com

fenyvesi, 2017-09-08:
It seems to me that if in the chk_enc() routine there is any delay (because of debug print or explicit delay() ), then the program works. Without them crashes immediately. Strange. If I added the two 1 ms delays, then it works.
I know that reading the program excerpt is not possible, but contrary that I have the feeling that it is related somehow to the encoder turning interrupt called frequently. Is delay() blocking isr(), too? 

The crash can be caused by memory violation (for example calling the readhostfrompref ( enc_preset ) with negative/invalid enc_preset. I see that it can't happen in the program sequence but this my feeling.
```
    case PRESET :
      enc_preset += rotationcount ;                           // Change preset
      if ( enc_preset < 0 )                                   // Check limits
      {
        enc_preset = 0 ;                                      // Not lower than 0
      }
// dbgprint ( "chk_enc 1 preset:%d, ",enc_preset);
      delay(1);												// crashes in readhostfrompref ( enc_preset ) !!
      tmp = readhostfrompref ( enc_preset ) ;                 // Get host spec and possible comment
//  dbgprint ( "chk_enc 11");
       if ( tmp == "" )                                        // End of presets?
      {
        delay(1);												// crashes in readhostfrompref ( enc_preset ) !!
//    dbgprint ( "chk_enc 12");
        enc_preset = 0 ;                                      // Yes, wrap
        tmp = readhostfrompref ( enc_preset ) ;               // Get host spec and possible comment
//   dbgprint ( "chk_enc 13");
      }

```

fenyvesi, 2017-09-08:
I checked the time costants of my cheap encoder on a scope.
If I turn it quickly, then the low phase is not less than 20 ms.
The push button is minimum 100 ms.

It seems to be huge.

Edzelf, 2017-09-08:
Nothing to worry about.  Mail me your sourcecode and I will try to reproduce the crash.

fenyvesi, 2017-09-08:
OK. Thank you. 

I checked the number of interrupts:
Switch: 4-30 /push (100-150 msec period)
Turning: max 300 /sec (1-51 interrupt/movement, 20-30 ms period)

This is a rar file, only zip is allowed. I renamed.
[ESP32_Radio.zip](https://github.com/Edzelf/ESP32-Radio/files/1287390/ESP32_Radio.zip)

Edzelf, 2017-09-08:
Sorry, I can not compile this version.  Please supply a version without SSD1306.

koskee, 2017-09-09:
I haven't been entirely following this thread, but to answer at least one of your questions:

Keep in mind that on both esp8266 and esp32 that the delay() function is tied to yield() and is used to pass control from your program to the underlying OS so it can maintain your wireless connection and other housekeeping that it needs to do. If you have the processor tied up for too long you will often get crashes. I think this was much more common on the 8266 however, as I haven't had nearly the trouble with the 32 thus far. Nevertheless, this may be why a delay is necessary for proper function, and in any case is something to keep in mind if you were not aware before. 

You also cannot use delay in an ISR function because they are meant to be kept extremely fast and basic (since the function is interrupting the main thread.)  Delay would be diverging entirely the opposite direction from this requirement. Get in, get out, and declare your variables volatile is what I always have in the back of my mind for working with ISRs

Sorry if this is completely irrelevant lol

fenyvesi, 2017-09-09:
Not irrelevant at all. There is no delay() in the isr(), only in the main thread. My feeling is on the basis of behaviour of the program ( not the written code) that without delay() the frequent isr() calling generates somehow a processor collision /memory allocation/ stack/ mutex error that produces the crash.

If the seemingly unnecessary delay() is included the critical timing for collision is avoided and the program works. No evidence only feeling.

koskee, 2017-09-09:
Well if you are doing non-atomic operations on variables in isr and the same variables in the main thread then collisions are possible. Especially since the esp32 has a dual core (ie. 2) processors running at any given time. Perhaps you need to declare the isr atomically when you enter the isr to prevent the other processor from modifying the variables at the same time. Don't ask me how to do this on the esp though, I haven't had to research that particular topic yet. The core concept applies to almost any multithreaded environment/platform though.

RTOS might likely be a good starting point if you're wondering what to research/search for - I believe the esp's firmware is based on it entirely, as far as I am aware.

When I have a chance I'll have a look at your source and see if i can find anything obviously out of place, but it might be a day or 2, and I'm by no means an expert lol.

Edzelf, 2017-09-09:
Most likely is a stack overflow problem.  The default stacksize is 4096 words.  Check if changing this solves the problem.  Its in ...\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp.

fenyvesi, 2017-09-10:
I doubled the stack, in vain. Without 4-600 microsec delay it crashes. As a workaround it works. Thank you the efforts has been taken.

fenyvesi, 2017-09-11:
I managed to analyze the crash dump.

The reason was in incorrect interrupt  handling. I was not able to pinpoint the exact reason but here are the results:

0x40080e05: isr_enc_turn() at C:\Users\Gyuri\Documents\Arduino\ESP32_Radio/Esp32_radio.ino line 2497
0x40080e05: isr_enc_turn() at C:\Users\Gyuri\Documents\Arduino\ESP32_Radio/Esp32_radio.ino line 2497
0x40081154: __onPinInterrupt at C:\Users\Gyuri\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-gpio.c line 212
0x40082255: _xt_lowint1 at xtensa_vectors.o line ?

The line in the issue is:
`__pinInterruptHandlers[pin]();`

closed 2017-09-11 by fenyvesi