Edzelf/ESP32-Radio #45: Encoder ISR routines

fenyvesi, 2017-09-06:
My rotary encoder didn't work perfectly, becuse when I turned it in one direction, than the volume went low and high. I tried to modify the debouncing time delays, in vain.

I separeted the encoder switch and encoder turning interrupt routines. The switch is the original, the turning uses the background that the encoder is Manchester-coded, so it doesn't need debouncing.

The routines with some comments:

```
void IRAM_ATTR isr_enc_switch()
{
  static uint32_t oldtime = 0 ;                        // Time in millis previous interrupt
  static bool     sw_state ;                               // True is pushed (LOW)
  static bool     clk_state ;                               // True is activated (LOW)
  bool            newstate ;                                // Current state of input signal
  uint32_t        newtime ;                               // Current timestamp

  // Read current state of SW pin
  newstate = ( digitalRead ( ini_block.enc_sw_pin ) == LOW ) ;
  newtime = millis() ;
  if ( newtime == oldtime )                                // Debounce
  {
    return ;
  }
  if ( newstate != sw_state )                              // State changed?
  {
    sw_state = newstate ;                                  // Yes, set current (new) state
    if ( !sw_state )                                       // SW released?
    {
      if ( ( newtime - oldtime ) > 1000 )                  // More than 1 second?
      {
        longclick = true ;                                 // Yes, register longclick
      }
      else
      {
        clickcount++ ;                                     // Yes, click detected
      }
      enc_inactivity = 0 ;                                 // Not inactive anymore 
    }
  }
  oldtime = newtime ;                                      // For next compare
 
}
void IRAM_ATTR isr_enc_turn()
{
// The encoder is a Manchester coded device, the outcomes (-1,0,1) of all the previous state and actual state are stored in the enc_states[].
// full_status is a 4 bit variable, the upper 2 bits are the previous encoder values, the lower ones are the actual ones
// 4 bits covers all the possible previous and actual states of the 2 PINs, so this variable is the index enc_states[].
// No debouncing is neded, because only the valid states produce values different from 0.

  static uint8_t full_status = 0x0001;                         //lookup table index 
  uint8_t act_status;                                          // the actual value of the two encoder PINs
  static const int8_t enc_states [] = 
		{0,-1,1,0,1,0,0,-1,-1,0,0,1,0,1,-1,0};         //encoder lookup table
  
  static int8_t num;
 
  full_status <<= 2;                                          // remember previous state ( shift to the 2 upper bits )
  act_status = 2 * digitalRead ( ini_block.enc_clk_pin ) + digitalRead ( ini_block.enc_dt_pin );  // Read current state of CLK, DT pin
  full_status |= act_status ;	                              // add the actual 2 read bits to the 2 lower bits
  num += enc_states[( full_status & 0x0F )];
  if ( ( num == 1 )|| ( num == 3 ))                           // four steps forward
    { 
      num = 0; 
      rotationcount++ ;                                       // Right rotation
   }
  else if (( num == -1 ) ||( num == -3 ) )                    // four steps backwards
    {  
      num = 0; 
      rotationcount-- ;                                      // Left rotation
    }
  enc_inactivity = 0 ;    
}

```

koskee, 2017-11-16:
I have been using these routines (slightly modified to change behaviour of the buttons to suit my application)  for some time now. They seemingly work quite well. Perhaps submit a pull request?

Edzelf, 2017-11-17:
I will include this in the next release.

Edzelf, 2017-11-27:
Updated now.