Edzelf/ESP32-Radio #46: ESP-32 radio

digits1, 2017-09-08:
PROMT
Hello! I very much like the project of radio on esp32. I have such question. Why only ili9163? With other lcd, for example St7735 will be options? Thanks.

Edzelf, 2017-09-08:
There are only a few lines of code for the display.  It should be easy to replace them with the code for any display.

derpicknicker1, 2017-10-16:
To do this, do the following:
1. Get the lib for ST7735 [here](https://github.com/adafruit/Adafruit-ST7735-Library) or install it with [Library Manager](https://www.arduino.cc/en/Guide/Libraries#toc3) (search for Adafruit_ST7735)
2. Include the header file for ST7735 instead of ILI9163. Change `#include <TFT_ILI9163C.h>` to `#include <Adafruit_ST7735.h> ` in [Line 105](https://github.com/Edzelf/ESP32-Radio/blob/master/Esp32_radio.ino#L105)
3. Change [line 224](https://github.com/Edzelf/ESP32-Radio/blob/master/Esp32_radio.ino#L224) from `TFT_ILI9163C*    tft = NULL ;` to `Adafruit_ST7735*    tft = NULL ;`
4. Change [lines 2763 to 2766](https://github.com/Edzelf/ESP32-Radio/blob/master/Esp32_radio.ino#L2763) which will look like this
```
    tft = new TFT_ILI9163C ( ini_block.tft_cs_pin,
                             ini_block.tft_dc_pin ) ;    
    tft->begin() ;                                   
    tft->fillRect ( 0, 0, 160, 128, BLACK ) ; 
```

to the following:
```
    tft = new Adafruit_ST7735 ( ini_block.tft_cs_pin,
                             ini_block.tft_dc_pin, 0 ) ;    
    tft->initR(INITR_BLACKTAB);                                      
    tft->fillScreen(ST7735_BLACK);
```
Thats all.