Edzelf/ESP32-Radio #48: Can not pass compile

lixianyu, 2017-09-24:
Hi, when I compile using Arduino IDE, there is an error: "fatal error: nvs.h: No such file or directory"
How to pass compile?Thanks.

koskee, 2017-09-24:
Compiler is having trouble finding the nvs library. This indicates that the library is not installed properly. 
I believe this is included with the ESP32 Core libraries (github.com/espressif/arduino-esp32) , so you will likely need to re-download the most current version and ensure that it is correctly installed (& make sure to follow the installation instructions that accompany the core)

lixianyu, 2017-09-25:
@koskee , Hi, Thank you very much!

koskee, 2017-09-29:
No problem, did you manage to get it working?

joekinley, 2018-02-04:
I have the same problem. Installed the Arduino-esp32 as stated in their install instructions, but I still get the same error.
Any help?

Edzelf, 2018-02-04:
See https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

joekinley, 2018-02-04:
I did that, and it didn't work, because the Arduino IDE didn't use my personal sketchbook directory, even though it was specified, so I had to install it into program files. But then it still doesn't work...

Edzelf, 2018-02-05:
For Windows, the nvs.h file should be in:
C:\Users\<USER>\Documents\Arduino\hardware\espressif\esp32\tools\sdk\include\nvs_flash

joekinley, 2018-02-22:
That's the thing, it is there, but I still get that error. I tried installing, reinstalling, putting it in the other Arduino folder.... nothing.

Any other ideas?

ericoud, 2019-01-06:
I have the same issue. For me it comes from the fact I have 2 different hardware library installed one from the Arduino project (ESP8266) and one from Expressif. (ESP32)..
I tried to compile for a generic ESP8266 which belong to the Arduino stuff and the library is not available. I copy paste the lib from ESP32 to ESP8266 lib and it works fine.