Edzelf/ESP32-Radio #49: ESP32 reboots periodically

MRACME, 2017-09-29:
D: Start server for commands
D: Rotary encoder is disabled (-1/-1/-1)
Guru Meditation Error of type InstrFetchProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x00000000  PS      : 0x00060630  A0      : 0x800de7f2  A1      : 0x3ffd8070  
A2      : 0x3ffc4e54  A3      : 0x3f4016f3  A4      : 0x00000000  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800de6f2  A9      : 0x3ffd8050  
A10     : 0x3ffc4e54  A11     : 0x00000001  A12     : 0x3ffd5890  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x00000000:0x3ffd8070 0x400de7ef:0x3ffd8090 0x400d52a2:0x3ffd80c0 0x400d6973:0x3ffd80e0 0x4012d424:0x3ffd8120

Rebooting...

Edzelf, 2017-09-29:
Replace your power supply.  The ESP32 is very picky on this.

MRACME, 2017-09-29:
I did it for several times, now I use a very powerfull supply with additional decoupling caps, but no changes at all :-(
With other WIFI applications the device is working as well as expected.

Edzelf, 2017-09-29:
Did you run the ini program in the tools directory successfully?

derpicknicker1, 2017-10-15:
I have the same issue

```E (91261) esp_pthread: pthread_getspecific: not supported!
E (91261) esp_pthread: pthread_setspecific: not supported!
abort() was called at PC 0x400de823 on core 1

Backtrace: 0x40087848:0x3ffd7cd0 0x40087947:0x3ffd7cf0 0x400de823:0x3ffd7d10 0x400de86a:0x3ffd7d30 0x400def6e:0x3ffd7d50 0x400deff7:0x3ffd7d70 0x400de47a:0x3ffd7d90 0x400de431:0x3ffd7db0 0x400d82f8:0x3ffd7dd0 0x400daf0e:0x3ffd7e10 0x400d695b:0x3ffd7e30 0x4012d728:0x3ffd7e70

Rebooting...
```

Added 2200µF to 5V and 3V3 but that did not help.
I ran the ini program successfully.

derpicknicker1, 2017-10-21:
I found what caused the periodically reboot for me. It was the `ArduinoOTA.handle()`.
Here is the error message:
```
E (612172) esp_pthread: pthread_getspecific: not supported!
E (612173) esp_pthread: pthread_setspecific: not supported!
abort() was called at PC 0x400de873 on core 1

Backtrace: 0x40087848:0x3ffd7da0 0x40087947:0x3ffd7dc0 0x400de873:0x3ffd7de0 0x400de8ba:0x3ffd7e00 0x400defbe:0x3ffd7e20 0x400df047:0x3ffd7e40 0x400de4ca:0x3ffd7e60 0x400de481:0x3ffd7e80 0x400d8348:0x3ffd7ea0 0x400da466:0x3ffd7ee0 0x400d69c7:0x3ffd7f00 0x4012d778:0x3ffd7f40

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
```

I decoded the backtrace with a [tool](https://github.com/me-no-dev/EspExceptionDecoder). Here is what i get:
```
0x400de873: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 112
0x40087848: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x40087947: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x400de873: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 112
0x400de8ba: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 112
0x400defbe: __cxa_get_globals at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_globals.cc line 133
0x400df047: __cxa_throw at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 65
0x400de4ca: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x400de481: operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc line 32
0x400d8348: WiFiUDP::parsePacket() at C:\Users\Flow\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiUdp.cpp line 129
0x400da466: ArduinoOTAClass::handle() at C:\Users\Flow\Documents\Arduino\hardware\espressif\esp32\libraries\ArduinoOTA\src/ArduinoOTA.cpp line 355
0x400d69c7: loop() at C:\Users\Flow\Documents\Arduino\ESP32_radio/Esp32_radio.ino line 3665
0x4012d778: loopTask(void*) at C:\Users\Flow\Documents\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 18 (discriminator 1)
```

As you can see, the issue comes from ArduinoOTA using `WiFiUDP::parsePacket()`. I don't know where exactly the issue is, but deactivating ArduinoOTA solved the problem for me. I removed `ArduinoOTA.handle() ;` from the main loop and now everything works fine. But no OTA possible...

May also be interesting for #47

koskee, 2017-10-27:
Would like to add that I've also been having trouble with the OTA  library. Since my firmware is heavily modified from the original and is seeing regular development work, I had assumed it was just me, something I had caused with the new code. 

Upon going over other program output samples, I realize that I am getting similar abort() calls,  causing reboots at various rates when the OTA.handle() function is included in the sketch. 

I can also pinpoint exactly when I noticed it wasn't working properly, as I had just started merging the update from (around Aug 30 I think? The one that included the time library, encoder, and was I believe the last major update as of now).
Prior to this update it was working properly, and I used it most of the time - which was pretty much daily at the time. I have been missing it for the convenience factor,  so would be good to have it back.

pieman64, 2017-11-03:
@nkolban I have the same error message and I'm running OTA.handle() like @koskee and @derpicknicker1
I am not knowingly doing any BLE stuff but the error messages initially took me to https://github.com/nkolban/esp32-snippets/issues/121
Is it likely the fixes for that issue will fix our issue?

koskee, 2017-11-08:
Not sure if anyone saw this, but i have a feeling this commit would be related to this issue. If my memory serves me, I remember seeing somewhere parseInt() was the cause of the issue while I was debugging. I suspect that updating one's libraries might see this resolved.

[Fixes for zero length packet bug, buffer overflow in parseInt(), added end() method](https://github.com/espressif/arduino-esp32/pull/757)

(I haven't tried it yet myself, but I haven't had the time)