Edzelf/ESP32-Radio #50: Amplifier Power Circuit

Kneheim, 2017-10-04:
Hi, 
great work but I am wondering if there is a chance to power the ESP and the amp with the same battery.
I am not very experienced in hardware-layout. 
what measures can be taken to reduce noise in the amp power supply?
Thanks and regards
Andreas

Edzelf, 2017-10-04:
I have not tried it myself, but you may connect the L and R signals of the headphone connector through 10 microFarad capacitors to your amplifier.  Do not use the common pin of the headphone connector.

Another possibility is to us a 5 volt DC-DC converter to isolate the amplifier from the VS1053.  Search for "B0505S" on webshops.
I have some converters ordered and I will test it in the coming days.

Edzelf, 2017-10-05:
I have tested the solution with the DC-DC converter and it works.

Kneheim, 2017-10-05:
Thank you very much for the quick response.

Will try it myself the next days.
 

 

Gesendet: Donnerstag, 05. Oktober 2017 um 11:40 Uhr

derpicknicker1, 2017-10-16:
> I have tested the solution with the DC-DC converter and it works.

Do you use the DC-DC converter for VS1053 or for the amp?
The B0505S which comes in 1W or 2W versions seems to be a bit underpowerd for an amp.

Edzelf, 2017-10-16:
I tested it for the amp.  1 watt is a lot of noise, knowing that 1 microwatt can keep you out of sleep.
But maybe it's better to use the 2 Watt converter for the ESP and the VS1053.  The idea is to get the power supply of the amp separated from the rest.

derpicknicker1, 2017-10-16:
I want to use one of the cheap PAM8403 2x3W D Amp modules. So the dc/dc converter really seems to be underpowerd for the amp.
Which amp du you use?

Edzelf, 2017-10-16:
I use the same amplifier.  See my comment "But maybe it's better to use the 2 Watt converter for the ESP and the VS1053."

WiGeorg, 2018-04-15:
Hello again!
I made a mistake with the first link. Please use this link here:
http://esp8266-server.de/Internetradio.html

SimonMerrett, 2019-01-18:
> Hello everybody!
> I realized the ESP-radio with only one alimentation.
> After some research I found out a solution in this link to eliminate the noise:
> http://www.enaos.net/P1220.aspx?IdPer=485253
> I realized this modification here:
> https://user-images.githubusercontent.com/37144462/38781681-433bbce2-40e9-11e8-97a4-6c7ad9ce619a.png
> 
> The capacitance of 47uF is soldered directly to the VREF-pin of the amplifier an to the GBUF-line at the other side. This small modification eliminates the noise completely!
> 
> I suppose that you can adapt this modification too to other amplifier.
> 
> Nice projects ESP-radio and ESP32-radio!!!!

The large capacitor sorted out my issue with PAM8403 connection too, thank you!