Edzelf/ESP32-Radio #53: Sound Distortion

MMSchneider, 2017-10-20:
Hello,

I am facing distortion on some radio stations. I am not sure if this is because of cabeling (other stations work fine), the station itself or the bitrate. An example stream which causes trouble is: swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3

Here a excerpt of the log:

D: New preset/file requested (12/0) from swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3
D: Connect to new host swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3
D: Connect to swr-swr3-live.cast.addradio.de on port 80, extension /swr/swr3/live/mp3/128/stream.mp3
D: Connected to server
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-description:Mehr Hits. Mehr Kicks. Einfach SWR3
D: Headerline: icy-genre:Pop Music
D: Headerline: icy-name:SWR3
D: Headerline: icy-pub:1
D: Headerline: icy-url:http://www.swr3.de
D: Headerline: Server: dg-swr-http_dus-dtag_edge_1c578a1c46f2c8ca01cf246adf834ac5
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: Close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 0, metaint is 16000
D: First chunk:
D: FF FB 94 60 C2 80 03 0E
D: 40 5B CB 0B 1A EA 43 61
D: AB 8D 2D 2B 37 0E 71 33
D: 69 2C 3D 0D A9 22 19 AE
D: Metadata block 48 bytes
D: Streamtitle found, 35 bytes

I can see that there is no icy-br:XXX at all. Can that cause the distortion?

Thanks in advance
Mike

derpicknicker1, 2017-10-20:
Works fine for me.

Edzelf, 2017-10-20:
Sound quality is good, but I do hear some short breaks (average about 2 per minute).  @Mike: is that the distortion you experience?
This can be heard during the handling of the "streamtitle" in the data stream.  The streamtitles is included in the datas tream very often for this station.  In my setup this streamtitle is transmitted to my mqtt server, which also takes some time to handle.
I changed the configuration (mqttbroker = none).  That makes no difference.
Then I changed the configuration to "no TFT".  That also makes no difference.
The conclusion sofar is that there must be some code error in the handling of the streamtitle.
I will test this later.

MMSchneider, 2017-10-21:
@Edzelf not quite. It is really like over-amplified and like saturation. But since @derpicknicker1 does not have this issue, I sense that it may come from the ESP32 <--> VS1053 SPI connection. Maybe I do not get enough throughput. Interestingly 64kBit Streams do not have this issue. I will replace cables and make them shorter. Lets see.  

AND: @Edzelf -> Thanks for this outstanding piece of Code!

koskee, 2017-10-31:
Sounds pretty good to me too, but like Ed said, the stream has a few hiccups once in a while... 

Another thing you might want to look into is noise on your power supply. Sharing a supply with other devices has caused me issues similar to what you're describing in the past. 
I've noticed that if I am powering from the USB bus on my laptop, and I plug in another device alongside the radio, I have on occasion got a volume spike/over saturated sound which lasts until manually rebooting the device. After that it is no problem.

You could try beefing up your power supply filter, either by adding more uF (maybe try doubling it) or by adding small ceramic capacitors of various values (for example, 1nF, 10nF, 100nF, 1uF and anything in between. The more, the better) The small ceramic caps serve to filter out specific frequencies, so if you can hit the correct value for any noise you may be encountering, it should filter it out. There's a formula you can use if you know the frequency, but you'd need an oscilloscope to quantify it. 

A much simpler approach might be to try a different PS, one rated for a at least 1A, and only attach the radio (if you had other things connected).

Just some food for thought.

MMSchneider, 2018-01-21:
SOLVED - Just want to mention that the distortion has gone after replacing the wires between ESP32 and VS1053. Seems that a good connection is key. 
Thanks for all the help.

closed 2018-01-21 by MMSchneider