Edzelf/ESP32-Radio #55: esp_pthread: pthread_getspecific: not supported! => reboot.

NonaSuomy, 2017-10-21:
Getting these reboots seems to be after an MQTT publish while using ihr streams.

D: New preset/file requested (10/0) from ihr/IHR_IEDM
D: Connect to new iHeartRadio host: IHR_IEDM
D: GET /api/livestream?version=1.5&mount=IHR_IEDMAAC&lang=en
D: Connected to playerservices.streamtheworld.com
D: XML parser processing...
D: Found: 14833.live.streamtheworld.com:3690/IHR_IEDMAAC_SC
D: Connect to new host 14833.live.streamtheworld.com:3690/IHR_IEDMAAC_SC
D: Connect to 14833.live.streamtheworld.com on port 3690, extension /IHR_IEDMAAC_SC
D: Connected to server
D: Publish to topic P83C40A24/ip : 10.13.37.177
D: Switch to HEADER
D: Headerline: Expires: Thu, 01 Dec 2003 16:00:00 GMT
D: Headerline: Cache-Control: no-cache, must-revalidate
D: Headerline: Pragma: no-cache
D: Headerline: Access-Control-Allow-Methods: GET, HEAD
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Credentials: true
D: Headerline: Set-Cookie: uuid=e6eaf42e-d62c-4ebf-be8a-088b2d116c65; expires=Sat, 22-Dec-2029 21:30:41 GMT; path=/; domain=.live.st
D: Headerline: Content-Type: audio/aacp
D: audio/aacp seen.
D: Headerline: Connection: close
D: Headerline: icy-br: 48
D: Headerline: icy-description: All Things Dance
D: Headerline: icy-genre: 
D: Headerline: icy-name: Evolution
D: Headerline: icy-url: 
D: Headerline: icy-metaint: 16000
D: Headerline: Server: MediaGateway 4.4.2-002
D: Switch to DATA, bitrate is 48, metaint is 16000
D: First chunk:
D: FF F1 5E 40 20 BF FC 01
D: 3A 36 2C AB E8 0A 91 86
D: A1 01 24 06 AC 0A 00 50
D: 92 92 83 C2 D7 38 12 08
D: Publish to topic P83C40A24/icy/name : Evolution
D: Metadata block 64 bytes
D: Streamtitle found, 52 bytes
D: StreamTitle='Cash Cash / Sofia Reyes - How To Love';
D: Publish to topic P83C40A24/icy/streamtitle : Cash Cash / Sofia Reyes - How To Love
E (27945) esp_pthread: pthread_getspecific: not supported!
E (27946) esp_pthread: pthread_setspecific: not supported!
abort() was called at PC 0x400de7ef on core 1

Backtrace: 0x400879c8:0x3ffd9100 0x40087ac7:0x3ffd9120 0x400de7ef:0x3ffd9140 0x400de836:0x3ffd9160 0x400def3a:0x3ffd9180 0x400defc3:0x3ffd91a0 0x400de446:0x3ffd91c0 0x400de3fd:0x3ffd91e0 0x400d8324:0x3ffd9200 0x400daeda:0x3ffd9240 0x400d6987:0x3ffd9260 0x4012e8ac:0x3ffd92a0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

derpicknicker1, 2017-10-21:
Get [this tool](https://github.com/me-no-dev/EspExceptionDecoder) to decode your error message and see what the real issue is.
I had the same errors and for me it was ArduinoOTA. See #49

NonaSuomy, 2017-10-22:
Thanks, I'll check that tip out, commented out arduinoOTA, so far stable, we'll see what time brings, I'll look into the debugger tool later. :+1:

NonaSuomy, 2017-10-22:
Well looks like it breaks the MQTT connection still just like on the ESP8266 but at least it doesn't reboot now.

D: Metadata block 64 bytes
D: Streamtitle found, 53 bytes
D: StreamTitle='Fall Out Boy - Sugar, We're Goin' Down';
D: Publish to topic espradio/icy/streamtitle : Fall Out Boy - Sugar, We're Goin' Down
D: Metadata block 48 bytes
D: Streamtitle found, 38 bytes
D: StreamTitle='MISSIO - Middle Fingers';
D: Publish to topic espradio/icy/streamtitle : MISSIO - Middle Fingers
D: Publish to topic espradio/ip : 10.13.37.177
D: (Re)connecting number 3 to MQTT 10.13.37.52
D: MQTT connection failed, rc=-2
D: STOP requested
D: Stopping client
D: Song stopped correctly after 0 msec
D: New preset/file requested (1/0) from airspectrum.cdnstream1.com:8114/1648_128
D: Connect to new host airspectrum.cdnstream1.com:8114/1648_128
D: Connect to airspectrum.cdnstream1.com on port 8114, extension /1648_128
D: Connected to server
D: (Re)connecting number 4 to MQTT 10.13.37.52
D: STOP requested
D: Stopping client
D: Song stopped correctly after 0 msec
D: New preset/file requested (2/0) from us2.internet-radio.com:8050
D: Connect to new host us2.internet-radio.com:8050
D: Connect to us2.internet-radio.com on port 8050, extension /
D: Connected to server
D: Publish to topic espradio/ip : 10.13.37.177
D: Switch to HEADER
D: Headerline: icy-name:CLASSIC ROCK MIAMI
D: Headerline: icy-genre:Classic Rock Album Light Rock 60s 70s 80s Oldies
D: Headerline: icy-url:http://www.ClassicRockmiami.online
D: Headerline: content-type:audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-pub:1
D: Headerline: icy-metaint:32768
D: Headerline: icy-br:256
D: Switch to DATA, bitrate is 256, metaint is 32768
D: First chunk:
D: 12 B8 B3 C7 7D AC 28 A1
D: 51 FC 05 05 11 10 18 48
D: AD 31 FD 52 75 2B 89 6D
D: 6B 5A 75 A4 8F CB 26 CA
D: Publish to topic espradio/icy/name : CLASSIC ROCK MIAMI
D: Metadata block 48 bytes
D: Streamtitle found, 46 bytes
D: StreamTitle='Advert: - Advert: ';StreamUrl='';
D: Publish to topic espradio/icy/streamtitle : Advert: - Advert: 
D: (Re)connecting number 5 to MQTT 10.13.37.52
D: MQTT connection failed, rc=-2
D: (Re)connecting number 6 to MQTT 10.13.37.52
D: MQTT connection failed, rc=-2
D: STOP requested
D: Stopping client
D: Song stopped correctly after 0 msec

NonaSuomy, 2017-10-22:
Guess some more debug sleuthing will be required...D: Metadata block 48 bytes
D: Streamtitle found, 36 bytes
D: StreamTitle='AFI - Girl's Not Grey';
D: Publish to topic espradio/icy/streamtitle : AFI - Girl's Not Grey
Guru Meditation Error: Core  0 panic'ed (Double exception)
Register dump:
PC      : 0x4011dbab  PS      : 0x00060036  A0      : 0x801202f0  A1      : 0x3ffde6e0  
A2      : 0x3ffde750  A3      : 0x00000000  A4      : 0x3ffe99fc  A5      : 0x000000ff  
A6      : 0x00002aaa  A7      : 0x3ffde780  A8      : 0x800fa100  A9      : 0x00000014  
A10     : 0x3ffe975c  A11     : 0x400ff1e8  A12     : 0x3ffea7e6  A13     : 0x00000006  
A14     : 0x00000aaa  A15     : 0x3ffde740  SAR     : 0x0000001e  EXCCAUSE: 0x00000002  
EXCVADDR: 0x00000014  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

Backtrace: 0x4011dbab:0x3ffde6e0 0x401202ed:0x3ffe9796

Rebooting...
ets Jun  8 2016 00:22:57

Edzelf, 2017-10-22:
Maybe this has something to do with the qoute in the streamtitle.

derpicknicker1, 2017-10-22:
@NonaSuomy 
Again: Get [this tool](https://github.com/me-no-dev/EspExceptionDecoder) to decode your error message and see what the real issue is. The backtrace you posted is not usable if not decoded.

Copy the `Backtrace: 0x4011dbab:0x3ffde6e0 0x401202ed:0x3ffe9796` into the tool and it will decode it and show you which function(s) caused the error.

Disabeling ArduinoOTA lets run my raido more stable, but i have some other errors too.

NonaSuomy, 2017-10-24:
Yup, I said I was going to get the stacktracer going later, was just showing that it got rid of the first error but still had other issues. 

This is the last stack trace...

D: StreamTitle='Klingande - Pumped Up';
D: Publish to topic espradio/icy/streamtitle : Klingande - Pumped Up
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40085eb1  PS      : 0x00060033  A0      : 0x8008475a  A1      : 0x3ffc05c0  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x007fe2b3  A5      : 0x00000002  
A6      : 0x00060a20  A7      : 0x3ffd4de0  A8      : 0x3ffc4cf8  A9      : 0x3ffc05a0  
A10     : 0x3ffc4b2c  A11     : 0x00000001  A12     : 0x00000005  A13     : 0x00000020  
A14     : 0x00000020  A15     : 0x3ffd8b90  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40085eb1:0x3ffc05c0 0x40084757:0x3ffc05e0 0x40086a2d:0x3ffc0600 0x4008272e:0x3ffc0610 0x400ea80f:0x00000000

Decoding 5 results
0x40085eb1: xTaskIncrementTick at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4482
0x40084757: xPortSysTickHandler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./port.c line 341
0x40086a2d: _frxt_timer_int at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portasm.S line 296
0x4008272e: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./xtensa_vectors.S line 1095
0x400ea80f: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 55

Try only putting ihr/CIMX streams in overnight and let me know if you still have the same success.

Heres one before that last one...

D: Publish to topic espradio/icy/streamtitle : Alice in Chains - Rooster
D: Publish to topic espradio/ip : 10.13.37.177
Guru Meditation Error of type StoreProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x40085876  PS      : 0x00060e33  A0      : 0x80087130  A1      : 0x3ffd8a00  
A2      : 0xfefefefe  A3      : 0x00000010  A4      : 0x0000000f  A5      : 0x00000001  
A6      : 0x00060820  A7      : 0x00000000  A8      : 0x0000abab  A9      : 0x0000abab  
A10     : 0xb33fffff  A11     : 0x0000cdcd  A12     : 0x40083034  A13     : 0x00000001  
A14     : 0x00060e20  A15     : 0x3ffc4054  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xfefefefe  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x40085876:0x3ffd8a00 0x4008712d:0x3ffd8a20 0x400876c1:0x3ffd8a40 0x40082997:0x3ffd8a60 0x40082a95:0x3ffd8a80 0x40082b4e:0x3ffd8aa0 0x4008303d:0x3ffd8ac0 0x4000bedd:0x3ffd8ae0 0x400dc15e:0x3ffd8b00 0x400dc1ac:0x3ffd8b20 0x400dc1cd:0x3ffd8b40 0x400dc207:0x3ffd8b60 0x400d42e3:0x3ffd8b80 0x400d490f:0x3ffd8be0 0x400d5949:0x3ffd8c00 0x400d6817:0x3ffd8c60 0x4012d120:0x3ffd8ca0

Rebooting...
ets Jun  8 2016 00:22:57

Decoding 17 results
0x40085876: uxPortCompareSet at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4482
:  (inlined by) vPortCPUAcquireMutexIntsDisabled at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portmux_impl.h line 86
:  (inlined by) vTaskEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4126
0x4008712d: multi_heap_malloc_impl at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap.c line 615 (discriminator 1)
0x400876c1: multi_heap_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap_poisoning.c line 270
0x40082997: heap_caps_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c line 143
0x40082a95: heap_caps_realloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c line 143
0x40082b4e: heap_caps_realloc_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c line 143
0x4008303d: _realloc_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./syscalls.c line 47
0x400dc15e: String::changeBuffer(unsigned int) at /home/nonasuomy/Arduino/hardware/espressif/esp32/cores/esp32/WString.cpp line 840 (discriminator 2)
0x400dc1ac: String::reserve(unsigned int) at /home/nonasuomy/Arduino/hardware/espressif/esp32/cores/esp32/WString.cpp line 840 (discriminator 2)
0x400dc1cd: String::copy(char const*, unsigned int) at /home/nonasuomy/Arduino/hardware/espressif/esp32/cores/esp32/WString.cpp line 840 (discriminator 2)
0x400dc207: String::String(char const*) at /home/nonasuomy/Arduino/hardware/espressif/esp32/cores/esp32/WString.cpp line 840 (discriminator 2)
0x400d42e3: handlebyte(unsigned char, bool) at /home/nonasuomy/Downloads/ESP32-Radio-master/Esp32_radio/Esp32_radio.ino line 3753
0x400d490f: handlebyte_ch(unsigned char, bool) at /home/nonasuomy/Downloads/ESP32-Radio-master/Esp32_radio/Esp32_radio.ino line 3732
0x400d5949: mp3loop() at /home/nonasuomy/Downloads/ESP32-Radio-master/Esp32_radio/Esp32_radio.ino line 3494
0x400d6817: loop() at /home/nonasuomy/Downloads/ESP32-Radio-master/Esp32_radio/Esp32_radio.ino line 3601
0x4012d120: loopTask(void*) at /home/nonasuomy/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 18 (discriminator 1)

derpicknicker1, 2017-10-24:
@NonaSuomy The second error you posted where the issue comes from `loop()->mp3loop()->handlebyte()` is my main cause of errors. I have not figured out whats going on until now.
I even don't know how to reproduce it. Sometimes it occurrs every once in 5 minutes. Now my radio plays for +48h without an error without changing anything.

NonaSuomy, 2017-10-24:
Can you try to only put ihr/CIMXFM streams in overnight and let me know if you still have the same success?

Edzelf, 2017-10-25:
I will do a test today for "ihr/IHR_IEDM".  "ihr/CIMXFM" does not seem to work.
Sofar it it running for 4.5 hours without problems.

derpicknicker1, 2017-10-25:
ihr/IHR_IEDM is running for 12hrs now...

NonaSuomy, 2017-10-26:
It seems the physical radio stations (XXXXFM) are more prone to drops than the IHR_XXXX streams :/ I eventually left it running so long where the glitches stop causing the board to output any backtrace comments very strange.

Edzelf, 2017-10-26:
Isn't ihr/IHR_IEDM a physical station?

NonaSuomy, 2017-10-27:
IHR_XXXX are only avail streaming online to everybody, the XXXXFM are probably geo location locked specific to the radio stations in your location as to why CIMXFM won’t load for you :(