Edzelf/ESP32-Radio #56: Unable to connect to the Wifi

jollepe, 2017-11-04:
Hey everyone,

I want to setup this nice project on my ESP32 wroom controller.
Actually I just only have the controller itself without any peripherials connected.
I wanted to test the webinterface.
But I'm not able to open the webinterface. I'm connected to the SSID ESP32Radio.
That is the message I get in chrome browser (sorry for german translation)
"Ihre Verbindung wurde unterbrochen

Eine Netzwerkänderung ist aufgetreten.
ERR_NETWORK_CHANGED"

I also typed in the ssid and password of my desired wlan into the defaultprefs.h.
This is an overview of the log:

`D: Connect to WiFi
D: WiFi Failed!  Trying to setup AP with name ESP32Radio and password ESP32Radio.
D: IP = 192.168.4.1
D: Start server for commands
D: Rotary encoder is disabled (-1/-1/-1)
Guru Meditation Error of type InstrFetchProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x00000000  PS      : 0x00060830  A0      : 0x800de7da  A1      : 0x3ffd8100  
A2      : 0x3ffc4e4c  A3      : 0x3f4016f3  A4      : 0x00000000  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800de6da  A9      : 0x3ffd80e0  
A10     : 0x3ffc4e4c  A11     : 0x00000001  A12     : 0x3ffd2ed8  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x00000000:0x3ffd8100 0x400de7d7:0x3ffd8120 0x400d529e:0x3ffd8150 0x400d696f:0x3ffd8170 0x4012d40c:0x3ffd81b0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14

D: Starting ESP32-radio Version Wed, 30 Aug 2017 08:30:00 GMT...  Free memory 166544
D: ir_pin pin set to -1
D: enc_clk pin set to -1
D: enc_dt pin set to -1
D: enc_sw pin set to -1
D: tft_cs pin set to -1
D: tft_dc pin set to -1
D: sd_cs pin set to -1
D: vs_cs pin set to -1
D: vs_dcs pin set to -1
D: vs_dreq pin set to -1
D: spi_sck pin set to -1
D: spi_miso pin set to -1
D: spi_mosi pin set to -1
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is LOW, probably no PULL-UP
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: SD Card Mount Failed!
D: Scan Networks
D: Scan completed
D: Number of available networks: 1
D:  1 - Futtertrog                Signal: -45 dBm, Encryption WPA_WPA2_PSK, 
D: End of list
D: VS1053 not properly installed!
D: Fast SPI, Testing VS1053 read/write registers again...
D: VS1053 error retry SB:0003 R1:0000 R2:0000
D: VS1053 error retry SB:0006 R1:0000 R2:0000
D: VS1053 error retry SB:0009 R1:0000 R2:0000
D: VS1053 error retry SB:000C R1:0000 R2:0000
D: VS1053 error retry SB:000F R1:0000 R2:0000
D: VS1053 error retry SB:0012 R1:0000 R2:0000
D: VS1053 error retry SB:0015 R1:0000 R2:0000
D: VS1053 error retry SB:0018 R1:0000 R2:0000
D: VS1053 error retry SB:001B R1:0000 R2:0000
D: VS1053 error retry SB:001E R1:0000 R2:0000
D: VS1053 error retry SB:0021 R1:0000 R2:0000
D: VS1053 error retry SB:0024 R1:0000 R2:0000
D: VS1053 error retry SB:0027 R1:0000 R2:0000
D: VS1053 error retry SB:002A R1:0000 R2:0000
D: VS1053 error retry SB:002D R1:0000 R2:0000
D: VS1053 error retry SB:0030 R1:0000 R2:0000
D: VS1053 error retry SB:0033 R1:0000 R2:0000
D: VS1053 error retry SB:0036 R1:0000 R2:0000
D: VS1053 error retry SB:0039 R1:0000 R2:0000
D: VS1053 error retry SB:003C R1:0000 R2:0000
D: endFillByte is 0
D: Connect to WiFi
D: WiFi Failed!  Trying to setup AP with name ESP32Radio and password ESP32Radio.
D: IP = 192.168.4.1
`

koskee, 2017-11-04:
You'll need to either log into the temporary WiFi hotspot and edit the configuration file to include your WiFi SSID/password, or run the radio_init sketch to upload the information to be stored in the NVS/preferences. Changing the defaultprefs.h file won't actually commit the changes to the NVS. 

I think you could also possibly just restore the edited default preferences with the 'default' button on the web interface, but if you haven't been able to pull up the web interface then that won't be possible.
Remember to use the correct ip address to pull up the Interface, it would be displayed in the console as well as the TFT screen if you have one. If you still can't find the correct ip, the other possibility would be to download a lan/wifi host scanner app/program and use that to search your lan for the correct IP.

The simplest way to get up and running is to use the radio_init.ino sketch (in the tools folder. Don't forget to edit the file appropriately first)

jollepe, 2017-11-04:
That was my way to get things running:
1. Start the radio_init and edited the wifi settings (e.g.):
 preferences.putString ( "wifi_00", "MYSSID/MYPASSWORD" ) |
  preferences.putString ( "wifi_01", "ssid2/pw2" ) |

Is this the correct syntax?

After that I uploaded this script to the esp32.
After that I started and uploaded the main esp32radio script.
But in the end the controller is not able to login in the wifi, insted I get the SSID Esp32Radio.
I get it working to login into this wifi connection, but I was not able to start the webvisualization.

jollepe, 2017-11-04:
Now I get it working to connect the Controller to my wifi network.
But actually I dont get the access to the webvisualization.
The IP adresse (found out in my fritzbox) is 192.168.1.48

koskee, 2017-11-04:
Umm.. Looks okay, just remember to end each statement with a semicolon-> ' ; '  instead of pipe ' | ' 

So you're saying it is connecting to the WiFi network or isn't?  I couldn't decide which you meant, but if it still isn't after putting the correct credentials into NVS, that should be all you need to do.
Perhaps your router is employing Mac address filtering or could be out of range of the WiFi, or maybe you need to double check the credentials for an error? 

Otherwise I'm not really sure what to suggest. Maybe Ed can offer some insight that I haven't thought of..

jollepe, 2017-11-04:
[image]

Here we can see, that the controller has logged in successfully.
I try to ping this IP adress (again german.. but means timeout):

Ping wird ausgeführt für 192.168.1.48 mit 32 Bytes Daten:
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.

Serial output of the loop:

`
D: Start server for commands
D: Rotary encoder is disabled (-1/-1/-1)
Guru Meditation Error of type InstrFetchProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x00000000  PS      : 0x00060230  A0      : 0x800de7da  A1      : 0x3ffd8070  
A2      : 0x3ffc4e4c  A3      : 0x00000000  A4      : 0x800846dc  A5      : 0x3ffde010  
A6      : 0x00000000  A7      : 0x3ffd61c0  A8      : 0x800de6da  A9      : 0x3ffd8050  
A10     : 0x3ffc4e4c  A11     : 0x00000001  A12     : 0x3ffd5888  A13     : 0x00000020  
A14     : 0x00000020  A15     : 0xff000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x00000000:0x3ffd8070 0x400de7d7:0x3ffd8090 0x400d529e:0x3ffd80c0 0x400d696f:0x3ffd80e0 0x4012d40c:0x3ffd8120

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14

D: Starting ESP32-radio Version Wed, 30 Aug 2017 08:30:00 GMT...  Free memory 166400
D: ir_pin pin set to -1
D: enc_clk pin set to -1
D: enc_dt pin set to -1
D: enc_sw pin set to -1
D: tft_cs pin set to -1
D: tft_dc pin set to -1
D: sd_cs pin set to -1
D: vs_cs pin set to -1
D: vs_dcs pin set to -1
D: vs_dreq pin set to -1
D: spi_sck pin set to -1
D: spi_miso pin set to -1
D: spi_mosi pin set to -1
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is LOW, probably no PULL-UP
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: SD Card Mount Failed!
D: Added Futtertrog to list of networks
D: Added ssid2 to list of networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 1
D:  1 - Futtertrog                Signal: -44 dBm, Encryption WPA_WPA2_PSK, Acceptable
D: End of list
D: Command: mqttbroker with parameter none
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: mqttpasswd with parameter ******
D: Command: volume with parameter 72
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: preset with parameter 6
D: Command: clk_server with parameter pool.ntp.org
D: Command: clk_offset with parameter 1
D: Command: clk_dst with parameter 1
D: VS1053 not properly installed!
D: Fast SPI, Testing VS1053 read/write registers again...
D: VS1053 error retry SB:0003 R1:0000 R2:0000
D: VS1053 error retry SB:0006 R1:0000 R2:0000
D: VS1053 error retry SB:0009 R1:0000 R2:0000
D: VS1053 error retry SB:000C R1:0000 R2:0000
D: VS1053 error retry SB:000F R1:0000 R2:0000
D: VS1053 error retry SB:0012 R1:0000 R2:0000
D: VS1053 error retry SB:0015 R1:0000 R2:0000
D: VS1053 error retry SB:0018 R1:0000 R2:0000
D: VS1053 error retry SB:001B R1:0000 R2:0000
D: VS1053 error retry SB:001E R1:0000 R2:0000
D: VS1053 error retry SB:0021 R1:0000 R2:0000
D: VS1053 error retry SB:0024 R1:0000 R2:0000
D: VS1053 error retry SB:0027 R1:0000 R2:0000
D: VS1053 error retry SB:002A R1:0000 R2:0000
D: VS1053 error retry SB:002D R1:0000 R2:0000
D: VS1053 error retry SB:0030 R1:0000 R2:0000
D: VS1053 error retry SB:0033 R1:0000 R2:0000
D: VS1053 error retry SB:0036 R1:0000 R2:0000
D: VS1053 error retry SB:0039 R1:0000 R2:0000
D: VS1053 error retry SB:003C R1:0000 R2:0000
D: endFillByte is 0
D: Connect to WiFi
D: Connected to Futtertrog
D: IP = 192.168.1.48

`

koskee, 2017-11-04:
If your wifi credentials are correct and in the NVS  you could try refreshing the page or restarting your browser and /or device but that really shouldn't be necessary I don't think.

jollepe, 2017-11-04:
I see that the controller is rebooting all the time! I read something to ensure a stable power supply.
But doesn't work for me

jollepe, 2017-11-04:
Now I'm able to connect to the webinterface.
THe program struggles with those  lines:
scanserial() ;  

  if ( time_req )                                           // Time has changed?
  {
    gettime() ;                                             // Show time on TFT (if any)
    time_req = false ;
    displayvolume() ;                                       // Show volume on display
  }
  chk_enc() ; 

After deleting, the webserver was accessable

Edzelf, 2017-11-04:
Can you find out what the problem is, scanserial, gettime or displayvolume?

jollepe, 2017-11-04:
The problem was definitely scanserial

naitsimp, 2018-04-04:
Hello,

I think I have got the same problem.
I connect my Lolin32/WEMOS Board to my Computer to upload the code, but the board restarts in a loop. 
I put my SSID and Password in the defaultprefs.h file.
I also tried to connect it to a different PSU after uploading the Code, but nothing happens, the ESP32 doesn't connect to my WIFI.
I am out of ideas what I should do to get this working.
Any help would be appreciated.

The Serial Monitor outputs this code:
D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 05 Mar 2018 10:52:00 GMT.  Free memory 172124
D: Partition nvs found, 20480 bytes
D: Read 0 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is LOW, probably no PULL-UP
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is HIGH
D: GPIO35 is HIGH
D: Create list with acceptable WiFi networks
D: End adding networks
D: Scan Networks
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x40085fc5  PS      : 0x00060033  A0      : 0x4008744b  A1      : 0x3ffd3140  
A2      : 0x00050023  A3      : 0x00000001  A4      : 0x3ffc8314  A5      : 0x3ffc84d4  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000016  A9      : 0x00000002  
A10     : 0x3ffc11c0  A11     : 0x00000000  A12     : 0x3ffc84d4  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000048  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40085fc5:0x3ffd3140 0x40087448:0x3ffd3160 0x400873fe:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58

naitsimp, 2018-04-05:
I tried to run the init sketch and it gives me:
ESP32_radio_init completed...

I also tried to run prefbug. Which gives me this:
Fill NVS
Loop number 1
TKEY_00 = <someting>
TKEY_01 = <someting>
TKEY_02 = <someting>
TKEY_03 = <someting>
TKEY_04 = <someting>
TKEY_05 = <someting>
TKEY_06 = <someting>
TKEY_07 = <someting>
TKEY_08 = <someting>
TKEY_09 = <someting>
TKEY_10 = <someting>
TKEY_11 = <someting>
TKEY_12 = <someting>
TKEY_13 = <someting>
TKEY_14 = <someting>
TKEY_15 = <someting>
TKEY_16 = <someting>
TKEY_17 = <someting>
TKEY_18 = <someting>
TKEY_19 = <someting>
TKEY_20 = <someting>
TKEY_21 = <someting>
TKEY_22 = <someting>
TKEY_23 = <someting>
TKEY_24 = <someting>
TKEY_25 = <someting>
TKEY_26 = <someting>
TKEY_27 = <someting>
TKEY_28 = <someting>
TKEY_29 = <someting>

Edzelf, 2018-04-05:
The output of both prefbuf and radio_init looks good.  The NVS is working.  Now edit your WiFi credentials on line 39 and 40 of radio_init and run it.  Do not run prefbug.  No run the radio and you should see the keys that are present in radio_init.  The radio should connect to WiFi.

naitsimp, 2018-04-05:
I might do something wrong or don`t understand how to do it.
I just open the "Esp32_radio_init.ino" file which is a a folder named "Esp32_radio_init".
I put my Wifi SSID and password in on line 39, line 40 stays as it is.
I upload this sketch to my Board.
A MQTT Broker is not present at the moment.
How do i see the keys? because the Serial Monitor just shows:
`ESP32_radio_init completed...`
It doesn´t show up as a connected device on my Router web interface.

Edzelf, 2018-04-05:
The output of radio-init is correct.  Then you must load and start the radio sketch.  This will show all keys that are also present in radio-init.  The MQTT settinf can be left as it is if you don't use it.

naitsimp, 2018-04-05:
ok, now I uploaded the "Esp32_radio.ino" file to the Board.
But I still have the restart thing.
Serial Monitor outputs this:
```
D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 05 Mar 2018 10:52:00 GMT.  Free memory 172124
D: Partition nvs found, 20480 bytes
D: Read 31 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is LOW, probably no PULL-UP
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is HIGH
D: GPIO35 is HIGH
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Create list with acceptable WiFi networks
D: Added Fanta_7 to list of networks
D: Added ADSL to list of networks
D: End adding networks
D: Scan Networks
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x40085fc5  PS      : 0x00060033  A0      : 0x4008744b  A1      : 0x3ffd3140  
A2      : 0x00050023  A3      : 0x00000001  A4      : 0x3ffc8314  A5      : 0x3ffc84d4  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000016  A9      : 0x00000002  
A10     : 0x3ffc11c0  A11     : 0x00000000  A12     : 0x3ffc84d4  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000048  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40085fc5:0x3ffd3140 0x40087448:0x3ffd3160 0x400873fe:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
```

Edzelf, 2018-04-05:
Looks like WiFi.scanNetworks is giving problems.  Try to make a short sketch that test WiFi.scanNetworks.  Could it be that there many (more than 25) networks in your surrounding?

naitsimp, 2018-04-05:
Thank you. 
I uploaded the WiFiScan sketch from the Examples.
The sketch finds 5-6 Networks around me without any problems.

Edzelf, 2018-04-06:
Sorry, I can't reproduce the problem.  I used get.exe to get the latest core library.
I will put a compiled version in the code (Esp32_radio.ino.bin) section of this repositry.  The load command is (on my configuration):
C:\Users\Ed\Documents\Arduino\hardware\espressif\esp32/tools/esptool.exe --chip esp32 --port COM4 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\Ed\Documents\Arduino\hardware\espressif\esp32/tools/partitions/boot_app0.bin 0x1000 C:\Users\Ed\Documents\Arduino\hardware\espressif\esp32/tools/sdk/bin/bootloader_dio_80m.bin 0x10000 C:\Users\Ed\AppData\Local\Temp\arduino_build_876232/Esp32_radio.ino.bin 0x8000 C:\Users\Ed\AppData\Local\Temp\arduino_build_876232/Esp32_radio.ino.partitions.bin 

This is the list of the libraries I used.
 Used: C:\Users\Ed\Documents\Arduino\hardware\espressif\esp32\libraries\SD
Using library pubsubclient-master at version 2.6 in folder: C:\Users\Ed\Documents\Arduino\libraries\pubsubclient-master 
Using library WiFi at version 1.0 in folder: C:\Users\Ed\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi 
Using library ESPmDNS at version 1.0 in folder: C:\Users\Ed\Documents\Arduino\hardware\espressif\esp32\libraries\ESPmDNS 
Using library TFT_ILI9163C-master at version 0.9 in folder: C:\Users\Ed\Documents\Arduino\libraries\TFT_ILI9163C-master 
Using library Adafruit_GFX_Library at version 1.2.3 in folder: C:\Users\Ed\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library FS at version 1.0 in folder: C:\Users\Ed\Documents\Arduino\hardware\espressif\esp32\libraries\FS 
Using library SD at version 1.0.5 in folder: C:\Users\Ed\Documents\Arduino\hardware\espressif\esp32\libraries\SD 
Using library SPI at version 1.0 in folder: C:\Users\Ed\Documents\Arduino\hardware\espressif\esp32\libraries\SPI 
Using library ArduinoOTA at version 1.0 in folder: C:\Users\Ed\Documents\Arduino\hardware\espressif\esp32\libraries\ArduinoOTA 
Using library Update at version 1.0 in folder: C:\Users\Ed\Documents\Arduino\hardware\espressif\esp32\libraries\Update 
Sketch uses 625114 bytes (47%) of program storage space. Maximum is 1310720 bytes.
Global variables use 60500 bytes (20%) of dynamic memory, leaving 234412 bytes for local variables. Maximum is 294912 bytes.

mrohner, 2018-11-21:
Thanks for this program.
I cannot connect to Wifi and thus not start the Web interface. The VS1053 is not installed yet.
Wifi credentials are defined in the ESP32_radio_init program.
AP address is what exactly? 192.168.4.1?? Where does that come from? I cannot connect to this address.
"D: STOP requested" I did not request STOP. I wouldn't know how, where does it come from??
Where to define the pins for the VS1053? Obviously the defaultprefs.h does nothing. What is defaultprefs.h used for when it has no effect??

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 04 Oct 2018 07:22:32 GMT.  Free memory 151784
D: Display type is OLED
D: Partition nvs found, 20480 bytes
D: Read 30 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to 22
D: pin_tft_sda set to 21
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to -1
D: pin_vs_dcs set to -1
D: pin_vs_dreq set to -1
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is LOW, probably no PULL-UP
D: GPIO26 is LOW, probably no PULL-UP
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Start display
D: Init SSD1306, I2C pins 21,22
D: Create list with acceptable WiFi networks
D: Added mmwireless to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Couldn't get a wifi connection
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: mqttbroker with parameter 192.168.178.59
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 8883
D: Command: mqtttpreset with parameter none
D: Command: mqttuser with parameter mqtt
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Connect to WiFi
D: Try WiFi mmwireless
D: WiFi Failed!  Trying to setup AP with name ESP32Radio and password ESP32Radio.
D: IP = 192.168.4.1
D: Start server for commands
D: Rotary encoder is enabled
D: STOP requested

Edzelf, 2018-11-21:
"STOP" in the logging is normal during startup.
Defaultprefs are used in the webinterfac, page "config".
It looks you radio does not see any WiFi network, otherwise the should be listed in the logging.
For AP you need to connect to the WiFi network supplied by the radio. Then access 192.168.4.1 for the webinterface.