Edzelf/ESP32-Radio #57: Problems with higher Bitrates

zapf2000, 2017-11-05:
Hi, first of all many thanks for this great work. I started with building a ESP8266 Radio. All works fine. But then I got the ESP32 board and gave it a try. Everything works (IR, SD card playing, ILI9341 display) but at higher bitrates > 160KBit (MP3 files and also webradio) it begins to stuck. I have heavy dropouts and a horrible sound. Has anybody an idea what that could be ? Perhaps it's my chaotic cable connection ? Below 160KBits all is fine and sound is clear.
[image]

Edzelf, 2017-11-05:
Can you give me the URL of the station that will demonstrate this problem?

zapf2000, 2017-11-06:
Yes, for example the 192KBit one from the default settings: airspectrum.cdnstream1.com:8000/1261_192

svins01, 2017-11-10:
I have the same problem
mp3 with 320 bitrate, stuttering
radio with 320 bitrates, stuttering
Edzelf can I ask you to release a light version

lecatla, 2017-11-10:
Never had this problem with karadio

koskee, 2017-11-10:
@Edzelf have you considered utilizing the freeRTOS to take advantage of the second core? I'm fairly sure the application is only using one of the two cores available, since it is essentially a sequential program. Utilizing tasks would likely speed up execution (& enable parallel processing, allowing the radio to output data to the mp3 codec at the _same_time_ as responding to the web interface/updating the tft) The changes would be quite minimal as per my initial research on the subject, since you already have the software split up in a compatible way. A few mutexes/semaphores to prevent concurrent access of the ring buffer, and you could essentially split off at least one task to handle the mp3 data without much effort at all.. Just a suggestion because I was considering implementing this myself, but it seems this project would stand to benefit anyway

Edzelf, 2017-11-11:
I tried airspectrum.cdnstream1.com:8000/1261_192.  Perfect sound, but after some minutes it starts to stutter.  By pressing the "Test"-button in the web interface, you can see that the number of bytes in the stream is too low.  Normally you will see something like:
"Free memory is 117960, ringbuf 19736, stream 2768", but sometimes the number of bytes in the stream will go to zero:
"Free memory is 122296, ringbuf 0, stream 0"
That means that there is no data in the stream.
I put a 320 kbps mp3 file on a remote server and it played without problems.  So I think the processing speed is fast enough for streams up to 320 kbps.  I raised the ringbuffer to 40 kByte.
Furthermore, the measured bitrate will be printed on teh "test"-command.
I still hear some disruptions while receiving the metadata.  I will check if this is a program error..

svins01, 2017-11-12:
applied a new firmware
I noticed a feature sometimes when playing MP3 tracks with 320 bitrates, ringbuf is always equal to 0, in this case I go to constant friezes.
Sorry for bad english, I use a translator :)

LordVanguard, 2017-11-14:
There is a lot of space still available on the TFT when playing a radio station even with all the track and radio information displayed. Would there be any benefit in having an option to allow users to have the display also show the current buffer capacity and stream rate. For example the screen readout could be something as simple as :- 

"19736-B / 2768-S"

It could be placed just above the volume bar and could be updated about once a second and provide a visual, almost real time, indication of how things are going without having to go to the web interface ?

Edzelf, 2017-11-27:
The 27-nov-2017 version will play streams up to 256 kbps without stuttering.

zapf2000, 2017-12-02:
Yes, now everything works very fine :)
Thanks a lot Ed.

udavst, 2017-12-31:
Where can I get the version 27-nov-2017? After all it is laid out from 28-jul-2017 (in README.md).

Edzelf, 2017-12-31:
beloeradio.ru:8080/whiteradio.mp3 plays very well here.
Free memory is 58444, chunks in queue 388, stream 4308, bitrate 192 kbps.
Check power supply and Internet connection.

udavst, 2017-12-31:
I, during freezes - Free memory is 95928, chunks in queue 0, stream 0, bitrate 193 kbps when playing - Free memory is 90312, chunks in queue 393, stream 4308, bitrate 193 kbps. There are no problems with the Internet, two more radio stations play in parallel, they are not interrupted and the channel is used by 20%. I'll try to replace the power, for sure ESP32 requires more than ESP. Thank you.

freey99a, 2018-02-01:
I also had big problems with the sound reproduction on the Internet radio strongly dropouts, the MP3 player well functions.
The settings of my router must be changed,  after I have moved the WIFI on 802.11B the massive dropouts were away.
I have found a web page where is described like the settings can be changed in the ESP. 
Maybe this helps in the mistake search:
https://www.reddit.com/r/esp8266/comments/5inhtw/how_to_change_radio_type_from_80211g_to_80211n/