Edzelf/ESP32-Radio #58: HTTPS/TLS handling?

koskee, 2017-11-09:
I'm having trouble accessing this stream: 
[stream.subcity.org:443/subcity.mp3](stream.subcity.org:443/subcity.mp3)

It seems the initial response contains a Location: redirect, but it redirects to https://... which looks like it isn't implemented. There are a whole bunch of examples online of how to use the TLS protocol to access these encrypted resources, so I was just wondering if you had any plans to include this? 

It doesn't seem to be all that common, but I think it would be a worthwhile goal to have the radio be able to access (almost) any resource.

Edzelf, 2017-11-13:
I've tried to connect to stream.subcity.org:443 through WiFiClientSecure, but the connect fails.
I don't understand why.  From the logging:

[I][ssl_client.cpp:153] start_ssl_client(): Performing the SSL/TLS handshake...
[E][ssl_client.cpp:25] handle_error(): SSL - A fatal alert message was received from our peer
[E][ssl_client.cpp:27] handle_error(): MbedTLS message code: -30592
[E][WiFiClientSecure.cpp:101] connect(): lwip_connect_r: 11
[I][ssl_client.cpp:194] stop_ssl_socket(): Cleaning SSL connection.
Connection failed!

koskee, 2017-11-13:
Hmm. In most of the examples I've seen they have a lengthy block of cyphertext (which would be the certificate/key I presume).. I'm guessing your sketch likely has this as well.

AFAIK, there are key generating services/Web applications online that you can acquire a new key, might be worth a shot. I'm guessing some sites might have the default key flagged as insecure since it is shared so widely and freely.

Regrettably my understanding of the whole mechanism behind the protocol that is modern day TLS is pretty basic/limited at best. I'm pretty sure that I don't understand it well enough to effectively contribute to troubleshooting with you. 

The error message doesn't seem to offer much by way of specifics,  neither.

koskee, 2017-11-13:
Don't stress about figuring it out if you don't have time, like i said, it's seemingly rare for stations to use encryption from what I've seen. If you need to come back to it later on, I'm in no rush.