Edzelf/ESP32-Radio #60: Can' compile: some objects aren't declared in this scope

daUser, 2017-11-10:
Hi,
i would like to try this ESP32-Radio, but i have some issues. The Arduiono-IDE throws some Errors while compiling:

```
D:\Eigene Dateien\Eigene Dokumente\Arduino\Esp32_radio\Esp32_radio.ino: In function 'esp_err_t nvssetstr(const char*, String)':

Esp32_radio:974: error: 'nvssearch' was not declared in this scope

   if ( nvssearch ( key ) )                                 // Already in nvs?

                        ^

D:\Eigene Dateien\Eigene Dokumente\Arduino\Esp32_radio\Esp32_radio.ino: In function 'int listsdtracks(const char*, int, bool)':

Esp32_radio:1365: error: 'mp3loop' was not declared in this scope

       mp3loop() ;                                       // Keep playing

               ^

D:\Eigene Dateien\Eigene Dokumente\Arduino\Esp32_radio\Esp32_radio.ino: In function 'bool connectwifi()':

Esp32_radio:1983: error: 'tftlog' was not declared in this scope

     tftlog ( pfs2 ) ;

                   ^

D:\Eigene Dateien\Eigene Dokumente\Arduino\Esp32_radio\Esp32_radio.ino: In function 'String readprefs(bool)':

Esp32_radio:2315: error: 'mp3loop' was not declared in this scope

           mp3loop() ;                                       // Yes, keep playing

                   ^

exit status 1
'nvssearch' was not declared in this scope
```

I had some problems with the libraries I could solve, but here I am stuck. Can sombody help me?

Thanks!

koskee, 2017-11-13:
Since all of those functions are in the main file (read: the compiler should be able to find them) most of the time I've seen this error it turns out to be a mismatched number of curly braces '{' or '}' - try reloading the sketch as it appears on Github (it is possible that you may have inadvertently deleted or added a bracket somewhere).

If you are sure that you haven't changed the file, then another possibility (that I have run into before on this project) is that you may need to add a function prototype for the functions that the compiler is complaining about at the top of the file (search for "prototype" and it should take you to the appropriate section, and you can use the prototypes that are already there as an example to write the new ones that you require. It's pretty simple to do). 

You can also move the function definitions to an earlier part of the file (must be before the first function call). 

The arduino environment is supposed to automatically generate these prototypes, but in my experience with the espressif chips, it doesn't always do this properly.

Edzelf, 2017-11-13:
@daUser.  Can you give me a full list of the "was not declared in this scope"-messages?  I will declare this as "forward" in the source code.  In my case, the compiler generates the prototype automatically, so it's hard to find the missing forward declarations.  I uploaded a new version with some extra forward declarations.

daUser, 2017-11-15:
@koskee 
Thank you, i will try that later, time is short.

@Edzelf 
The Log in the start post is the full list that the arduino-IDE gives me.

I have try it now with Microsoft Visual Studio & Visual Micro. There are other Errors:

```
Compiling 'Esp32_radio' for 'ESP32 Dev Module'
 
Sd2Card.h:26: In file included from
SdFat.h:29: from
SD.h:20: from
File.cpp:15: from
 
Sd2PinMap.h: 510:2: error: #error Architecture or board not supported
   #error Architecture or board not supported
 
Sd2Card.h:26: In file included from
SdFat.h:29: from
SD.h:20: from
SD.cpp:53: from
 
Sd2PinMap.h: 510:2: error: #error Architecture or board not supported
   #error Architecture or board not supported
 
Sd2Card.h:26: In file included from
Sd2Card.cpp:22: from
 
Sd2PinMap.h: 510:2: error: #error Architecture or board not supported
   #error Architecture or board not supported
 
Sd2Card.h:26: In file included from
SdFat.h:29: from
SdFile.cpp:20: from
 
Sd2PinMap.h: 510:2: error: #error Architecture or board not supported
   #error Architecture or board not supported
 
Sd2Card.h:26: In file included from
SdFat.h:29: from
SdVolume.cpp:20: from
 
Sd2PinMap.h: 510:2: error: #error Architecture or board not supported
   #error Architecture or board not supported
Error compiling libraries
Build failed for project 'Esp32_radio'
```

I don't need SD-Card-Support or the TFT.
The documentationPDF says, i need the TFT- and Adafruit-Libraries only when configurated. How can i disable that?

Edzelf, 2017-11-15:
You may remove all the stuff that the SD card is using, but it is easier to include the libraries.

daUser, 2017-11-16:
@Edzelf they are included. I think thats an Error of VisualMicro. I used your newest Esp32_radio.ino with the ArduinoIDE and it works.

Thank You!