Edzelf/ESP32-Radio #61: Sound to Fast and to High

BorisJB, 2017-11-14:
Hi! Thanks for this great piece of Software. It helps a lot for my project of a personalized Alarm-Radio-Clock.

But I have an issue with the decoding of the mp3 stream.
The Sound is too fast and too high. And because of the too fast playback there are gaps.
This happens with all the radio stations.
I'm using the "Sparkfun ESP32 thing" and the 
"VLSI VS1003B MP3 Decoder Module DevelopmentVLSI VS1003B MP3 Decoder Module Development" with 
"Built-in VS1053 Decoder Chip" from here https://eckstein-shop.de/VLSI-VS1003B-MP3-Decoder-Module-Development-Board-Arduino-AVR-PIC
I conncected the 8 connection of the MP3 Decoder as described in the script.
And I connected the RST pin to the RST pin of the ESP32.

It would be great if anybody has an idea to solve the problem.

Thanks

Boris

Edzelf, 2017-11-14:
Can you tell me the bitrate you are using.  I know that the current version of the radio does not handle bitrates higher than 192 kbps well.  I am working on a solution.

BorisJB, 2017-11-15:
Hi! I tried several streams with different bitrates, example here 56 and 192, problem occurs with both. The bitrate is given by the website, right? Or can I change it? Here are the debug outputs:

D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Connected to server
D: Switch to HEADER
D: Headerline: Server: Icecast 2.4.2
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: Cache-Control: no-cache
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: Pragma: no-cache
D: Headerline: icy-br:192
D: Headerline: ice-audio-info: samplerate=48000;channels=2;bitrate=192
D: Headerline: icy-br:192
D: Headerline: icy-genre:Talk
D: Headerline: icy-metadata:1
D: Headerline: icy-name:NPO Radio1
D: Headerline: icy-pub:0
D: Headerline: icy-url:http://www.radio1.nl
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 192, metaint is 16000
D: First chunk:
D: EA D6 F2 E3 7B FF FF FF
D: FF FD 6B FF 8F FE BD B1
D: A8 FF 7A 8F 6C E9 D8 01
D: 38 32 CD 34 45 33 25 1C
D: Metadata block 64 bytes
D: Streamtitle found, 55 bytes
D: StreamTitle='NPO Radio 1 - NOS-Radio 1-Journaal - NOS';

D: Connect to new host mdr-284310-0.cast.mdr.de/mdr/284310/0/mp3/low/stream.mp3
D: Connect to mdr-284310-0.cast.mdr.de on port 80, extension /mdr/284310/0/mp3/low/stream.mp3
D: Connected to server
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:56
D: Headerline: icy-description:MDR KULTUR - Das Radio
D: Headerline: icy-genre:culture
D: Headerline: icy-name:MDR KULTUR
D: Headerline: icy-pub:0
D: Headerline: icy-url:http://www.mdr.de/radio
D: Headerline: Server: dg-mdr-http_dus-dtag_edge_e7c8b7f5467f2df30818790db6cd94a9
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: Close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 56, metaint is 16000

Thanks!

Edzelf, 2017-11-15:
The bitrate can not be changed for a station.  The radio should handle bitrates up to 192 kbps without problems.  Are you sure there is a **VS1053B** chip on your board?  A VS1003B chip is not good enough.

BorisJB, 2017-11-15:
On the board it says "Vs1003/1053" and also "Built-in VS1053 Decoder Chip"  in the description.
The (very small) text on the cips says VS1003B.
What board would you recommend or worked for you?

Edzelf, 2017-11-15:
Yes, there are some fake VS1053 boards on the market.
I use [this](https://www.aliexpress.com/item/1PCS-New-VS1053-MP3-Module-Development-Bard-on-board-recording-function/32354143509.html?spm=a2g0s.9042311.0.0.vJ04b7) one.

BorisJB, 2017-11-15:
That looks exactly like mine, which I got from a good store here.

BorisJB, 2017-11-16:
Ok, Thank You. I probably order a new one.

koskee, 2017-11-16:
Issue previously discussed here : #11