Edzelf/ESP32-Radio #62: Plays stream but doesn't save it in the Config

daUser, 2017-11-16:
Hi,
i have some streams of the German Radio "Radio Bob!". They are like this:
`http://bob.hoerradar.de/radiobob-live-mp3-hq?sABC=5n0qo9s3%230%230pq7n9286r599r9589o9pqs9763o1o32%23fgernzf.enqvbobo.qr&amsparams=playerid:streams.radiobob.de;skey:1510849011`

ESP32Radio plays this streams over the Webinterface ("Control" => "Enter a station/file here..." => "Play"). I go to the "config", add this stream like:
`preset_00 = http://bob.hoerradar.de/radiobob-live-mp3-hq?sABC=5n0qo9s3%230%230pq7n9286r599r9589o9pqs9763o1o32%23fgernzf.enqvbobo.qr&amsparams=playerid:streams.radiobob.de;skey:1510849011 # 0 - Radio Bob`
then I save this and go to the control. When I go back to "Config" this line is deleted. Other lines that have been changed at the same time are there.

I also have this URL: http://streams.radiobob.de/bob-live/mp3-192/mediaplayer but ESP32Radio doesn't play this one.

Edzelf, 2017-11-17:
Problem 1:
The maximal size of a string that can be stored in NVS is defied on line 147 (NVBUFSIZE, value 150).
You may change this number if you want to store longer strings.
Problem 2:
The ":" in the URL is not handled correctly by the current software.  You may change line 1881:
from:
  inx = host.indexOf ( ":" ) ;                           // Search for separator
To:
  inx = hostwoext.indexOf ( ":" ) ;                 // Search for separator

I will correct this in the next release.

daUser, 2017-11-24:
Hi,
that works!
Thank you!