Edzelf/ESP32-Radio #63: Two rotary encoder

cypeks, 2017-11-25:
Is it possible to add a second rotary encoder? As in the traditional radio: one for volume control, the other for radio change, etc ...

Edzelf, 2017-11-25:
Yes, you need 2 free input pins.  And you need to change the software.
It's easier to use a potentiometer instead of a rotary switch for the volume.

cypeks, 2017-11-25:
Can you tell me how to adjust the volume in the program? Using the vs1053player.setVolume() object?

Edzelf, 2017-11-26:
Just change ini_block.reqvol to a value between 0 and 100.

Edzelf, 2017-11-26:
I just realized that you may connect 2 second rotary switch for just the volume in parallel of the first.  If you do not connect the SW-pin, the function of this switch will be limited to "volume" only.
No need to change the software and no extra I/O pins required.

cypeks, 2017-11-26:
Not exactly good because I need to connect two rotary encoders. One to adjust the volume, the other to change the station.
I'll try to add enc2 support.
Where in the code are detected connected accessories? - I can not find it.

Edzelf, 2017-11-26:
Yes, that's what I meant.  One for volume, one for the rest.

DaniilMir, 2018-03-04:
Hello All,
I started implementation of two encoders.
Simply cloned ISR Routine and chk_enc routine. Also cloned all encoder variables.
But there is an issue with preset switching. Original routine has a hard-coded timeout : 
"case PRESET:
	{	waittime = 10;"
because of this timeout preset change works really slow comparing to the web interface command.
without timeout it works pretty fast, but I get another bug - encoder can skip presets.
It would be great to have a flag which shows that preset is already in switch process.
i tried : "if (datamode == STOPREQD) return; "
but it does not help. 

Even web interface command can switch two presets down without waiting for connecting to stream.
how can I identify that connecting process is already finished to allow second preset change command?

Edzelf, 2018-03-04:
Yes, the rotary switch function is not very well implemented.  I will redesign it.

koskee, 2018-03-04:
Here is a pretty good article I came across today in my inbox.. It shows only three shares, so I'm guessing it's new(ish). Although I haven't tried it, you might try this method out because I have found my encoder to be quite problematic (when using it, it will do any of the following: skip multiple increments, move in the wrong direction, or sometimes cause a crash/reboot).

This method sounds good on paper, from the description (the second method described) but as I haven't experimented with it, I do not know for sure. You might want to give it a try, since it would be as simple as copy/paste and could save you a bunch of time if it works well. 

 http://www.best-microcontroller-projects.com/rotary-encoder.html

Edzelf, 2018-03-04:
That's exactly the method being used in the radio.  The difficulty is how to handle the results of the movements.

DaniilMir, 2018-03-04:
best option is to split all functionality 
first: split encoder rotation and encoder clicks
second: spit encoder rotation and commands processing.

in current case we have scope to change volume and preset.
We have commands like  
"upvolume" -  "downvolume"  
"uppreset" - "downpreset"

on rotate action we can send this command and wait for a flag say"command process on"
till flag is on - ignore all other rotations. 
from encoder rotation process we expect direction only (even in current implementation)
flag processing can work even with existing setup with one encoder.  
we can bypass this flag with define option.

koskee, 2018-03-04:
Well yes, it is similar in its use of tables, however I don't see any sort of denounce filter as he has described. 

I think that many of the issues (that I'm running into, anyway) could logically be attributed to contact bounce.. Either way, I'll try it when I  have a moment.

Edzelf, 2018-03-05:
No need for debounce if a state machine is used to detect rotation.  I've improved the interface in the new version.  Added some debug output to show the detected clicks and rotation.

dl6dbh, 2020-04-21:
Hallo  DaniilMir ,

have you finish the  implementation of two encoders? 
i am interested for an test 

Meinolf

greppp, 2020-09-09:
> I just realized that you may connect 2 second rotary switch for just the volume in parallel of the first. If you do not connect the SW-pin, the function of this switch will be limited to "volume" only.
> No need to change the software and no extra I/O pins required.

Hi @Edzelf Edzelf, Could you elaborate a bit more on this? Would like to implement it. Thanks!

Edzelf, 2020-09-09:
Plus, ground and the two rotation signals of the sevond encoder in parallel of the first.  What can I say more?

greppp, 2020-09-12:
@Edzelf Thanks for your answer. Let's call "**encoder A**" the one that doesn't have the SW-pin connected and "**encoder B**" the one that has the SW-pin connected.
If I turn encoder A, the volume goes up or down. I get it. If I turn encoder B (knowing the rotations signal are in parallel of the other encoder), how come it will change the stations and not adjust the volume?
Thanks for your time, this is really appreciated.

Edzelf, 2020-09-15:
I just test the radio with 2 encoders.  I put them in full parallel mode (all 5 pins).  It worked as expected.  I could use both encoders to change the volume.

mousemat86, 2021-08-06:
I'm looking add adding a second encoder – encoder A for presets only and encoder B for volume only. 

Could any one explain running them in parallel – it's not clear from the above comments, are the encoders wired to seperate pins? Where would I declare the second encoder in the config? @Edzelf @greppp

Edzelf, 2021-08-06:
Why not use a potentiometer for the volume?
For an extra rotary encoder you need at least 2 free I/O pins.  And some changes in the software.

greppp, 2021-08-06:
@mousemat86 I never finished the project, but if you use a potentiometer for volume control then the added bonuses are that:
- It become more responsive
- Smooth transition from low to high volume (no steps), which will give you a nicer feel.
- No need for extra pins
- Less software change

good luck!