Edzelf/ESP32-Radio #64: ILI9341 2.4'' custom color

techbv, 2017-12-03:
Some changes have been made in the code, colors have changed. Used library Adafruit_ILI9341. Displays the date.

[image]
[Esp32_radio_1.zip](https://github.com/Edzelf/ESP32-Radio/files/1525152/Esp32_radio_1.zip)

ats3788, 2018-05-23:
Hello techbv 
I use your Sketch How do you switch on the TFT LED ?
I can't find any clue in the sketch.

techbv, 2018-05-23:
hello ats3788.
TFT LED is turned on permanently. It is simply connected by a VCC contact
(3.3V).

Feuer001, 2018-05-23:
I want to use a 2.8 " (320x240) TFT display, but I can not modify the code because it's too heavy for me.

Edzelf, 2018-05-23:
I ordered one.  I will write the driver for it.

Feuer001, 2018-05-24:
@Edzelf: Thanks!! You are the best!!

ariehoffman, 2020-02-07:
Where is the full kit in details so we can build it ourselves?
Erik Hoffman