Edzelf/ESP32-Radio #65: MQTT status

eXadra, 2017-12-06:
Before all, congratulations, great project and excellent programming skills.
I’m using this radio with my home automation system using OpenHab, I can turn it on and off, change channels and volume inside Openhab.
Also Openhab starts the Radio at normal go to bed time with a smooth volume and slow music preset and changes it to a more active preset  with higher volume at wake up time :)
My issue is that there is no MQTT status message every time the radio changes from STOP,RESUME, volume change and or selected station, …, this means that if radio setting are changed in the encoder or in the web server, the Openhab gets out of sync.
I just made some changes to accomplish this in your code, but because I don’t know your code extensively they are not in the best rules of programming :(, so I don’t dare to publish :(, anyway just as example:
<topic>/status  <state (on,off)>,<preset radio>,<volume>

Ps: just remembered, my SSID password starts with a !, I inserted the password in the config and when the device starts it hangs, hardcoded the password in the code… must check …

Edzelf, 2017-12-06:
I guess it would be sufficient to publish the status changes to mqtt.
I will see if I can implement this.

Edzelf, 2017-12-08:
The 8 december version has some extra MQTT items.  See the docs.
A password starting with an exclamation mark should be handled normally.

NonaSuomy, 2017-12-08:
@eXadra this is what I did https://github.com/Edzelf/Esp-radio/issues/77