Edzelf/ESP32-Radio #66: Audio Noise (solved)

eXadra, 2017-12-07:
I'm using to amplify the audio an PAM8610 module amplifier bought in eBay, like discussed before, the Vs1053 audio output is not referenced to ground but to GbuF  and it cannot be grounded.
As an alternative we can reference the 2 output lines to ground instead of Gbuf but then we have a lot of digital noise, also we can like advanced here,  use 2 different power supplies but this is not a practical solution.
Yesterday night I remember to use 2 isolation 600:600R transformers and the result is amazing, no digital noise, clear crystal audio :). I connected de audio left and Gbuf directly to the primary of one transformer and audio right and Gbuf to the other transformer, the secondary wiring was connected to ground and amplifier output for each of the transformers.
This module board that I bought on eBay (most cheap and common) doesn’t have volume control and internally the volume is connected for maximum output power, so the audio level is controlled by the ESP. When the audio is on mute or the radio is Stopped mode, we can hear in the speakers low static noise, to solve this I connected the SW (shutdown) of the audio amplifier to one free digital  pin of the ESP32 and in the software I check the variable "datamode" in the main loop, if it is different of 256 then I put the output pin to HIGH, if not, back to LOW. Now if there is no music I have no static noise :).
In the next days I’m going to draw  a motherboard PCB to place all the modules, afterwards I will share It to all.
[image]

Edzelf, 2017-12-07:
Maybe it's better to set/reset the shutdown output in stopSong()  and startSong().  The dataqueue with mp3 data may not be empty is datamode changes to "STOPPED".
I will implement this feature in the software and add a "pin_shutdown = x" to the preferences.

Edzelf, 2017-12-08:
The 8 december version will handle the shutdown.

locki-cz, 2018-02-01:
Nice i have same board PAM8610 from Ali! :) I must bought 600:600R transformers, how did you solved powering ESP32 and this amlifer? How it looks with your PCB?

eXadra, 2018-02-09:
Hi locki-cz
Sorry for the delayed answer :)
3 modules, ESP32, VS1053 and display work at 3.3v  the amplifier works at 12v, ESP32 and the VS1053 have a linear regulator AMS117 that will overheat and wast energy if powered from 12V. Solution, I bought from Ebay an switched 5v regulator that will reduce the power from the 12v to 5 and feed the ESP and the VS. Display is powered by the ESP.

Attacked You have the PCB for the Radio.
Also because I use the ESP32Radio as an AlarmClock (3 working at the moment :) ) the backlight of the display is dimmed when dark. An LDR connected to one analog port of the ESP and a PWM output with some driving transistors power the LCD backlight, if this option is not used just shunt the R2 and don´t install LDR, transistors ans resistors. Please pay attension that are several ESP32 modules with diferent pinouts available, the modules that I used are (just copy past ion Ebay):
*DC-DC Buck Step Down Converter Regulator Power Supply Module 9V 12V 24V to 5V 3A
*ESP-32 ESP32S Development Board 2.4GHz Dual-Mode WiFi+Bluetooth Antenna Module M
*1.8 Inch Mini Serial SPI TFT LCD Module Display PCB Adapter ST7735S For Arduino
*1PCS VS1053 MP3 Module Development Bard (on-board recording function) MO
*PAM8610 Digital Audio Amplifier Board 2*15W Dual Channel Stereo AMP Class D
*10X Audio Signal Transformers 600:600 Ohm Europe 1:1 EI14 Isolation Transformer
*1/2/5PCS Rotary encoder with switch EC11 Audio digital potentiometer 20mm handle
*6*6*(4.3-15mm) Momentary Tactile Push Button Switch Small Mini Micro PCB Mounted
*capacitors, diode connectors, you can find in any local electronics store
(all arrounf 20€ )

If needed please ask and I will send to you all the addon code for the LCD backlight control :)
To finissh the project I just need to buy an acrilic housing that shows all the electronics (steampunk style :) )
[ESP32Radio_pcb.pdf](https://github.com/Edzelf/ESP32-Radio/files/1710907/ESP32Radio_pcb.pdf)
[ESP32radio_silk.pdf](https://github.com/Edzelf/ESP32-Radio/files/1710908/ESP32radio_silk.pdf)
[image]

locki-cz, 2018-02-09:
Nice! I already have all at home also stepdown :D but i need it little bit smaller to hide after standard ABB swith, 
[image]

I must build on ESP32s with some SMDs and ST7735 without board (board si to big :) )
It should look like this , 15W speakers in ceiling :
[image]

Edzelf, 2018-02-09:
Looks very nice!

Primus007, 2018-02-19:
Nice Work!! 
Please can you upload a circuit diagram? I would like to know how to wire the Audio Signal Transformers. I am using the minimal Version with an Wemos D1.

Thank you!

Edzelf, 2018-05-30:
Good stuff.  I ordered one.

Primus007, 2018-09-12:
> 
> 
> i Use thise:
> 
> The BA3121,BA3121F and BA3121N are ground isolation amplifiers developed for use in car audio applications.
> this does not cut the frequency

please can you give us a wiring diagram or a fritzing?

ART032, 2018-09-13:
> > i Use thise:
> > The BA3121,BA3121F and BA3121N are ground isolation amplifiers developed for use in car audio applications.
> > this does not cut the frequency
> 
> please can you give us a wiring diagram or a fritzing?

HI, I took it from datashit.

http://www.datasheet-pdf.com/mobile/233367/BA3121N.html

but, you can bay it in ALYEXPRESS 
https://ru.aliexpress.com/item/12V-Audio-Noise-Filter-Loop-Isolator-Eliminating-Noise-from-Car-Amplifier-PC-BA3121N/32570550241.html
(DC 12V)

Primus007, 2018-09-13:
Thank you.
and you connect this between vs1053 and PAM8610 ?

ART032, 2018-09-13:
Yes.

ememem13, 2019-03-04:
Can I get a circuit diagram?
Condenser, resistance values, etc.

ememem13, 2019-03-04:
> Kaphatok egy áramköri rajzot? 
> Kondenzátor, ellenállás értékek stb.

> Hi locki-cz 
> Elnézést a késleltetett válaszért: 
> 3 modul, az ESP32, a VS1053 és a 3.3v-os megjelenítés az erősítő 12v-en működik, az ESP32 és a VS1053 egy lineáris AMS117 szabályozóval rendelkezik, amely 12 V-os áramforrásból túlmelegszik és pazarolja az energiát. Megoldás, megvettem az Ebay-től egy kapcsolt 5v-es szabályozót, amely csökkenti a 12v-től 5-ig terjedő teljesítményt, és betáplálja az ESP-t és a VS-t. A kijelzőt az ESP táplálja.
> 
> Megtámadta a rádió PCB-jét. 
> Továbbá azért, mert az ESP32Radio-t AlarmClock-ként használom (jelenleg 3 munka): a kijelző háttérvilágítása sötétben sötét. Az ESP egyik analóg portjához csatlakoztatott LDR és néhány vezetési tranzisztorral rendelkező PWM kimenet táplálja az LCD háttérvilágítását, ha ezt az opciót nem használják, csak az R2-t rontjuk, és ne telepítsük az LDR-t, a tranzisztorok ellenállását. Kérjük, vegye figyelembe a különböző ESP32 modulokat, amelyek különböző pinoutok állnak rendelkezésre, az általam használt modulok (csak az eBay-en töltöttem): 
> * DC-DC Buck Step Down átalakító szabályozó tápegység modul 9V 12V 24V - 5V 3A 
> * ESP-32 ESP32S Fejlesztési tanács 2,4 GHz-es kettős módú WiFi + Bluetooth antenna-modul M 
> * 1,8 hüvelykes mini soros SPI TFT LCD modul PCB adapter ST7735S megjelenítéséhez
> * 1PCS VS1053 MP3 modul fejlesztés Bard (fedélzeti rögzítési funkció) MO 
> _PAM8610 digitális audió erősítő tábla 2_ 15W kétcsatornás sztereó AMP osztály D 
> * 10X audio jel transzformátorok 600: 600 Ohm Európa 1: 1 EI14 szigetelő transzformátor 
> _1/2 / 5PCS Rotary kódoló ___ EC11 _kapcsolóval Audió digitális potenciométer 20mm fogantyú ___6___ 6_ (4,3-15 mm) Pillanatnyi tapintható nyomógombos kapcsoló Kicsi Mini Micro PCB-szerelt 
> * kondenzátorok, dióda csatlakozók, minden helyi elektronikai áruházban 
> (minden arrounf 20 €)
> 
> Szükség esetén kérjük, kérdezze meg, és elküldöm neked az LCD háttérvilágítás vezérlésére szolgáló összes kiegészítő kódot :) 
> A projekt befejezéséhez csak egy akril házat kell vásárolnom, amely az összes elektronikát mutatja (steampunk stílus :)) 
> [ESP32Radio_pcb.pdf ](https://github.com/Edzelf/ESP32-Radio/files/1710907/ESP32Radio_pcb.pdf)
> [ESP32radio_silk.pdf](https://github.com/Edzelf/ESP32-Radio/files/1710908/ESP32radio_silk.pdf)
> [image]

Can I get a circuit diagram?
Condenser, resistance values, etc.

Rhanem, 2019-03-30:
Hello ... I just received Audio Transformer 600: 600 Ohm and I can now play the esp32 radio with the same power supply without noise. 
i buyed this: https://www.aliexpress.com/item/2657136/32827431499.html

Primus007, 2019-03-31:
@ Rhanem, i have bought the same Transformers but i don't know ho to connect/wire. Can you please explain me how to wire the transformers or make a Photo ? Thank you.

Rhanem, 2019-03-31:
this is the diagram that I used and it also works for me without the 100 ohm resistors
[image]

Edzelf, 2019-03-31:
In fact, the common line of the input should be cooected to the common line of the vs1053 output, not to ground.

Rhanem, 2019-03-31:
yes, I mean the common line.. Thanks for correcting

Primus007, 2019-04-01:
@Rhanem, only to eliminate misunderstandings. 
You have it connected between VS1053 and PAM8610 ? VS1053 to in and Out to PAM8610 ?

Rhanem, 2019-04-01:
> @Rhanem, only to eliminate misunderstandings.
> You have it connected between VS1053 and PAM8610 ? VS1053 to in and Out to PAM8610 ?

Yes, exactly

h1aji, 2019-04-21:
Will add my 10 cents in this conversation :)
I got tired of fighting this noise. Tried transformers and it didnt work. 
Instead of using audio transformers and DC-DC step down converter, 5V can be supplied by isolated DC-DC converter like Mornsun B1205S-2W 
[image]

magpa, 2019-05-28:
Hi guys. Compliments to eXadra for very good solution. 
I have almost the same problem. I'd like to connect out of intercom to adc0 of ESP32 pin36.
I put this amplifier:    http://educ8s.tv/part/PAM8403 I put also 600:600isolator (one terminal at ground ) and othe to esp32 pin36. In my phone I hear much noise. If I use only amplifier and the output is a localy speaker I hear very good sound.
If I use microphone: 
https://it.aliexpress.com/item/32810488574.html?spm=a2g0s.9042311.0.0.27424c4dqC5hmb 
and link Vout directly to esp32 pin 36 I hear a very good sound speaking directly to microphone. 
What I have to put between out of my amplifier and pin36 to obtain good sound? Help me please I stay on this problem from day and day. 
Thanks Thanks a lot for your time. I trust on you.

ART032, 2019-05-29:
I soldered the Board but the power volt and all the noise went completely. the lower and upper frequencies were not cut off.

https://ru.aliexpress.com/item/12V-Audio-Noise-Filter-Loop-Isolator-Eliminating-Noise-from-Car-Amplifier-PC-BA3121N/32570550241.html?isOrigTitle=true

BA3121N
http://partnumber.datasheet.netdna-cdn.com/pdfhtml/0582/233367/page2.png

magpa, 2019-05-29:
very good, but you link the output of this to esp32 adc_0 pin 36? 
do you understand also why if I USE MICROPHONE like  https://it.aliexpress.com/item/32810488574.html?spm=a2g0s.9042311.0.0.27424c4dqC5hmb the sound is very good. It is very strange. When I use amplifier for output intercom and put output of amplifier to ESP 32 too much noise. 
I'll try anyway your solution. 
Thanks a lot

ART032, 2019-05-29:
the cause of the noise is a single digital and analog ground. PAM separates these lands. and noises from the digital circuit do not fall into the analog circuit of the amplifier.

thijstriemstra, 2020-07-02:
> Instead of using audio transformers and DC-DC step down converter, 5V can be supplied by isolated DC-DC converter like Mornsun B1205S-2W

Will give the isolated DC-DC converter a try! I'll use a PAM8610 and [Recom RO-1205S/P](https://recom-power.com/pdf/Econoline/RO.pdf):

[image]

h1aji, 2020-07-05:
@thijstriemstra 1W may not be enough

thijstriemstra, 2020-07-05:
> 1W may not be enough

The input device is a YX5300 Serial MP3 Player (Catalex Module) and some 6ohm JVC speakers I found in a thrift store. Before I could hear the SD-card spinning over the speakers and with this isolator it's all good. I agree 1W is low but for my purposes it's loud enough. But 2W is not enough either than is it (e.g. the Mornsun B1205S-2W)? You don't want to know how much time I spent trying to get rid of this noise and I'm so glad I came across this thread.