Edzelf/ESP32-Radio #67: Feature Request: OGG encoding and streaming ?

zapf2000, 2017-12-11:
Hi Ed,
wouldn't it be possible to use the VS1053 as an OGG live encoder and stream the content to an icecast server ? I know this is the complete opposite to a webradio player, but i think the hardware is capable of doing it. For me personally, it would be a great compact method to do webradio the easy way, without having a PC running.

Edzelf, 2017-12-11:
It is possible, in fact I tried that.  But the difficulty is the synchronisation of the sender with the receiver(s).  If a receiver is too slow (bitrate is a little off), the sender has to skip some frames.  If a receiver is too fast, the sender has to insert some extra frames.

zapf2000, 2017-12-11:
Well, I think you mention that the ESP plays the server for the clients. What I mean is that the esp streams the packets out to an icecast server (running anywhere), which handels the client connections. The icecast would see a normal streaming source, but not from a standard encoder (ices, butt, nicecast). The esp is the streaming source. I think it's just an implementation of the libshout getting the packets from the esp and VS1053.

Edzelf, 2017-12-11:
In that case it would be possible, I think.

zapf2000, 2017-12-12:
Well if you need an icecast server to test, feel free to contact me directly. I can provide a testing server, that you use whenever you like (also when you're not testing ;))

Aggebitter, 2017-12-22:
Well this is exactly what I'm working on. but only P2P