Edzelf/ESP32-Radio #68: Can't run on new firmware version

svins01, 2017-12-14:
`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034

D: Include file config_html has the wrong version number!  Replace header file.
D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Wed, 13 Dec 2017 12:10:00 GMT.  Free memory 180864
D: Partition nvs found, 20480 bytes
D: Read 40 keys from NVS
D: pin_ir pin set to -1
D: pin_enc_clk pin set to -1
D: pin_enc_dt pin set to -1
D: pin_enc_sw pin set to -1
D: pin_tft_cs pin set to 15
D: pin_tft_dc pin set to 2
D: pin_sd_cs pin set to 21
D: pin_vs_cs pin set to 5
D: pin_vs_dcs pin set to 16
D: pin_vs_dreq pin set to 4
D: pin_shutdown pin set to -1
D: pin_spi_sck pin set to 18
D: pin_spi_miso pin set to 19
D: pin_spi_mosi pin set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Start TFT
Guru Meditation Error of type LoadStoreError occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x40001277  PS      : 0x00060930  A0      : 0x800ec800  A1      : 0x3ffd32e0  
A2      : 0x40080e70  A3      : 0x3f40223e  A4      : 0x00000004  A5      : 0x3ffcf7ac  
A6      : 0x44530064  A7      : 0x44530064  A8      : 0x80000000  A9      : 0x3ffd3300  
A10     : 0x3ffcf7a8  A11     : 0x00000002  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffd24e0  SAR     : 0x00000010  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40080e70  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

Backtrace: 0x40001277:0x3ffd32e0 0x400ec800:0x3ffd32f0 0x400ec819:0x3ffd3310 0x400da1ff:0x3ffd3380 0x400d9643:0x3ffd33b0 0x400d60b7:0x3ffd33e0 0x4012730a:0x3ffd3450

CPU halted.
`
from 13-11-2017 everything works

Edzelf, 2017-12-15:
Notice the first debug line.  You have to replace "config_html.h".
If that does not help, use the "ESP exception decoder" of the IDE to narrow down the cause of the crash.

svins01, 2017-12-15:
the crash occurs if I connect a LCD or SD
config_html.h explain in more English is not my native language

Edzelf, 2017-12-15:
Hold on, I can reproduce it here.  I will try to fix it.

Edzelf, 2017-12-15:
Download config_html.h from the github code page.

Edzelf, 2017-12-15:
The problem is probably a weak power supply.  I can reproduce the error if I add an extra device to the powersupply. The LCD display starts normally when I use a good powersupply.  Try to run the radion om a powerbank.

svins01, 2017-12-15:
Thanks, tonight will try
the voltage is unlikely. on old firmware everything works fine.

svins01, 2017-12-16:
if Esp32_radio_init not add 
preferences.putString ( "vs_cs", "5" ) |
preferences.putString ( "vs_dcs", "16" ) |
preferences.putString ( "vs_dreq", "4" ) | 
do not start

svins01, 2017-12-16:
did not help all the same

svins01, 2017-12-16:
what we found out
1. download Esp32_radio_init.ino
```
//********************************************************************************************
// Initialize the preferences op ESP32-radio.  They can be edited later by the webinterface. *
//********************************************************************************************
// 27-04-2017, ES: First set-up, derived from preferences example sketch.                    *
//********************************************************************************************

#include <Preferences.h>

// Note: Namespace name is limited to 15 chars.  Same name must be used in ESP32-radio.
#define NAME "ESP32Radio"
// Adjust size of buffer to the longest expected string for nvsgetstr
#define NVSBUFSIZE 150

Preferences preferences ;         // Instance of Preferences

//******************************************************************************************
//                              S E T U P                                                  *
//******************************************************************************************
void setup()
{
  String       str ;
  
  Serial.begin ( 115200 ) ;
  Serial.println() ;
  preferences.begin ( NAME, false ) ;                 // Open for read/write
  preferences.clear() ;                               // Remove all preferences under opened namespace
  // Store the preferences for ESP32-radio
  
  preferences.putString ( "mqttbroker",   "none" ) |
  preferences.putString ( "mqttport",     "1883" ) |
  preferences.putString ( "mqttuser",     "none" ) |
  preferences.putString ( "mqttpasswd",   "none" ) |
  preferences.putString ( "mqtttpreset",  "none" ) |
  //
  preferences.putString ( "wifi_00", "HOME/12344321" ) |
  preferences.putString ( "wifi_01", "Gargoyle/" ) | 
  //
  preferences.putString ( "volume", "72" ) |
  preferences.putString ( "toneha", "0" ) |
  preferences.putString ( "tonehf", "0" ) |
  preferences.putString ( "tonela", "0" ) |
  preferences.putString ( "tonelf", "0" ) |
  preferences.putString ( "vs_cs", "5" ) |
  preferences.putString ( "vs_dcs", "16" ) |
  preferences.putString ( "vs_dreq", "4" ) |
  preferences.putString ( "enc_clk", "25" ) |
  preferences.putString ( "enc_dt", "26" ) |
  preferences.putString ( "enc_sw", "27" ) | 
   //
  preferences.putString ( "preset", "00" ) |
  preferences.putString ( "preset_00", "ep256.hostingradio.ru:8052/europaplus256.mp3  #Europa +" ) ; 
  //
  preferences.putString ( "clk_server", "pool.ntp.org" ) ;        // Setting for Time Of Day clock on TFT
  preferences.putString ( "clk_offset", "3" ) ;
  preferences.putString ( "clk_dst",    "0" ) ;
  preferences.end() ;
  delay ( 1000 ) ;
}

//******************************************************************************************
//                              L O O P                                                    *
//******************************************************************************************
void loop()
{
  Serial.println ( "ESP32_radio_init completed..." ) ;
  delay ( 10000 ) ;
}

```
 2. download - Esp32_radio.ino
**radio works**
if Esp32_radio_init.ino to add a line to the - preferences.putString ( "sd_cs", "21" ) |- then there is a collapse
Also if the collapse happens in the web browser tab, and manually add pin_sd_cs = 21

Edzelf, 2017-12-16:
I do not understand it completely, but in the preferences the pinnames are changed.  New names are:
    - "pin_ir"
    - "pin_enc_clk"
    - "pin_enc_dt"
    - "pin_enc_sw"
    - "pin_tft_cs"
    - "pin_tft_dc"
    - "pin_sd_cs"
    - "pin_vs_cs"
    - "pin_vs_dcs"
    - "pin_vs_dreq"
    - "pin_shutdown"
    - "pin_spi_sck"
    - "pin_spi_miso"
    - "pin_spi_mosi"

svins01, 2017-12-16:
explain what I'm doing wrong?
I Add
pin_sd_cs = 21 
to activate SD and all at once collapse

Edzelf, 2017-12-18:
Please show me the preferences on the config page, so I can try to reproduce the error.
Are you sure GPIO21 is connected correctly?

svins01, 2017-12-18:
I managed to download your latest firmware, but the error not found, I only added my settings in your config by default

Edzelf, 2017-12-18:
There is a conflict:
gpio_**21** = station = icecast.omroep.nl:80/radio1-bb-mp3
and:
pin_sd_cs = **21**

Please remove "gpio_21 = station = icecast.omroep.nl:80/radio1-bb-mp3" if you want to use pin 21 for SD.