Edzelf/ESP32-Radio #69: Will not compile on Nano32

Aggebitter, 2017-12-22:
Did a lot of debugging, but still some issues 

I think links in source to all used libs would be useful.

Edzelf, 2017-12-22:
I have added the URLs of the the special libs to the documentation.

Aggebitter, 2017-12-23:
Thanks Now it builds. Add this to source ? 
##elif defined ( ARDUINO_FEATHER_ESP32 ) || defined ( ARDUINO_NANO32 ) 
#define VS1053_CS     32
#define VS1053_DCS    33
#define VS1053_DREQ   15
#define SDCARDCS      14

Aggebitter, 2017-12-23:
Now working on code for 
#elif defined ( ARDUINO_Heltec_WIFI_LoRa_32 ) 
It got I2C OLED 128x64 and a Semtec SX1276 LoRa RF module 
VS1053b is running and playing but initial volume register test and SD do not pass yet. more fiddling with pins needed 
porting to u8g2 lib (faster and cleaner than adafruit I've heard) need the speed due I2C

Edzelf, 2017-12-23:
You do not need #defines for VS1053_CS, VS1053_CS, etc.
Just specify the pins in the preferences like:
pin_vs_cs = 32        # GPIO Pin number for VS1053 CS
pin_vs_dcs = 3      # GPIO Pin number for VS1053 DCS
pin_vs_dreq = 15      # GPIO Pin number for VS1053 DREQ
pin_sd_cs = 14       # GPIO Pin number for SD card CS

Aggebitter, 2017-12-23:
Well will do it as a #elif defined to adjust the pre compiler for the I2C OLED display on this board (Heltech WiFi LoRa 32).  If it got a Display why not use it ;-)
Will GIT it at my github later when more stable.

Might have to fork yours later. Going to reuse the code for some other stuff (Ogg low bitrate streaming over the LoRa RF)

Noticed some "Ear" danger when testing the volume bar on the OLED. The web interface does a round robin when passing 0 volume. Then it starts over at 100% volume . A funktion for holding it at zero should be in there.

Thanks for a good job on porting! 👍

closed 2017-12-23 by Aggebitter

koskee, 2017-12-24:
In regards to the volume jump, it is because it is defined as an unsigned value... The variable will roll over to a value of 255 and not raise an error. (and although the code will change it to 100 it still isn't what we want)  

All I did was change the uint8_t to int8_t, then check for the volume going below 0 and adjust if necessary.