Edzelf/ESP32-Radio #72: how to do it Power on - not autoplay

udavst, 2018-01-02:
Tell me how to make sure that the radio does not turn on itself, when power is applied?
And for pin_shutdown to be used only when STOP is pressed, but not when switching stations?

PS Sorry for my English, this is an interpreter.

Edzelf, 2018-01-03:
Solution 1:
Just add the line:
currentpreset = ini_block.newpreset ;
just before the end of the setup() function.

Solution 2:
remove lines 3049, 3050 and 3052 ( 2 januari 2018 version ).

closed 2018-01-03 by Edzelf

udavst, 2018-01-03:
This is inconvenient (1 point), now the radio is turned on at all without a preset, and after switching on from the IR does not play anything, there are still stations to switch.
I want to be in the STOP mode when starting the radio.