Edzelf/ESP32-Radio #73: VS1053 replacement with I2S DAC + software stream decoder

jgrulich, 2018-01-04:
And it's here again. The question about VS1053 replacement with the I2S DAC and software stream decoding.
This is really wonderful project, but there is still remaining VLSI decoder.
There are several issues about this:
1) It's not easy to get the IC from official distribution and is found, the price is over $20 per part !!!
Forget the Ali boards due to the following point.
2) Another point, which was never mentioned here is the "Licenses" paragraph, see the page 6. of the Datasheet.

Is there an idea when, or if ever will be the version with the software decoding?

koskee, 2018-01-05:
The original link supplied by @Edzelf to buy the module on Ali at a cost of approx $10) should get you a vs1053b chip (although whether it is an authentic VLSI chip and not just a Chinese knockoff remains to be seen - but I haven't had any issues to date).

I have had experience with getting a vs board with a lower model chip attached (vs1003 I think?)
The thing is, it still works, but isn't able to decode AAC streams/files. (for instance, IHeartRadio) However, as per the licencing section, if you don't have a licence for AAC, you're technically breaking the law by using it (and especially if you're making money from such a thing) as it remains a patented codec.  Meaning that you aren't supposed to use it anyway, and (probably) should have it disabled as well, to be entirely on the right side of the law. There are several formats that the datasheet mentions that are under a similar status.

So, while software decoding is possible,  (and an ESP32 implementation for MP3 does exist - Google it),  one must consider that the benefit would do little more than save $10 on the vs1053b chip, but would restrict its use to only MP3 streams/files. (MP3 became patent-free on 16 April 2017, when U.S. Patent 6,009,399, held and administered by Technicolor, expired.)

Unfortunately, software decoding or not, most of the licence(s) would still be required and if anything, this audio decoding board gives a bit of plausible deniability for having the feature available -  by using the same logic VLSI uses in the datasheet. 

Software decoding *could* negatively effect the reliability because then you are requiring the processor to perform decoding on top of everything else it does, (but would likely be solved with a little optimization regardless.) 
What wouldn't be so easily solved would be the additional codebase bloat required to store the algorithm(s) & codecs that the current system supports. (consuming more flash & RAM) 
It would also increase the complexity of the code and could make maintainability a big hassle. (also not very noob-friendly..) 
Another point to consider (that you touched on above) is that if @Edzelf did decide to provide code for some of the other formats the board supports, it could quite likely open the door for patent infringement/lawsuit against him. 

Now, I'M NOT A LAWYER, but as far as I know/have heard, companies don't normally go after a single user too often, as it would likely not be worth the expense, but could be quite likely to go after someone contributing to widespread infringement).. 

I see this project as a best of both worlds - it's still cheap enough, and using an audio co-processor to offload the decoding means there's a lot more CPU cycles available to customize the hell out of it, in any way you want.

jgrulich, 2018-01-05:
@koskee thank's a lot for such a deep investigation into the Licenses issue and explanation. It only confirms my worry. The root cause why I started to look around was, that @Edzelf did excellent work and I liked to somehow to donate his effort with some nice board like made Dimitris for the Ka-Radio here:
https://hackaday.io/project/11570-wifi-webradio-on-esp8266-or-esp32/log/52061-some-hardware-projects-based-on-karadio.
But there is this most important paragraph in the Licenses: "**This product is protected by certain intellectual property rights of Microsoft and cannot be used or further distributed without a license from Microsoft.**"
If I understood well, it's prohibited not only the usage = playing the file, but even the usage the IC on the board itself and it's distribution, independently on the software code used.
On the other hand, based on the last paragraph of the Licenses, the distribution of the IC on the board is probably not a issue, but it's important to disable the AAC, WMA, PS and SBR usage as default.
As you've mentioned, the VS1003 is the major solution point. There is no AAC, PS and SBR at all and only WMA need to be disabled by code. Personally guess, that this format is not needed at all for radio streaming.

koskee, 2018-01-05:
@jgrulich Yep, I'm pretty sure WMA based radio stations are relatively rare, as I can't recall seeing one recently, if at all.

And in that case - if you were ok with having ONLY MP3 support - then a software implementation would at least start to make sense. That being said, it would render the IHR sections of the code useless, along with a good amount of other stations as well.

@Edzelf I wonder if you couldn't utilize the esp32 webradio project as a loosely defined library that can be enabled (through compiler pre-proc defines or something equivalent) for the case where somebody either doesn't have their vs10xx board (yet) or had theirs fail? Then it would be functional out of the box with minimal hardware, and wouldn't take up any extra flash/RAM space once they do have one to utilize. 

It doesn't seem worthwhile to code it yourself without a bigger goal to do so most likely,  but since the code already exists for this architecture, maybe it would be easy to drop in the decoding logic? 

Ps.. Ed, Did you ever end up having any luck with getting that SSL/TLS  client mode functionality going? 

From the reading I've done regarding hosting a server capable of SSL , it seems people have got a client working with the arduino libraries, but when it comes to server code, the only library implementation currently available seems to be written for the IDF only. Or at least there isn't any sort of arduino server example that I've been able to locate at all... 

I haven't had time to figure out how easily one can cross compile between the two environments or specifics of what would be required.

jgrulich, 2018-01-05:
@koskee Yes, you're right that only MP3 playback is great degradation of this amazing project and it's functionality. AAC is really important point. Not sure, if already exists software decoder library for that.
For only MP3 version the software decoder with I2S is not a benefit, because the official distribution cost of VS1003 is only $4, which is equal to I2S approach.
The WMA playback may be disabled by firmware at all and than everything seems to be legal.
And for those, who will like the full featured project, there may be customized board option with the VS1053 and no firmware inside.
I don't want to load the @Edzelf more, than is really needed. My target is to help him with the project HW support and donation, for sure if he will agree with that.

jgrulich, 2018-01-05:
I've found this lib for software decoding: https://github.com/earlephilhower/ESP8266Audio, but the AAC still need to be licensed.

koskee, 2018-01-05:
As far as hardware, I'm nearing completion of a third prototype board, which I'm hoping could double as a final prototype as well.
For some reason on this project, I keep opting to add more features and having to rework the whole thing. Its getting to be a bit tedious. So going to finalize it asap and then I can work on v2, I suppose

jgrulich, 2018-01-05:
@koskee please be so kind and provide more info about your HW at my e-mail: jan@grulich.eu to prevent double work.

jgrulich, 2018-01-05:
Side of WMA, AAC Licensing issues, I've found similar with the IHR. Now it's not available for Europe any more. Than only MP3 playback is enough and VS1003 seems to be the right solution of this Licensing issue.

jgrulich, 2018-01-08:
Solved. I've used ARM SoC based board with software MP3 decoder.
It doesn't need any VS board and it's with the same price of $10 like ESP-32 and VS1053 board.

closed 2018-01-08 by jgrulich