Edzelf/ESP32-Radio #74: Change the volume through MQTT

udavst, 2018-01-04:
Works perfectly team theme: **P06A4AE31/command** data: **stop**, but the theme does not work at all: **P06A4AE31/volume** data: **55**. How to specify the desired volume through mtqq?

eXadra, 2018-01-05:
Hello all.

Just sen dthe command below (in this case volume 88)
**P37A4AE30/command volume=88**

ps: Change topic to your own :)

udavst, 2018-01-05:
I can do any topic, but I need only the numbers in the data. All MQTT programs for android are able to change the number in the data field (1, 2, 88 ..), but they can not write text ( aka volume = ).

udavst, 2018-03-29:
Topic:**P37A4AE30/command** Case:**volume=55** - work, но такой case очень неудобен. Как сделать, чтобы case содержал только цифру громкости (55)?
Topic:**P37A4AE30/command volume=** Case:**55** - not work
Topic:**P37A4AE30/command volume** Case:**55** - not work

Edzelf, 2018-03-29:
You have the choice of changing the android app or changing the radio sketch.  It would not be too difficult.

bagotu, 2018-11-15:
I see requests in MQTT.fx, but on ESP32, Volume level does not work

{ mqtt=">[mosquitto:P05A4AE30/volume:command:*:${command}], - not work

<[mosquitto:P05A4AE30/volume:state:default]" } - it is work!

can you help me?