Edzelf/ESP32-Radio #75: VS1053 not properly installed!

AST-esp-rasp, 2018-01-06:
Hello after checking twice the connection and using a second VS1003/1053 Module i stil get the error message "VS1053 not properly installed!"
Is there any known reason?
Do i have to pull up / down some of the GPIO ports connected to the Decoder module?
k.r.
Andreas

Edzelf, 2018-01-06:
Is the reset  line high?

AST-esp-rasp, 2018-01-06:
found the problem at least the program is running now.

was due to missing settings for the  
#define VS1053_CS     5
#define VS1053_DCS    16
#define VS1053_DREQ   4
because i have used the esp32_radio_init.ino before and my preferences there did not set the definitions
Is this esp32_radio_init.ino still necessary? or does the normal program working without previously using it?
But now i get some noise out of my headphones but it sounds shredded

Edzelf, 2018-01-06:
The init sketch is only necessary to allow the webinterface to work.  The pins must be set in the preferences.

AST-esp-rasp, 2018-01-06:
okay found it was a broken gnd cable everything works now

closed 2018-01-06 by AST-esp-rasp

some89, 2018-11-06:
hi, any other posibilitis for this fault? I have chcked every connection and rebuild it 3 times.

I have only the vs1053 connected nothing else. the reset pin is at 3.8V

the wifi also does not connect, does it connect after verifing the vs1053 board?

D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Connect to WiFi
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e080f  PS      : 0x00060630  A0      : 0x800d2c06  A1      : 0x3ffb1e50  
A2      : 0x0000001c  A3      : 0x3f401317  A4      : 0x3ffca3a8  A5      : 0x3ffb1e8c  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800e080f  A9      : 0x3ffb1e10  
A10     : 0x0000001b  A11     : 0x0000001c  A12     : 0x3f401317  A13     : 0x3ffb1ed0  
A14     : 0x3ffb1e50  A15     : 0x00000008  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x400e080f:0x3ffb1e50 0x400d2c03:0x3ffb1ef0 0x400d3ac9:0x3ffb1f10 0x400da06c:0x3ffb1f30 0x4013bffb:0x3ffb1fa0

Rebooting...

Edzelf, 2018-11-06:
Try to run without any wites attached.  It should connect to WiFi then.  Show me the logging.

some89, 2018-11-06:
No luck

this is just the uploaded  ESP32 radio skatch, the init skatch was left untouched. I have added my wifi login under defultprefs.

this is esp32 without anything

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 04 Oct 2018 07:22:32 GMT.  Free memory 233608
D: Display type is NEXTION
D: Partition nvs found, 20480 bytes
D: Read 31 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to -1
D: pin_vs_dcs set to -1
D: pin_vs_dreq set to -1
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Create list with acceptable WiFi networks
D: Added XUK5 to list of networks
D: Added ADSL to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 2
D:  1 - T-2_40cb2b                Signal: -55 dBm, Encryption WPA_WPA2_PSK, 
D:  2 - XUK5                      Signal: -70 dBm, Encryption WPA2_PSK, Acceptable    **this is the added network wifi_00**
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqtttpreset with parameter none
D: Command: mqttuser with parameter none
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Connect to WiFi
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e080f  PS      : 0x00060630  A0      : 0x800d2c06  A1      : 0x3ffb1e50  
A2      : 0x0000001c  A3      : 0x3f401317  A4      : 0x3ffca3a8  A5      : 0x3ffb1e8c  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800e080f  A9      : 0x3ffb1e10  
A10     : 0x0000001b  A11     : 0x0000001c  A12     : 0x3f401317  A13     : 0x3ffb1ed0  
A14     : 0x3ffb1e50  A15     : 0x00000008  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x400e080f:0x3ffb1e50 0x400d2c03:0x3ffb1ef0 0x400d3ac9:0x3ffb1f10 0x400da06c:0x3ffb1f30 0x4013bffb:0x3ffb1fa0

Rebooting...

Edzelf, 2018-11-06:
Add your wifi credentials to the init-sketch and run it once.
Changing defaultpresf has no effect at this stage.  It is only used in the config page of the webinterface.

trombose009, 2020-09-03:
I have this issue:
`
D: Scan Networks
D: Scan completed
D: Number of available networks: 2
D:  1 - dampfmaschine2            Signal: -73 dBm, Encryption WPA2_PSK, 
D:  2 - fritzboxdampf             Signal: -83 dBm, Encryption WPA_WPA2_PSK, 
D: End of list
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Start CH376
`

I did upload the init ino with wifi credentials. 
Then it shows me:
"ESP32_radio_init completed..."

Then I uploaded the main file and I got this output. I have no display so I set it to dummy. 
The ESP does not appear in the list of DHCP devices at my router. 

Any Ideas?